ros-drivers/flir_camera_driver #1: compile problem

weisongwen, 2018-07-24:
hi, thanks for your great work.  I got this issue when compiling.
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avformat_alloc_output_context2@LIBAVFORMAT_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator+=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_write_trailer@LIBAVFORMAT_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::replace(unsigned long, unsigned long, char const*, unsigned long)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <char, std::char_traits<char>, std::allocator<char> >(std::basic_ostream<char, std::char_traits<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::basic_ofstream<char, std::char_traits<char> >::basic_ofstream(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::_Ios_Openmode)@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avpicture_alloc@LIBAVCODEC_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_last_of(char, unsigned long) const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `VTT for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_open2@LIBAVCODEC_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__throw_out_of_range_fmt(char const*, ...)@GLIBCXX_3.4.20'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_first_not_of(char const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::append(char const*, unsigned long)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator+=(char const*)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign(char const*)@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avformat_new_stream@LIBAVFORMAT_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream(std::_Ios_Openmode)@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_flush_buffers@LIBAVCODEC_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::_M_create(unsigned long&, unsigned long)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::c_str() const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_erase(unsigned long, unsigned long)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::begin()@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream(std::_Ios_Openmode)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::basic_filebuf<char, std::char_traits<char> >::operator=(std::basic_filebuf<char, std::char_traits<char> >&&)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_aux(unsigned long, unsigned long, unsigned long, char)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_encode_video2@LIBAVCODEC_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `typeinfo for std::ios_base::failure[abi:cxx11]@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::rfind(char const*, unsigned long) const@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_register_all@LIBAVCODEC_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_first_not_of(char const*, unsigned long) const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::append(char const*)@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_free@LIBAVUTIL_FFMPEG_54'
usr/lib/libSpinnaker.so: undefined reference to `std::runtime_error::runtime_error(std::runtime_error const&)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::[100%] allocator<wchar_t> >::find(wchar_t const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::substr(unsigned long, unsigned long) const@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::~basic_stringstream()@GLIBCXX_3.4.21'
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_frame_alloc@LIBAVUTIL_FFMPEG_54'
usr/lib/libSpinnaker.so: undefined reference to `std::basic_istream<wchar_t, std::char_traits<wchar_t> >& std::getline<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >(std::basic_istream<wchar_t, std::char_traits<wchar_t> >&, std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >&, wchar_t)@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const@GLIBCXX_3.4.21'
[... cut ...]
/home/wenws/blackflyS/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avio_open@LIBAVFORMAT_FFMPEG_56'
usr/lib/libSpinnaker.so: undefined reference to `VTT for std::__cxx11::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4.21'
usr/lib/libSpinnaker.so: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(unsigned long, char, std::allocator<char> const&)@GLIBCXX_3.4.21'

do you have any idea on this?

weisongwen, 2018-07-24:
Ubuntu
ros jade

weisongwen, 2018-07-24:
Hi, 
can the ros version be the problem ?
thanks

weisongwen, 2018-07-25:
hi @mhosmar-cpr ,
thanks for your kind information. I test your package in Ubuntu 16.04 + ROS kinetic. the package can be successfully be compiled. My camera is BlackFlyS (BFS-U3-31S4C-C). 

I plugin the usb of camera into the computer and launch the launch file testspinnaker.launch. However, the output shows:

number of interfaces detected: 3
of connected camera : 0
No Cameras Connected!

this seems the driver cannot identify the camera. Do I need to configure something in the package?

thanks

mhosmar-cpr, 2018-07-25:
@weisongwen The test Spinnaker launch file was only really used early in testing. Have you tried camera.launch?

weisongwen, 2018-07-25:
hi, @mhosmar-cpr  thanks for your reply. I just tried the camera.launch and it still do not work. Following output is obtained. 
[image]

it seems that the driver cannot find the camera.

weisongwen, 2018-07-25:
if i use the lsusb command, i can get the following output
[image]
the computer seems detected the USB device of the camera. very appreciate it if you can provide some advice to me.

best
welson

mhosmar-cpr, 2018-07-25:
There is a possibility you need a udev rule which is normally provided by the Spinnaker SDK.

can you run the `spin-conf` script which will be in the build folder if you have compiled once before.
In this folder:
`workspace/build/spinnaker_camera_driver/spinnaker-1.13.0.31-amd64/`

Alternatively you could install the udev rule yourself which is in the debian folder.

you may need to reboot once the udev is installed.

weisongwen, 2018-07-26:
@mhosmar-cpr  thanks for your kind information and reply. I successfully install the udev rule and the driver can connect the camera now. But after I launch the camera.launch, there is no camera related topic being published. following is the output (I set the output="screen" for the other node): 
[image]

run rostopic list, the output is:
 
[image]

Do you have any advice on this?

best 
welson

tfoote, 2018-07-26:
@weisongwen please don't use screenshots, but copy and paste the text so it's searchable and copyable in the future.

weisongwen, 2018-07-26:
@tfoote  ok, sorry for this

weisongwen, 2018-07-26:
@mhosmar-cpr  thanks for your help. The probelm is solved.
best
welson

closed 2018-07-26 by mhosmar-cpr