ros-drivers/flir_camera_driver #6: Failed to set configuration

alfred31, 2018-08-07:
Hello,
After having installed the driver in my ROS workspace I am trying to run the camera.launch file. The camera is being detected but it displays an error, so the topics of the camera are empty. Here is the outcome:
SUMMARY
========

PARAMETERS
 * /camera/spinnaker_camera_nodelet/frame_id: camera
 * /camera/spinnaker_camera_nodelet/serial: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    image_proc_debayer (nodelet/nodelet)
    spinnaker_camera_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [11922]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 15d2bd2a-9a0b-11e8-9546-c85b76a8f2ac
process[rosout-1]: started with pid [11935]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [11942]
process[camera/spinnaker_camera_nodelet-3]: started with pid [11953]
process[camera/image_proc_debayer-4]: started with pid [11954]
[ INFO] [1533623304.401713233]: Initializing nodelet with 4 worker threads.
[ INFO] [1533623306.203160039]: [SpinnakerCamera]: Number of cameras detected: 1
[ INFO] [1533623306.303620179]: [SpinnakerCamera::connect]: Using Serial: 18013295
[ INFO] [1533623306.303692870]: [SpinnakerCamera::connect]: Detected device type: GEV
[ INFO] [1533623306.740861063]: [SpinnakerCamera::connect]: Camera model name: Blackfly S BFS-PGE-50S5C
[ INFO] [1533623306.751486209]: [SpinnakerCamera]: (18013295) DeviceLinkThroughputLimit set to 12500000.
[ INFO] [1533623306.990190413]: [SpinnakerCamera]: (18013295) BinningHorizontal set to 2.
[ INFO] [1533623307.004701024]: [SpinnakerCamera]: (18013295) BinningVertical set to 2.
[ WARN] [1533623307.006970626]: [SpinnakerCamera]: (18013295) Feature DecimationHorizontal not writable.
[ WARN] [1533623307.008475084]: [SpinnakerCamera]: (18013295) Feature DecimationVertical not writable.
[ INFO] [1533623307.020437135]: [SpinnakerCamera]: (18013295) OffsetX set to 0.
[ INFO] [1533623307.029691618]: [SpinnakerCamera]: (18013295) OffsetY set to 0.
[ INFO] [1533623307.040225700]: [SpinnakerCamera]: (18013295) Width set to 1224.
[ INFO] [1533623307.048865311]: [SpinnakerCamera]: (18013295) Height set to 1024.
[ INFO] [1533623307.055117262]: [SpinnakerCamera]: (18013295) OffsetX set to 0.
[ INFO] [1533623307.061442761]: [SpinnakerCamera]: (18013295) OffsetY set to 0.
[ INFO] [1533623307.074728797]: [SpinnakerCamera]: (18013295) PixelFormat set to Mono8.
[ INFO] [1533623307.079206047]: [SpinnakerCamera]: (18013295) AcquisitionFrameRateEnable set to 1.
[ERROR] [1533623307.082655973]: Reconfigure Callback failed with error: [Camera::setNewConfiguration] Failed to set configuration: Spinnaker: GenICam::OutOfRangeException= Value 9.619247 must be smaller than or equal 9.619247. : OutOfRangeException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 85) [-2002]

Any help on this?

Thank you very much!

alfred31, 2018-08-07:
I solved the problem by setting the acquisition_frame_rate to 9.5 in the cfg file.

Kind regards

closed 2018-08-07 by alfred31

haozhang666, 2022-04-02:
@alfred31  hello I changed acquisition_frame_rate to 9.5 ,but it do not active . Could you tell me more information? thanks