ros-drivers/flir_camera_driver #7: Failed to display the /camera/image_raw

Rockyzyk, 2018-08-15:
@mhosmar-cpr Hello, 
After having installed the driver in my ROS workspace I am trying to run the camera.launch file. The camera is being detected and I can see the published topics:

`/camera/camera_info
/camera/camera_nodelet_manager/bond
/camera/image
/camera/image_color
/camera/image_color/compressed
/camera/image_color/compressed/parameter_descriptions
/camera/image_color/compressed/parameter_updates
/camera/image_color/compressedDepth
/camera/image_color/compressedDepth/parameter_descriptions
/camera/image_color/compressedDepth/parameter_updates
/camera/image_color/theora
/camera/image_color/theora/parameter_descriptions
/camera/image_color/theora/parameter_updates
/camera/image_mono
/camera/image_mono/compressed
/camera/image_mono/compressed/parameter_descriptions
/camera/image_mono/compressed/parameter_updates
/camera/image_mono/compressedDepth
/camera/image_mono/compressedDepth/parameter_descriptions
/camera/image_mono/compressedDepth/parameter_updates
/camera/image_mono/theora
/camera/image_mono/theora/parameter_descriptions
/camera/image_mono/theora/parameter_updates
/camera/image_proc_debayer/parameter_descriptions
/camera/image_proc_debayer/parameter_updates
/camera/image_raw
/camera/image_raw/compressed
/camera/image_raw/compressed/parameter_descriptions
/camera/image_raw/compressed/parameter_updates
/camera/image_raw/compressedDepth
/camera/image_raw/compressedDepth/parameter_descriptions
/camera/image_raw/compressedDepth/parameter_updates
/camera/image_raw/theora
/camera/image_raw/theora/parameter_descriptions
/camera/image_raw/theora/parameter_updates
/camera/spinnaker_camera_nodelet/parameter_descriptions
/camera/spinnaker_camera_nodelet/parameter_updates
/diagnostics
/rosout
/rosout_agg`

but after running:

> rostopic echo /camera/image_raw

the following error is ocurred:
`[ERROR] [1534300532.174811311]: [SpinnakerCamera::grabImage] Image received from camera 17406103 is incomplete.
[ERROR] [1534300532.176756070]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
^C[camera/image_proc_debayer-4] killing on exit
[camera/spinnaker_camera_nodelet-3] killing on exit
[camera/camera_nodelet_manager-2] killing on exit
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Can't clear a camera because something still holds a reference to the camera [-1004]
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

If I use worth LAN which transfer speed is slow, I can see the camera image instantaneously. But for 10GBASE LAN, I can not see any images.
My camera is a Blackfly S ethernet camera: https://www.ptgrey.com/blackfly-s-color-32-mp-gige-vision-sony-imx265
By the way, I use the virtual box, and spinview in the virtual box is worked for showing images of the camera.

Would you please help me to check the error? Thanks for helping!

mhosmar-cpr, 2018-08-17:
Hello,
Gige cameras have not yet been implemented in this driver. The driver is likely using default parameters which is why you see any output at all. See [here](https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L208) for where gige was planed on being added. 

I've opened #10 to discuss adding gige support.

closed 2018-08-17 by mhosmar-cpr

SeamusShan, 2018-10-08:
> @mhosmar-cpr Hello,
> After having installed the driver in my ROS workspace I am trying to run the camera.launch file. The camera is being detected and I can see the published topics: [...]

Hi Rockyzyk, I am using Blackfly Gige and I had a similar issue reporting incomplete image using this driver. I solved it myself by increasing UDP buffer size to 25MB by 'sudo sysctl -w net.core.rmem_max=26214400' and 'sudo sysctl -w net.core.rmem_default=26214400'. I hope this works for you too.

vivekarvind, 2018-10-22:
@SeamusShan I'm using `Oryx 12.3 MP Color 10GigE` camera and increasing the buffer size gives a rostopic echo correctly. But when I'm using rqt image view to visualize, I get the above error. Any ideas?

SeamusShan, 2018-10-23:
> 
> 
> @SeamusShan I'm using `Oryx 12.3 MP Color 10GigE` camera and increasing the buffer size to 25 MB gives a `rostopic echo` correctly. But when I'm using `rqt image view` to visualize, I get the above error. Any ideas?

Hi vivekarvind, I do not use Oryx, but I guess it is because the computer processing could not catch up with the incoming packets when you visualize the images. Apart from increasing the UDP buffer, you can try increasing the RX DMA ring (another buffer) in network adapter by
sudo ethtool -G <interface name> rx 4096
Hope this works for you.

mhosmar-cpr, 2018-10-23:
Hi @SeamusShan  and @vivekarvind.  Thanks for testing Gige and 10Gige.  If you solve any problems please consider helping future users by creating a pull request. Even if it is only a note in the readme.

msha3421, 2021-06-08:
> Hello @msha3421 could you solve this issue?

Hi please see my previous comment for my suggestion. Also, I found that so far only Intel chipset based NIC can support modification of RX ring buffer by ethtool. Tried Realtek but no luck. I found this webpage very helpful to me: https://flir.custhelp.com/app/answers/detail/a_id/2817/related/1