ros-drivers/flir_camera_driver #8: Camera crash on trigger timeout

jewnicorn27, 2018-08-16:
The driver crashes when the camera times out on a trigger. If set to external triggering, and no trigger is received, the driver crashes. Hard to provide a more detailed description.

```sh
[ERROR] [1534454603.501352073]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
[ERROR] [1534454603.501684359]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
```

**Launch config:**
`      <param name="enable_trigger" value="On" />
      <param name="trigger_source" value="Line2" />
      <param name="trigger_activation_mode" value="RisingEdge" />
      <param name="trigger_overlap_mode" value="Off" />
`

mhosmar-cpr, 2018-08-20:
Hi @jewnicorn27,
I don't have hardware to test with right now. Would your camera work with flycapture?  That feature came from pointgrey_camera_driver which this driver is based on.  Could you maybe test with that? It would tell us if its a problem with the new driver.

jewnicorn27, 2018-08-20:
@mhosmar-cpr thanks for getting back to me, I am using a BFS camera, which I believe isn't supported by flycapture.

raabuchanan, 2018-11-14:
@ben2789 I'm also trying to externally trigger the camera and I applied your fix but it still doesn't respond to an external signal. Could you share your complete launch file and what electrical signal you use for triggering? I've tried 3.3v 25Hz 50% duty cycle from a signal generator.

ben2789, 2018-11-14:
@raabuchanan I believe the launch file I use is a simple modification of  flir_camera_driver/spinnaker_camera_driver/launch/camera.launch

```
<launch>
  <arg name="mono_name" value="Blackfly S BFS-U3-16S2M" />
  <arg name="mono_camera_serial" value="17513930" />

  <arg name="calibrated" default="0" />

  <!-- Mono Camera -->
  <group ns="BFS_MONO">
    <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" cwd="node" output="screen"/>

    <node pkg="nodelet" type="nodelet" name="spinnaker_camera_nodelet"
          args="load spinnaker_camera_driver/SpinnakerCameraNodelet camera_nodelet_manager" >

      <!-- Triggering -->
      <param name="frame_id" value="camera" />
      <param name="serial" value="$(arg mono_camera_serial)" />
      <param name="enable_trigger" value="On" />
      <param name="trigger_source" value="Line2" />
      <param name="trigger_activation_mode" value="RisingEdge" />
      <param name="trigger_overlap_mode" value="Off" />

      <!-- Exposure -->
      <param name="exposure_auto" value="Off" />
      <param name="exposure_time" value="40000" />
      <param name="auto_exposure_time_upper_limit" value = "300000" />
      <param name="auto_gain" value="Off" />

      <!-- Use the camera_calibration package to create this file -->
      <param name="camera_info_url" if="$(arg calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
    </node>

    <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer camera_nodelet_manager">
    </node>
  </group>
</launch>
```

raabuchanan, 2018-11-16:
Thanks @ben2789 I realized my issue was using Line0 which is opto isolated and requires a pull up resistor. Using line3 worked for me.

ben2789, 2018-11-16:
@raabuchanan I found it easier to work with the BFS camera by modifying the example code rather than editing this driver. If the functionality you're after is fairly simple e.g. no dynamic reconfigure. Adding basic ROS functionality to that may be easier.

Kaju-Bubanja, 2019-08-23:
Why was this closed? The issue still persists at least for me and I see no fix.

Zebrafish007, 2020-02-27:
... GetNextImage(timeout_10) and GetNextImage(timeout_)is raising an error (-1073740791) under PySpin. This also happens with GetNextImage(10), etc....

ParthAD3, 2021-06-01:
Hi @ben2789 
Could you solve this issue?
Image received from camera 20513343 is incomplete.
[ERROR] [1622558664.970472153]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]

ParthAD3, 2021-06-01:
I can launch the camera normaly, but after some time this error appers...