ros-drivers/flir_camera_driver #10: Add Gige Support to Driver

mhosmar-cpr, 2018-08-17:
Currently Gige cameras are not supported.  Discuss here about implementing Gige.

aurochs1627, 2018-09-14:
Following up on this issue. Is there plan to add Gige support to this driver anytime soon?

mhosmar-cpr, 2018-11-28:
Hi everyone,

Unfortunately I am not able to work on this driver.
If you take a look at issue #7 other people have been able to run the driver with GigE cameras. You may want to try it and see if it is already working for you.  What's missing is the ability to change the GigE specific settings. I believe the driver currently just uses defaults.  This would be implemented here [camera.h#L95](https://github.com/ros-drivers/flir_camera_driver/blob/94b5aa0e90fbe6089fd938e0efd439bde83b5078/spinnaker_camera_driver/include/spinnaker_camera_driver/camera.h#L95) and [SpinnakerCamera.cpp#L208](https://github.com/ros-drivers/flir_camera_driver/blob/94b5aa0e90fbe6089fd938e0efd439bde83b5078/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L208)

If the driver connects but doesn't show an image it might be because the model you are using hasn't been added yet.  Different models can have different parameters that need to be set correctly. (See: Cm3.cpp vs Camera.cpp)

Sorry I couldn't help more,

Mike