ros-drivers/flir_camera_driver #9: Failed to load nodelet

eduardoiscar, 2018-08-17:
Hi, 
I have downloaded and compiled the package on Ubuntu 16.04 with Kinetic. After installing, I ran the spin-conf script to set udev permissions and rebooted.

Whenever I run the camera.launch file, the camera is detected but I get the following error about the nodelet not being loaded. It happens both with a BFS-U3-51S5C and a BFS-U3-51S5M. Any ideas? 

 

>  SUMMARY
>  ========
> PARAMETERS
>  * /camera/spinnaker_camera_nodelet/frame_id: camera
>  * /camera/spinnaker_camera_nodelet/serial: 0
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.13
> 
> NODES
>   /camera/
>     camera_nodelet_manager (nodelet/nodelet)
>     spinnaker_camera_nodelet (nodelet/nodelet)
> 
> auto-starting new master
> process[master]: started with pid [9679]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to fd4411e2-a265-11e8-88c1-9cb6d0f4ddf5
> process[rosout-1]: started with pid [9692]
> started core service [/rosout]
> process[camera/camera_nodelet_manager-2]: started with pid [9699]
> process[camera/spinnaker_camera_nodelet-3]: started with pid [9708]
> [ INFO] [1534541956.363345983]: Initializing nodelet with 4 worker threads.
> [ INFO] [1534541957.856720197]: [SpinnakerCamera]: Number of cameras detected: 1
> [ INFO] [1534541957.979883873]: [SpinnakerCamera::connect]: Using Serial: 18284882
> [ INFO] [1534541957.979983317]: [SpinnakerCamera::connect]: Detected device type: U3V
> [ INFO] [1534541958.145472911]: [SpinnakerCamera::connect]: Camera model name: Blackfly S BFS-U3-51S5M
> [ INFO] [1534541958.148914877]: [SpinnakerCamera]: (18284882) DeviceLinkThroughputLimit set to 500000000.
> [FATAL] [1534541958.259820636]: Failed to load nodelet '/camera/spinnaker_camera_nodelet` of type `spinnaker_camera_driver/SpinnakerCameraNodelet` to manager `camera_nodelet_manager'
> [camera/camera_nodelet_manager-2] process has died [pid 9699, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/eiscar/.ros/log/fd4411e2-a265-11e8-88c1-9cb6d0f4ddf5/camera-camera_nodelet_manager-2.log].
> log file: /home/eiscar/.ros/log/fd4411e2-a265-11e8-88c1-9cb6d0f4ddf5/camera-camera_nodelet_manager-2*.log
> [camera/spinnaker_camera_nodelet-3] process has died [pid 9708, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load spinnaker_camera_driver/SpinnakerCameraNodelet camera_nodelet_manager __name:=spinnaker_camera_nodelet __log:=/home/eiscar/.ros/log/fd4411e2-a265-11e8-88c1-9cb6d0f4ddf5/camera-spinnaker_camera_nodelet-3.log].
> log file: /home/eiscar/.ros/log/fd4411e2-a265-11e8-88c1-9cb6d0f4ddf5/camera-spinnaker_camera_nodelet-3*.log
>

mhosmar-cpr, 2018-08-20:
Hi @eduardoiscar,

Are you able to get video with spinview? You can install it from the sdk deb. The spinview deb actually would be in your build folder if you want to install it from there.

eduardoiscar, 2018-08-24:
Spinview starts and detects the camera, but as soon as I click to start streaming it crashes with a invalid double linked list pointer message. I will try and contact Flir support to troubleshoot this issue.

closed 2019-06-12 by mhosmar-cpr