ros-drivers/flir_camera_driver #12: USBFS Too Small by Default

mhosmar-cpr, 2018-10-23:
Increasing the USB buffer is outlined here: [Using Linux with USB 3.1]
(https://www.ptgrey.com/tan/10685#ConfiguringUSBFS) But isn't mentioned in the readme.

Plan:
1. Add note to readme.
2. Have deb automatically increase USBFS size.

mikepurvis, 2018-10-24:
My opinion has long been that this should be documented, and possibly checked at runtime by the driver with a warning dumped to the console, but otherwise that it's inappropriate for an individual driver like this to be manipulating a system-wide configuration such as the usb kernel setup.

mhosmar-cpr, 2018-10-24:
@mikepurvis thanks for the input.  The new goal will be to include a runtime check.  The driver currently dies without error and libusb segfaults when trying to allocate more than the default 32MB buffer.

bk-mtg, 2019-11-21:
In case it helps anyone else in the future, I had my USBFS set too small for the number of frames the system was trying to capture, and the behavior I saw was

    Reconfigure Callback failed with error: [SpinnakerCamera::start] Failed to start capture with error: Spinnaker: Could not start acquisition. Please try reconnecting the device. [-1010]

I was able to fix this by increasing the USBFS memory, or by decreasing the required buffer size (though I think that's not supported by the current ROS driver?).

mikepurvis, 2019-11-22:
@bk-mtg There's a note about this on the legacy pointgrey_camera_driver wiki page, see:

http://wiki.ros.org/pointgrey_camera_driver#Troubleshooting

If relevant, please copy this to the page for the spinnaker fork.

bk-mtg, 2019-11-22:
@mikepurvis that's the relevant note, yup.  The main thing I wanted to add that I didn't see anywhere online was a pointer that this is the action to take to resolve an error -1010 from the Spinnaker SDK when trying to start image acquisition.

I don't currently have wiki access, but working to obtain that.