ros-drivers/flir_camera_driver #14: /image_color remains gray scale.

wep21, 2018-12-13:
I display /image_color with blackflyS(GigE), but it remains gray scale.
I also check display image with spinview, it was rgb scale.
Do you have any thoought about this?
image_proc/debayer cannot convert bayer into rgb?

raabuchanan, 2019-01-31:
Have you made sure to change ```image_format_color_coding``` to something other than mono8? Take a look at camera.launch in my fork https://github.com/raabuchanan/flir_camera_driver

wmmc88, 2019-05-28:
I'm also getting a greyscale image from all the topics. is a camera calibration required to get the color mapping?

heethesh, 2019-10-26:
I get the warning `[SpinnakerCamera]: (12345678) Entry name BGR8 not available.` Same with `BayerRG8`

vSensorRT, 2019-11-08:
@heethesh where did you change image_format_color_coding? in Spinnaker.cfg

heethesh, 2019-11-08:
@vSensorRT I did as @raabuchanan suggested [here](https://github.com/raabuchanan/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/launch/camera.launch#L75). I'm working with the Chameleon3 model.

tiexuedanxin, 2020-11-13:
> Have you made sure to change `image_format_color_coding` to something other than mono8? Take a look at camera.launch in my fork https://github.com/raabuchanan/flir_camera_driver

hello, how do you know the parameter has image_format_color_coding、auto_white_balance and the like.Do you know where the code to load these parameters.