ros-drivers/flir_camera_driver #15: "Something still holds a reference to a camera" error with multiple Blackfly S cameras

ebretl, 2019-03-06:
Setup:
- Ubuntu 16.04, ROS Kinetic
- Two FLIR Blackfly S cameras
- Spinnaker 1.13.0.31

Problem Description:
- On first launch, ROS driver works fine with one or two cameras
- After terminating the launch file with `Ctrl+C`, the first nodelet is able to unload, but the second one is not. The error encountered is:
```
[camera_nodelet_manager-1] killing on exit
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Can't clear a camera because something still holds a reference to the camera [-1004]
```
- On subsequent launches, the nodelet manager will crash and the error comes up again:
```
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Can't clear a camera because something still holds a reference to the camera [-1004]
================================================================================REQUIRED process [camera_nodelet_manager-1] has died!
```
- When launching only one camera, the error does not happen

keenan-burnett, 2019-03-07:
I've also seen this error come up in my current set up which is identical to ebretl.

@ebretl Would you mind sharing your launch file setup for running multiple blackfly S cameras at once? I've been struggling to get this working without throwing an error.

ebretl, 2019-03-08:
The launch file is here: https://github.com/AutoRally/autorally/blob/evan-flir-kinetic/autorally_core/launch/camera_flir.launch. This launch file is included for each camera (in cameras.launch), and both camera nodelets are loaded into the same manager. 

Also, try checking out my pull request and see if that fixes your issue. If not, we should work through it.

keenan-burnett, 2019-03-08:
Thanks for this. I was able to resolve my current issue with using multiple cameras after all. Although the issue I was seeing may be specific to my system. I have observed that the spinnaker camera driver will not allow two cameras to be running on the same PCIe card. I'll see if this issue crops up for me again, but it doesn't happen consistently.

keenan-burnett, 2019-03-18:
I haven't seen this error since I last commented.
I was able to resolve my own issues by setting usbcore.usbfs_memory_mb=10000 using the following instructions:
https://www.ptgrey.com/tan/10685

I incorrectly assumed that 1000 MB would be enough for two of my cameras.

mhosmar-cpr, 2019-06-12:
Thanks marking as duplicate of #12

PR #16 (2021-02-11): Fixes SpinnakerCamera teardown
Fixes #15. Please review the issue description before the following.

Running the nodelet manager in valgrind shows that the crash is happening in the destructor of SpinnakerCamera's system_ object, which is calling ReleaseInstance for a second time. Removing the first call prevents the crash from occuring and the "something still has a reference to the camera" error from appearing in the console. 

Through trial and error, removing camList_.Clear() was found to prevent a further issue where all nodelets under the nodelet manager will break their bonds and exit a [... cut]
closed 2019-06-12 by mhosmar-cpr