ros-drivers/flir_camera_driver #18: Auto-download no longer works

mincrmatt12, 2019-05-02:
Pointgrey has merged their site with FLIR's, and as a result the `ptgrey.com` links no longer download properly, and instead all redirect to `flir.com`

davidbsp, 2019-05-07:
Facing the same issue here. 
As a consequence, the compilation on melodic fails with error:

`
/home/user/catkin_ws/src/flir_camera_driver/spinnaker_camera_driver/include/spinnaker_camera_driver/set_property.h:29:10: fatal error: Spinnaker.h: No such file or directory
 #include "Spinnaker.h"
          ^~~~~~~~~~~~~
compilation terminated.
`

davidbsp, 2019-05-07:
Hi! i was able to download the file from another source. 
You can find an updated "download_spinnaker" file at the end of this comment.

I have a new issue: I am trying to get this to work in Ubuntu 18.04 and ROS Melodic.

**a) Using the spinnaker-1.13.0.31 lib version, I get the following linking issues when compiling the pkg:**

```
/usr/bin/ld: warning: libswscale-ffmpeg.so.3, needed by //home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libavcodec-ffmpeg.so.56, needed by //home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libavutil-ffmpeg.so.54, needed by //home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libavformat-ffmpeg.so.56, needed by //home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1, not found (try using -rpath or -rpath-link)
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avformat_alloc_output_context2@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_write_trailer@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avpicture_alloc@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_open2@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avformat_new_stream@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_flush_buffers@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_encode_video2@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_register_all@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_free@LIBAVUTIL_FFMPEG_54'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_frame_alloc@LIBAVUTIL_FFMPEG_54'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_rescale_q@LIBAVUTIL_FFMPEG_54'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avformat_write_header@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avformat_free_context@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `sws_getContext@LIBSWSCALE_FFMPEG_3'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avio_open@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_interleaved_write_frame@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_find_encoder@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_dump_format@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_init_packet@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avio_close@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `av_register_all@LIBAVFORMAT_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `avcodec_close@LIBAVCODEC_FFMPEG_56'
//home/david/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libspinvideoencoder.so.1: undefined reference to `sws_scale@LIBSWSCALE_FFMPEG_3'
collect2: error: ld returned 1 exit status
flir_camera_driver/spinnaker_camera_driver/CMakeFiles/spinnaker_test_node.dir/build.make:206: recipe for target '/home/david/catkin_ws/devel/lib/spinnaker_camera_driver/spinnaker_test_node' failed
make[2]: *** [/home/david/catkin_ws/devel/lib/spinnaker_camera_driver/spinnaker_test_node] Error 1
[... cut ...]
    for files in xml_files:
        if files.endswith(".xml"):
            origin_file = os.path.join(xml_path, files)
            print('Copying ' + origin_file + ' to ' + destination_folder)
            shutil.copy(origin_file, destination_folder)

#-- libSpinnaker library: ~/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/lib/libSpinnaker.so
#-- libSpinnaker include: ~/catkin_ws/build/flir_camera_driver/spinnaker_camera_driver/usr/include/spinnaker
```

I would love some feedback on this. Thanks!

mincrmatt12, 2019-05-12:
Are you sure this doesn't violate the license? Unless that site has got express permission to redistribute the library it wouldn't be valid.

mincrmatt12, 2019-05-24:
There appears to be an API for box, so we might be able to download it through that with a dummy box account.

mhosmar-cpr, 2019-06-12:
@mincrmatt12 Have you looked at the box API any further is it a simple implementation?

zlacelle, 2019-06-21:
I don't think we should be downloading the Spinnaker library automatically. Tightly coupling it is causing strange issues with newer versions of the Spinnaker library.

The SDK is available for free, and is easy enough to install. Decoupling this package from the underlying FLIR SDK will save pain down the road, I think.

mhosmar-cpr, 2020-05-04:
Fixed in fec86c4

closed 2020-05-04 by mhosmar-cpr

rdesc, 2020-05-13:
Facing same issue `fatal error: Spinnaker.h: No such file or directory` it doesn't seem like the spinnaker sdk was properly downloaded. Trying to run with ROS melodic and Ubuntu 18.04

mhosmar-cpr, 2020-05-13:
@rdesc Can you provide more of the compilation output.  If its failing to download there should be signs before this stage.