ros-drivers/flir_camera_driver #19: Launchfile parameters out of date

YoshuaNava, 2019-05-16:
HI,
Thank you for writing this driver.

I'm currently using it with a set of BlackFly cameras, and it works really well.

Nonetheless, setting it up at first was a bit confusing due to two parameter names which are not correctly refered in the launchfiles.
* frame_rate, which should be acquisition_frame_rate
https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/launch/camera.launch#L45

* acquisition_frame_rate_enable, which doesn't appear at all in the launchfiles, and enables manual frame rate settings.

I could write a PR in the near future to fix this.

Best,
Yoshua Nava

mikehosmar, 2019-05-17:
@YoshuaNava thanks for finding this.  A PR would be appreciated.

YoshuaNava, 2019-06-24:
@mikehosmar I submitted a PR last month. Could you review it?

Link: https://github.com/ros-drivers/flir_camera_driver/pull/20

Thanks in advance.

closed 2020-05-04 by mhosmar-cpr