ros-drivers/flir_camera_driver #22: Build issues

SK19, 2019-05-28:
Hello,

I'm trying to build the flir_camera_driver package, however have been running into the below error.
I have also tried building the ETH ASL fork of this driver with the same error.

Would anyone have any ideas on what is going wrong/how I might fix it?

I'm running Ubuntu 16.04 on an ARM64 system (Nvidia TX2).

Any help would be greatly appreciated.
Thanks!
```
nvidia@tegra-ubuntu:~/flir2_ws/src$ catkin build
----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/nvidia/flir2_ws
----------------------------------------------------------
Build Space:        [exists] /home/nvidia/flir2_ws/build
Devel Space:        [exists] /home/nvidia/flir2_ws/devel
Install Space:      [unused] /home/nvidia/flir2_ws/install
Log Space:         [missing] /home/nvidia/flir2_ws/logs
Source Space:       [exists] /home/nvidia/flir2_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
----------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                                                                                                                                                            
[build] Updating package table.                                                                                                                                                                                                       
Starting  >>> catkin_tools_prebuild                                                                                                                                                                                                   
Finished  <<< catkin_tools_prebuild                  [ 4.4 seconds ]                                                                                                                                                                  
Starting  >>> spinnaker_camera_driver                                                                                                                                                                                                 
______________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << spinnaker_camera_driver:make /home/nvidia/flir2_ws/logs/spinnaker_camera_driver/build.make.000.log                                                                                                                      
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x30): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator=(long)'
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x38): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator=(decltype(nullptr))'
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x68): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator==(decltype(nullptr)) const'
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x78): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator==(long) const'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/nvidia/flir2_ws/devel/.private/spinnaker_camera_driver/lib/spinnaker_camera_driver/spinnaker_test_node] Error 1
make[1]: *** [CMakeFiles/spinnaker_test_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/nvidia/flir2_ws/build/spinnaker_camera_driver; catkin build --get-env spinnaker_camera_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................................................................................................................................................................................
Failed     << spinnaker_camera_driver:make           [ Exited with code 2 ]                                                                                                                                                           
Failed    <<< spinnaker_camera_driver                [ 33.7 seconds ]                                                                                                                                                                 
[build] Summary: 1 of 2 packages succeeded.                                                                                                                                                                                           
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                                                                                      
[build]   Warnings:  None.                                                                                                                                                                                                            
[build]   Abandoned: None.                                                                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                                               
[build] Runtime: 38.1 seconds total.                                                                                                                                                                                                  
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

d181, 2019-06-03:
Hi SK19, I'm getting the same issue. Had any success working out the problem yet?

SK19, 2019-06-03:
Not yet unfortunately. Getting the same issue on my desktop so it doesn't seem to be an issue specifically related to the TX2 at least. Haven't been able to narrow it down yet. I've double checked I had all of the dependencies fulfilled as per the documentation.

I'm having trouble finding where the reference to Spinnaker::BasePtr is coming from so I'm quite stuck here. I'll let you know if I work it out.

SK19, 2019-06-17:
Update:
Seems to have been a working directory issue. Fixed now.

closed 2019-06-17 by SK19