ros-drivers/flir_camera_driver #24: Build issues : Not able to find Spinnaker.h

Singh-sid930, 2019-06-06:
Hi,

I am trying to build the flir_camera_driver in my repo to use chameleon3 cameras. The following error pops up. 

```
Errors     << spinnaker_camera_driver:make /home/server/catkin_ws/logs/spinnaker_camera_driver/build.make.000.log                                                                                          
In file included from /home/server/catkin_ws/src/camera/flir_camera_driver/spinnaker_camera_driver/include/spinnaker_camera_driver/camera.h:32:0,
                 from /home/server/catkin_ws/src/camera/flir_camera_driver/spinnaker_camera_driver/src/camera.cpp:25:
/home/server/catkin_ws/src/camera/flir_camera_driver/spinnaker_camera_driver/include/spinnaker_camera_driver/set_property.h:29:10: fatal error: Spinnaker.h: No such file or directory
 #include "Spinnaker.h"
```
`` 

Also, Will this driver work with multiple cameras if I want to run more than one cameras(with different namespaces?)

SK19, 2019-06-06:
Hi Singh,

I was having other issues with building also.
I think this one may be solved by installing the spinnaker sdk:
https://www.flir.com.au/support-center/iis/machine-vision/downloads/spinnaker-sdk-flycapture-and-firmware-download/ 

Hopefully this helps. I'm curious if you get the other build issue which I got after this one (issue below yours).

Singh-sid930, 2019-06-06:
Hi SK19! 
Thanks for that, I was able to build the package however there were a bunch of warnings. But the package build was completed. I am gonna go ahead and try my stuff now. 
What exactly were the errors you were getting ?

Singh-sid930, 2019-06-06:
Fixed after installing spinnakker

closed 2019-06-06 by Singh-sid930

SK19, 2019-06-07:
Hi Singh, glad I could help.

This is what I get when I try to build.

```
nvidia@tegra-ubuntu:~/flir_ws/src/flir_camera_driver$ catkin build
--------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/nvidia/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/nvidia/flir_ws
--------------------------------------------------------------------------
Build Space:        [exists] /home/nvidia/flir_ws/build
Devel Space:        [exists] /home/nvidia/flir_ws/devel
Install Space:      [unused] /home/nvidia/flir_ws/install
Log Space:         [missing] /home/nvidia/flir_ws/logs
Source Space:       [exists] /home/nvidia/flir_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                                                                                                                                                            
[build] Updating package table.                                                                                                                                                                                                       
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                                                                                                                                                   
Finished  <<< catkin_tools_prebuild                  [ 5.0 seconds ]                                                                                                                                                                  
Starting  >>> spinnaker_camera_driver                                                                                                                                                                                                 
______________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << spinnaker_camera_driver:make /home/nvidia/flir_ws/logs/spinnaker_camera_driver/build.make.000.log                                                                                                                       
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x30): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator=(long)'
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x38): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator=(decltype(nullptr))'
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x68): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator==(decltype(nullptr)) const'
CMakeFiles/spinnaker_test_node.dir/src/spinnaker_test_node.cpp.o:(.rodata._ZTVN9Spinnaker9CameraPtrE[_ZTVN9Spinnaker9CameraPtrE]+0x78): undefined reference to `Spinnaker::BasePtr<Spinnaker::Camera, Spinnaker::ICameraBase>::operator==(long) const'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/nvidia/flir_ws/devel/.private/spinnaker_camera_driver/lib/spinnaker_camera_driver/spinnaker_test_node] Error 1
make[1]: *** [CMakeFiles/spinnaker_test_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/nvidia/flir_ws/build/spinnaker_camera_driver; catkin build --get-env spinnaker_camera_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................................................................................................................................................................................
Failed     << spinnaker_camera_driver:make           [ Exited with code 2 ]                                                                                                                                                           
Failed    <<< spinnaker_camera_driver                [ 38.3 seconds ]                                                                                                                                                                 
[... cut ...]
[build]   Abandoned: None.                                                                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                                               
[build] Runtime: 43.3 seconds total.                           
```

Any ideas?

tiexuedanxin, 2020-11-10:
> Hi Singh,
>
> I was having other issues with building also.
> I think this one may be solved by installing the spinnaker sdk: [...]

hello, could you tell me how to install the spinnaker sdk, thanks very much

CrazyPopLin, 2020-11-14:
@tiexuedanxin , you can download the spinnaker SDK and install it. Select right version, and install instruction is in readme. The lib and header should be in /opt/spinnaker.

tkazik, 2021-01-08:
Hi all, I am getting the same error while trying to build (`fatal error: Spinnaker.h: No such file or directory`). 

- Spinnaker is installed and I am able to get an image/video through `spinview`.
- The lib and header are available in `/opt/spinnaker`

Yet, while building it cannot find the header. Any other thing I am missing?
PS: Sorry for hijacking this issue.

lchop, 2021-01-11:
You should install the previous version of spinnaker (2.0.0), it works for me.

eleathers, 2021-02-03:
> Hi all, I am getting the same error while trying to build (`fatal error: Spinnaker.h: No such file or directory`).
>
> * Spinnaker is installed and I am able to get an image/video through `spinview`.
> * [...]

I am having the same issue. Were you ever able to resolve it?

Karthikeyanc2, 2021-06-15:
I am also facing the same issue,

**with 2.4.0,** 

fatal error: Spinnaker.h: No such file or directory

Spinnaker is installed and I am able to get an image/video through spinview.
The lib and header are available in /opt/spinnaker

**with 2.0.0**

this error which is still open,

https://github.com/FLIR/flir_adk_ethernet/issues/6

Please guide me what to do,

Thanks,
Karthik