ros-drivers/flir_camera_driver #25: Not able to set BinningHorizontal for Chamaleon chameras

Singh-sid930, 2019-06-06:
Hi, 

I tried to launch Acquisition but I am running into an obscure error that the cameras do not support color. 

```
[FATAL] [1559835607.362157355]: Unable to set BinningHorizontal to 2 (ptr retrieval). Aborting...
[FATAL] [1559835607.362460003]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 77) [-2006]
[ WARN] [1559835607.362504777]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]

```

Also, I am trying to use two cameras and this is the modified configuration file : 
`cam_ids:
- 15508253
- 16466237
cam_aliases:
- cam0
- cam1
master_cam: 16466237

# Assign all the follwing via launch file to prevent confusion and conflict

save_path: ~/catkin_ws
save_type: .bmp #binary or .tiff or .bmp
binning: 1 # going from 2 to 1 requires cameras to be unplugged and replugged
color: true
#frames: 50
soft_framerate: 4 # this frame rate reflects to the software frame rate set using ros::rate
exp: 997
#to_ros: true  #When to_ros is not selected, but live is selected, pressing 'space' exports single image to ROS`

The cameras are these : 
https://www.flir.com/products/chameleon3-usb3/?model=CM3-U3-13Y3C-CS

Which I am pretty sure are color cameras

oysteinvolden, 2019-06-26:
Did you figure it out?

I'm struggling with the same problem. I use Blackfly BFLY-PGE-23S6C. I don't think enabling color mode is the problem for me either, my camera supports color (have checked it using spinview).

Singh-sid930, 2019-06-27:
Hi, No I was not. 
I just switched to a different driver. 
https://github.com/daniilidis-group/cam_sync
Try this. I work in this group and if you have any issues hit me up .

closed 2019-06-27 by Singh-sid930

addext, 2019-08-27:
hi,
I tried to use cam_sync because I had the same problem with spinnaker_sdk.

with cam_sync, I am getting an error:
`ERROR: cannot launch node of type [cam_sync/cam_sync_node]: can't locate node [cam_sync_node] in package [cam_sync]`

when trying `rosrun cam_sync cam_sync_node`:

`[rosrun] Couldn't find executable named cam_sync_node below /home/dflap3/cam_sync`

I'm fairly new to ROS as this is my first project - do you have any advice regarding this problem?

Thanks & regards