ros-drivers/flir_camera_driver #33: Problem with roslaunching the acquisition.launch file

alankwok97, 2019-07-18:
Hi there,

I am currently working on a new project using ROS to get data from a FLIR Blackfly camera(I am still quite new to this language). I followed the instructions written on the ROS wiki web page (including changing the serial number) but still couldn't launch the acquisition.launch file. Here are the FATAL errors that I have:

[FATAL] [1563449758.815448351]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1563449758.815547185]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1563449758.934733561]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Camera is not started. [-1002]
[vision_nodelet_manager-2] process has died [pid 31124, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/vision_nodelet_manager-2.log].
log file: /home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 31125, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/acquisition_node-3.log].
log file: /home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/acquisition_node-3*.log

And here is the text in test_params.yaml:
cam_ids:
- 19138453
cam_aliases:
- cam0
master_cam: 19138453
skip: 20
delay: 1.0

# Assign all the follwing via launch file to prevent confusion and conflict

#save_path: ~/projects/data
#save_type: .bmp #binary or .tiff or .bmp
#binning: 1 # going from 2 to 1 requires cameras to be unplugged and replugged
#color: false
#frames: 50
#soft_framerate: 20 # this frame rate reflects to the software frame rate set using ros::rate
#exp: 997
#to_ros: true  #When to_ros is not selected, but live is selected, pressing 'space' exports single image to ROS

#Camera info message details
distortion_model: plumb_bob
image_height: 1536
image_width: 2048
distortion_coeffs:
- [-0.021141875266089592, -0.3733872931278025, 2.385982550579459, 3.2824571732099725]

#specified as [fx  0 cx 0 fy cy 0  0  1]
intrinsic_coeffs:
- [1886.9232141485886, 0.0, 604.7214878709341, 0.0, 1886.6668765711668, 493.47726714719823, 0.0, 0.0, 1.0]

rectification_coeffs:
- [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]

projection_coeffs:
- [913.700317, 0.000000, 953.448302, 0.000000, 0.000000, 1063.296631, 777.871993, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

Currently I am using ros melodic (Ubuntu 18.04 Bionic) and I know that that package is written for kinetic but are there ways to fix these problems and get it working? It would mean so much to me. 

Many thanks.

closed 2019-07-18 by alankwok97

PascPeli, 2019-09-11:
Hello alankwok97,
Did you resolve this issue? if yes how?

AdriannaGmz, 2021-06-22:
2021 and I also have the same problem