ros-drivers/flir_camera_driver #35: Image incomplete with image status Image data is incomplete

JetGang, 2019-10-25:
@mhosmar-cpr Hello,

I use three BFS-U3-28S5C-C cameras and try to trigger them simultaneously, but it crashed sometimes(not every time) when I launch the driver and try to display the images. I also tried to adjust the official `Trigger.cpp` and make it work with multi-cameras, and I don't meet any similar error so far.

> PARAMETERS
>  * /center/camera/spinnaker_camera_nodelet/acquisition_frame_rate: 30
>  * /center/camera/spinnaker_camera_nodelet/auto_exposure_time_upper_limit: 15000
>  * /center/camera/spinnaker_camera_nodelet/auto_gain: Continuous
>  * /center/camera/spinnaker_camera_nodelet/auto_white_balance: Continuous
>  * /center/camera/spinnaker_camera_nodelet/enable_trigger: On
>  * /center/camera/spinnaker_camera_nodelet/exposure_auto: Continuous
>  * /center/camera/spinnaker_camera_nodelet/frame_id: blackflys_FC
>  * /center/camera/spinnaker_camera_nodelet/gamma: 0.8
>  * /center/camera/spinnaker_camera_nodelet/image_format_color_coding: BayerRG8
>  * /center/camera/spinnaker_camera_nodelet/line_mode: Output
>  * /center/camera/spinnaker_camera_nodelet/line_selector: Line2
>  * /center/camera/spinnaker_camera_nodelet/line_source: ExposureActive
>  * /center/camera/spinnaker_camera_nodelet/serial: 19075864
>  * /center/camera/spinnaker_camera_nodelet/trigger_activation_mode: RisingEdge
>  * /center/camera/spinnaker_camera_nodelet/trigger_overlap_mode: Off
>  * /center/camera/spinnaker_camera_nodelet/trigger_selector: FrameStart
>  * /center/camera/spinnaker_camera_nodelet/trigger_source: Line3
>  * /left/camera/spinnaker_camera_nodelet/acquisition_frame_rate: 30
>  * /left/camera/spinnaker_camera_nodelet/auto_exposure_time_upper_limit: 15000
>  * /left/camera/spinnaker_camera_nodelet/auto_gain: Continuous
>  * /left/camera/spinnaker_camera_nodelet/auto_white_balance: Continuous
>  * /left/camera/spinnaker_camera_nodelet/enable_trigger: On
>  * /left/camera/spinnaker_camera_nodelet/exposure_auto: Continuous
>  * /left/camera/spinnaker_camera_nodelet/frame_id: blackflys_FL
>  * /left/camera/spinnaker_camera_nodelet/gamma: 0.8
>  * /left/camera/spinnaker_camera_nodelet/image_format_color_coding: BayerRG8
>  * /left/camera/spinnaker_camera_nodelet/line_mode: Output
>  * /left/camera/spinnaker_camera_nodelet/line_selector: Line2
>  * /left/camera/spinnaker_camera_nodelet/line_source: ExposureActive
>  * /left/camera/spinnaker_camera_nodelet/serial: 19075863
>  * /left/camera/spinnaker_camera_nodelet/trigger_activation_mode: RisingEdge
>  * /left/camera/spinnaker_camera_nodelet/trigger_overlap_mode: Off
>  * /left/camera/spinnaker_camera_nodelet/trigger_selector: FrameStart
>  * /left/camera/spinnaker_camera_nodelet/trigger_source: Line3
>  * /right/camera/spinnaker_camera_nodelet/acquisition_frame_rate: 30
>  * /right/camera/spinnaker_camera_nodelet/auto_exposure_time_upper_limit: 15000
>  * /right/camera/spinnaker_camera_nodelet/auto_gain: Continuous
>  * /right/camera/spinnaker_camera_nodelet/auto_white_balance: Continuous
>  * /right/camera/spinnaker_camera_nodelet/enable_trigger: On
>  * /right/camera/spinnaker_camera_nodelet/exposure_auto: Continuous
>  * /right/camera/spinnaker_camera_nodelet/frame_id: blackflys_FC
>  * /right/camera/spinnaker_camera_nodelet/gamma: 0.8
>  * /right/camera/spinnaker_camera_nodelet/image_format_color_coding: BayerRG8
>  * /right/camera/spinnaker_camera_nodelet/line_mode: Output
>  * /right/camera/spinnaker_camera_nodelet/line_selector: Line2
>  * /right/camera/spinnaker_camera_nodelet/line_source: ExposureActive
>  * /right/camera/spinnaker_camera_nodelet/serial: 19075865
>  * /right/camera/spinnaker_camera_nodelet/trigger_activation_mode: RisingEdge
>  * /right/camera/spinnaker_camera_nodelet/trigger_overlap_mode: Off
>  * /right/camera/spinnaker_camera_nodelet/trigger_selector: FrameStart
>  * /right/camera/spinnaker_camera_nodelet/trigger_source: Line3
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
>   /right/camera/
>     camera_nodelet_manager (nodelet/nodelet)
>     image_proc_debayer (nodelet/nodelet)
>     spinnaker_camera_nodelet (nodelet/nodelet)
>   /center/camera/
>     camera_nodelet_manager (nodelet/nodelet)
>     image_proc_debayer (nodelet/nodelet)
>     spinnaker_camera_nodelet (nodelet/nodelet)
>   /left/camera/
>     camera_nodelet_manager (nodelet/nodelet)
[... 228 lines cut ...]
> [ERROR] [1572026509.341576391]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
> ^C[center/camera/image_proc_debayer-10] killing on exit
> [center/camera/spinnaker_camera_nodelet-9] killing on exit
> [center/camera/camera_nodelet_manager-8] killing on exit
> [left/camera/image_proc_debayer-4] killing on exit
> [right/camera/spinnaker_camera_nodelet-6] killing on exit
> [right/camera/camera_nodelet_manager-5] killing on exit
> [right/camera/image_proc_debayer-7] killing on exit
> [left/camera/camera_nodelet_manager-2] killing on exit
> [left/camera/spinnaker_camera_nodelet-3] killing on exit
> terminate called after throwing an instance of 'Spinnaker::Exception'
>   what():  Spinnaker: Can't clear a camera because something still holds a reference to the camera [-1004]
> terminate called after throwing an instance of 'Spinnaker::Exception'
>   what():  Spinnaker: Can't clear a camera because something still holds a reference to the camera [-1004]
> ^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^Cterminate called after throwing an instance of 'Spinnaker::Exception'
>   what():  Spinnaker: Can't clear a camera because something still holds a reference to the camera [-1004]
> ^C^C^C^C^C[rosout-1] killing on exit
> ^C^C^C^C^C^C^C[master] killing on exit
> ^C^C^C^C^C^C^Cshutting down processing monitor...
> ... shutting down processing monitor complete
> done

By the way, it also causes my desktop's crash after this error shows up.

Would you please help me to check the error? Thanks for helping!

Cristian-wp, 2021-02-09:
Hi,
I have your exact problem.
I launch the driver during boot in order to start recording a video .mp4 and a rosbag immediately, but after 1/2 minutes I have this error:

spinnaker-start[843]: [ INFO] [[ERROR] [1612878222.719928123]: [SpinnakerCamera::grabImage] Image received from camera ID is incomplete.

After this error the recording stops, but the rosbag is not affected.

Did you found any solution?

andreucm, 2021-03-30:
Hi all

### Description

Same issue here with: 
```
...
[ INFO] [/camera/camera_nodelet_manager]: [SpinnakerCamera]: (21023432) AutoAlgorithmSelector set to Ae.
[ INFO] [/camera/camera_nodelet_manager]: [SpinnakerCamera]: (21023432) AasRoiEnable set to 1.
[ INFO] [/camera/camera_nodelet_manager]: [SpinnakerCamera]: (21023432) AutoExposureLightingMode set to Normal.
[ INFO] [/camera/camera_nodelet_manager]: using default calibration URL
[ INFO] [/camera/camera_nodelet_manager]: camera calibration URL: file:///home/andreu/.ros/camera_info/0.yaml
[ INFO] [/camera/camera_nodelet_manager]: Unable to open camera calibration file [/home/andreu/.ros/camera_info/0.yaml]
[ WARN] [/camera/camera_nodelet_manager]: Camera calibration file /home/andreu/.ros/camera_info/0.yaml not found.
[ERROR] [/camera/camera_nodelet_manager]: [SpinnakerCamera::grabImage] Image received from camera 2 is incomplete.
[ERROR] [/camera/camera_nodelet_manager]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
```
The error happens when some other node subscribes to some published topic, for instance when I run: 
`rosrun image_view image_view image:=/camera/image_raw`
but not necessarily just after calling image_view. It might happen after some seconds

### OS/SW/HW versions

- SW versions: Ubuntu18.04 + ROSmelodic + Spinnaker-2.3.0.77
- branch/commit: kinetic-devel / faf7cd658821c9d2fa3fbfea20b1e1c33530fe5d
- Blackfly S BFS-PGE-200S6C , GigE, 20MP

### Hypothesis
After doing on the code folder: 
`grep -R "Image received from camera"`

It seems that the exception is thrown here: 
https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L339

So may be related with an issue on setting a too short timeout, specially for high-res cameras (i.e 20MP): 
https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L458

Timeout is set from here: ( grep -R "setTimeout" ):
https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/src/nodelet.cpp#L522

### Actions
I tried to drastically increase the timeout but the error still happens. 

Any suggestion will be welcome. Thanks a lot.

Andreu

andreucm, 2021-03-31:
As a "non-serious" workaround, you can replace the exception thrown [here](https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L339) by a simple ROS_WARN message, and the driver execution carries on ...

imshrish64, 2023-01-12:
For 3 Flir GigE Camera initialisation simultaneously. Please set Acquisition Frame Rate less than 17.
My Code works I have used 10 meter Cat6 cable for each and set Acquisition Frame Rate less than 12 and It worked for me.
cam.AcquisitionFrameRateAuto = "Off"
cam.AcquisitionFrameRateEnable.SetValue(True)
cam.AcquisitionFrameRate.SetValue(12)
cam.AcquisitionFrameRate_set = 12  
cam.BeginAcquisition()

jiheddachraoui, 2024-06-07:
anyone have solved the problem?

imshrish64, 2024-07-09:
Also Try to fix the Bandwidth for multiple camera