ros-drivers/flir_camera_driver #36: Firefly S driver support

jacksparkman, 2019-10-31:
Has this driver been tested with the Firefly S model FFY-U3-16S2M: 1.6 MP, 60 FPS, SONY IMX296, MONO

Ador2, 2021-08-05:
yes, it works great

snakehaihai, 2023-06-26:
will not work directly. you can install from sudo apt install. 
My experience is that there will be error marked in red say cant CONNECT or some similar error. 

The original package of flir_camera_driver was build with spinnaker 1.27

[reference](https://flir.custhelp.com/app/answers/detail/a_id/3656/~/flir-ir-cameras-mounted-on-robots.-compatibility-with-ros-%28robot-operating)

The default one that came with apt install is not. I tested 1.27 and 3.1 version and it works with ubuntu 20.04. On 3.1 the driver is more stable, but have initial error such as 

`[ERROR] [1687759854.204442536]: [SpinnakerCamera]: (20129701) Feature name DecimationHorizontal not implemented.
[ERROR] [1687759854.204454731]: [SpinnakerCamera]: (20129701) Feature name DecimationVertical not implemented.
`