ros-drivers/flir_camera_driver #37: Timestamp set incorrectly?

themightyoarfish, 2019-11-05:
These lines don't make sense to me. Since GetTimeStamp() is just the nanoseconds, but the ROS timestamp wants the `nsec` field to be the nanoseconds since `sec`, more arithmetic needs to be done here, no?

https://github.com/ros-drivers/flir_camera_driver/blob/8c2c23824916e52d3ff7812f951daa23ef475bb9/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L344

mhosmar-cpr, 2020-05-04:
The nanoseconds will be truncated when assigned to the uint32_t `sec`  The line below sets the `nsec`.

closed 2020-05-04 by mhosmar-cpr

IMBAfarmer09, 2020-12-21:
The `nsec` is truncated incorrectly, with values ranging form 0 to 4,294,967,295, instead of 0 to 999,999,999.

https://github.com/ros-drivers/flir_camera_driver/blob/8c2c23824916e52d3ff7812f951daa23ef475bb9/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L345