ros-drivers/flir_camera_driver #38: spinnaker camera driver for ROS Melodic on Ubuntu 18.04

vtanya05, 2019-11-18:
The driver for spinnaker camera available on github is for ROS Kinetic. The same driver can't run in ROS Melodic because the packages, `image_exposure_msgs` and `wfov_camera_msgs` are not available in Melodic. 

Is there a driver available that's compatible with ROS Melodic Ubuntu 18.04?

davizinho5, 2020-04-02:
Hello, I had this very same problem. 

I solved it by downloading these packages from https://github.com/ros-drivers/pointgrey_camera_driver and compiling them under ROS-Melodic.

mhosmar-cpr, 2020-05-04:
`pointgre_camera_driver` is now released on Melodic and you should not need to compile locally anymore.

closed 2020-05-04 by mhosmar-cpr

zgxsin, 2020-05-05:
> `pointgre_camera_driver` is now released on Melodic and you should not need to compile locally anymore.

But `pointgre_camera_driver` is based on FlyCapture2 SDK. Is it also compatible with Spinnaker SDF

tiexuedanxin, 2020-11-11:
> Hello, I had this very same problem.
> 
> I solved it by downloading these packages from https://github.com/ros-drivers/pointgrey_camera_driver and compiling them under ROS-Melodic.

hello, have you use the flir camera with melodic? I come across the following error when roslaunch spinnaker_camera_driver camera.launch, Could you help me.

 [ERROR] [1605084836.155191601]: Failed to load nodelet [/camera/spinnaker_camera_nodelet] of type [spinnaker_camera_driver/SpinnakerCameraNodelet] even after refreshing the cache: Failed to load library /home/yong/catkin_ws/devel/lib//libSpinnakerCameraNodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libSpinnaker.so.1: cannot open shared object file: No such file or directory)
[ERROR] [1605084836.155212658]: The error before refreshing the cache was: Failed to load library /home/yong/catkin_ws/devel/lib//libSpinnakerCameraNodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libSpinnaker.so.1: cannot open shared object file: No such file or directory)
[FATAL] [1605084836.155344067]: Failed to load nodelet '/camera/spinnaker_camera_nodelet` of type `spinnaker_camera_driver/SpinnakerCameraNodelet` to manager `camera_nodelet_manager'
[camera/spinnaker_camera_nodelet-3] process has died [pid 3173, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load spinnaker_camera_driver/SpinnakerCameraNodelet camera_nodelet_manager __name:=spinnaker_camera_nodelet __log:=/home/yong/.ros/log/6e0b3874-23fb-11eb-97ec-a4bb6dc92670/camera-spinnaker_camera_nodelet-3.log].
log file: /home/yong/.ros/log/6e0b3874-23fb-11eb-97ec-a4bb6dc92670/camera-spinnaker_camera_nodelet-3*.log

davizinho5, 2020-11-11:
Hello, 

I used it the FLIR camera with melodic 6 to 8 months ago. 
I don't have a camera with me anymore, so I cannot make any test now. 

From the message you are getting, I'd bet the error comes from the compilation configuration of your nodes, but I'm not completely sure.

Sorry for not being able to be more helful.

tiexuedanxin, 2020-11-11:
> Hello,
>
> I used it the FLIR camera with melodic 6 to 8 months ago.
> I don't have a camera with me anymore, so I cannot make any test now.
>
> From the message you are getting, I'd bet the error comes [...]

hello, i can provide more error message, could you help debug, very thank you.

davizinho5, 2020-11-11:
Sorry, I cannot help if I cannot reproduce the error. 

Good luck

anoneuro, 2021-06-14:
@tiexuedanxin Hi! Did you by any chance ever get this driver to work? I am also trying to solve this issue using ROS Melodic on Ubuntu 18.04. Thanks!

sarahsallah, 2022-07-07:
Hi, I'm facing an error during catkin_make for spinnaker in ros melodic.. anyone can help me ? tq

Hateley123, 2022-09-02:
> > `pointgre_camera_driver` is now released on Melodic and you should not need to compile locally anymore.
> 
> But `pointgre_camera_driver` is based on FlyCapture2 SDK. Is it also compatible with Spinnaker SDK

was this solved?