ros-drivers/flir_camera_driver #39: Varous runtime errors, not able to get colour etc.

LukeAI, 2019-11-26:
Blackfly BFLY-PGE-31S4C a number of features are not writable or not implemented:

The camera basically works in that I can see images but I can't get colour, high resolution or set the frame-rate.

Warnings:
BinningHorizontal, DecimationHorizontal, AcquisitionFrameRate, LineSource, GainSelector

Not implemented errors:
AcquisitionFrameRateEnable, AutoExposureExposureTimeUpperLimit, GammaEnable

Full output:
```
process[camera/camera_nodelet_manager-1]: started with pid [2098]
process[camera/spinnaker_camera_nodelet-2]: started with pid [2099]
process[camera/image_proc_debayer-3]: started with pid [2100]
[ INFO] [1574784402.353763089]: Initializing nodelet with 12 worker threads.
[ INFO] [1574784408.701658810]: [SpinnakerCamera]: Number of cameras detected: 1
[ INFO] [1574784408.867011834]: [SpinnakerCamera::connect]: Using Serial: 19095433
[ INFO] [1574784408.867087172]: [SpinnakerCamera::connect]: Detected device type: GEV
[ INFO] [1574784409.656906252]: [SpinnakerCamera::connect]: Camera model name: Blackfly BFLY-PGE-31S4C
[ INFO] [1574784409.671622500]: [SpinnakerCamera]: (19095433) DeviceLinkThroughputLimit set to 125000000.
[ WARN] [1574784409.671769296]: SpinnakerCamera::connect: Could not detect camera model name.
[ WARN] [1574784409.899066243]: [SpinnakerCamera]: (19095433) Feature BinningHorizontal not writable.
[ INFO] [1574784409.902410337]: [SpinnakerCamera]: (19095433) BinningVertical set to 1.
[ WARN] [1574784409.903575022]: [SpinnakerCamera]: (19095433) Feature DecimationHorizontal not writable.
[ INFO] [1574784409.907756162]: [SpinnakerCamera]: (19095433) DecimationVertical set to 1.
[ INFO] [1574784409.913556481]: [SpinnakerCamera]: (19095433) OffsetX set to 0.
[ INFO] [1574784409.914956894]: [SpinnakerCamera]: (19095433) OffsetY set to 0.
[ INFO] [1574784409.917245078]: [SpinnakerCamera]: (19095433) Width set to 2048.
[ INFO] [1574784409.918432521]: [SpinnakerCamera]: (19095433) Height set to 1536.
[ INFO] [1574784409.919768754]: [SpinnakerCamera]: (19095433) OffsetX set to 0.
[ INFO] [1574784409.921089129]: [SpinnakerCamera]: (19095433) OffsetY set to 0.
[ INFO] [1574784409.928689112]: [SpinnakerCamera]: (19095433) PixelFormat set to Mono8.
[ERROR] [1574784409.928819828]: [SpinnakerCamera]: (19095433) Feature name AcquisitionFrameRateEnable not implemented.
[ WARN] [1574784409.934272066]: [SpinnakerCamera]: (19095433) Feature AcquisitionFrameRate not writable.
[ERROR] [1574784409.934426260]: [SpinnakerCamera]: (19095433) Feature name AcquisitionFrameRateEnable not implemented.
[ INFO] [1574784409.937633064]: [SpinnakerCamera]: (19095433) TriggerMode set to Off.
[ INFO] [1574784409.939980283]: [SpinnakerCamera]: (19095433) TriggerSource set to Line0.
[ INFO] [1574784409.941372040]: [SpinnakerCamera]: (19095433) TriggerSelector set to FrameStart.
[ INFO] [1574784409.942596100]: [SpinnakerCamera]: (19095433) TriggerActivation set to FallingEdge.
[ INFO] [1574784409.944807673]: [SpinnakerCamera]: (19095433) TriggerMode set to Off.
[ INFO] [1574784409.947047021]: [SpinnakerCamera]: (19095433) LineSelector set to Line0.
[ INFO] [1574784409.954509834]: [SpinnakerCamera]: (19095433) LineMode set to Input.
[ WARN] [1574784409.967091959]: [SpinnakerCamera]: (19095433) Enumeration LineSource not available.
[ INFO] [1574784409.969333417]: [SpinnakerCamera]: (19095433) ExposureMode set to Timed.
[ INFO] [1574784409.976590234]: [SpinnakerCamera]: (19095433) ExposureAuto set to Continuous.
[ERROR] [1574784409.976761286]: [SpinnakerCamera]: (19095433) Feature name AutoExposureExposureTimeUpperLimit not implemented.
[ WARN] [1574784409.976894345]: [SpinnakerCamera]: (19095433) Enumeration GainSelector not writable.
[ INFO] [1574784409.988170583]: [SpinnakerCamera]: (19095433) GainAuto set to Continuous.
[ INFO] [1574784409.998541485]: [SpinnakerCamera]: (19095433) BlackLevel set to 1.70898.
[ERROR] [1574784409.998633443]: [SpinnakerCamera]: (19095433) Feature name GammaEnable not implemented.
[ WARN] [1574784410.003143493]: [SpinnakerCamera]: (19095433) Feature Gamma not writable.
```

doubleonick, 2020-02-20:
[ INFO] [1574784409.928689112]: [SpinnakerCamera]: (19095433) PixelFormat set to Mono8.

I am also getting this.  I edit the pixel format in spinview, and the changes do not carry over (I get warnings about the device not being open).  I have also tried adding the two lines below to "camera.launch" to no avail.  How can we get color images?

 <arg name="color"		    default="true"
<param name="color"             value="$(arg color)" />

mhosmar-cpr, 2020-05-04:
This camera is not currently supported as denoted by the line: `[ WARN] [1574784409.671769296]: SpinnakerCamera::connect: Could not detect camera model name.` and in the readme.  Please refer to the contributing section of the readme and consider making a pull request.  Different camera models have implemented similar functions with different parameter names and so this driver can only work with models which have had a customization class made.

As you saw `AquisitionFramerateEnable` doesn't exist for this camera and must be called something else, so you won't be able to set the frame rate.  You may be able to get color, try launching dynamic reconfigure and adjusting the parameters while the driver is running.  Once you have parameters that work there.  You can add those to a launch file so they will be set on launch.

closed 2020-05-04 by mhosmar-cpr