ros-drivers/flir_camera_driver #49: [Issue with the latest spinnaker SDK (spinnaker-2.0.0.109)] Cannot start acquisition

zgxsin, 2020-04-30:
I am using ROS Melodic on Ubuntu 18.04 with spinnaker SDK version 2.0.0.109. My Camera is BFS-U3-13Y3C-C. I have no issue with spinnaker SDK version  1.23.0.27. I guess something needs to updated in this ros driver to work with the latest SDK.

```
$ roslaunch spinnaker_camera_driver camera.launch 
... logging to /home/guzhou/.ros/log/35050248-8ae6-11ea-bfb4-4c1d96b04f6e/roslaunch-guzhou-laptop-17915.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/guzhou/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://guzhou-laptop:36211/

SUMMARY
========

PARAMETERS
 * /camera/spinnaker_camera_nodelet/frame_id: camera
 * /camera/spinnaker_camera_nodelet/serial: 19060447
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    image_proc_debayer (nodelet/nodelet)
    spinnaker_camera_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17925]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 35050248-8ae6-11ea-bfb4-4c1d96b04f6e
process[rosout-1]: started with pid [17936]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [17943]
process[camera/spinnaker_camera_nodelet-3]: started with pid [17944]
[ INFO] [1588253194.227189722] [/camera/camera_nodelet_manager] [Loader::Impl::advertiseRosApi] [7f52c4081780]: Initializing nodelet with 12 worker threads.
process[camera/image_proc_debayer-4]: started with pid [17969]
[ INFO] [1588253195.975878142] [/camera/camera_nodelet_manager] [SpinnakerCamera::connect] [7f52c4081780]: [SpinnakerCamera]: Detected device type: U3V. Camera model name: 'Blackfly S BFS-U3-13Y3C' with serial '19060447'.
[ERROR] [1588253195.981307863] [/camera/camera_nodelet_manager] [SpinnakerCameraNodelet::paramCallback] [7f52c4081780]: Reconfigure Callback failed with error: [SpinnakerCamera::start] Failed to start capture with error: Spinnaker: Could not start acquisition. Please try reconnecting the device. [-1013]

```

zgxsin, 2020-06-10:
@civerachb-cpr Could you have a look at this?

buckleytoby, 2023-02-21:
I'm having the same issue with ubuntu 20 and spinnaker 3.0, any ideas?