ros-drivers/flir_camera_driver #57: Image color not Working correct

diogon01, 2020-10-09:
Dears, 
I try set image color and this not working, i use the params "BayerRG8", "RGB8" and "BGR8"

My result: 
[image]

This my launch file.
```
<launch>
   <!-- Determine this using rosrun spinnaker_camera_driver list_cameras.
       If not specified, defaults to first camera found. -->
  <arg name="camera_name"               default="pointgrey" />
  <arg name="camera_serial"             default="" />
  <arg name="calibrated"                default="0" />
 <arg name="image_format_color_coding" value="BayerRG8" />

  <!-- When unspecified, the driver will use the default framerate as given by the
      camera itself. Use the parameter 'control_frame_rate' to enable manual frame 
      rate control, and 'frame_rate' to set the frame rate value. -->
  <arg name="control_frame_rate"        default="false" />
  <arg name="frame_rate"                default="30" />

  <group ns="$(arg camera_name)">
    <!-- Nodelet manager -->
    <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" cwd="node" output="screen"/>

    <!-- Camera nodelet -->
    <node pkg="nodelet" type="nodelet" name="spinnaker_camera_nodelet"
          args="load spinnaker_camera_driver/SpinnakerCameraNodelet camera_nodelet_manager" >

      <param name="frame_id"                        value="camera" />
      <param name="serial"                          value="$(arg camera_serial)" />

      <!-- Frame rate -->
      <param name="acquisition_frame_rate_enable"   value="$(arg control_frame_rate)" />
      <param name="acquisition_frame_rate"          value="$(arg frame_rate)" />

      <!-- Color Format -->
	  <param name="image_format_color_coding" 		value="$(arg image_format_color_coding)" />
	  
	  
	  <!-- Use the camera_calibration package to create this file -->
      <param name="camera_info_url" if="$(arg calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
    </node>

    <!-- Debayering nodelet -->
    <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer camera_nodelet_manager">
    </node>
  </group>
</launch>
```

tiexuedanxin, 2020-11-13:
> Dears,
> I try set image color and this not working, i use the params "BayerRG8", "RGB8" and "BGR8"
> 
> My result:
> [image]
> 
> This my launch file.
> 
> ```
> <launch>
>    
>   <arg name="camera_name"               default="pointgrey" />
>   <arg name="camera_serial"             default="" />
>   <arg name="calibrated"                default="0" />
>  <arg name="image_format_color_coding" value="BayerRG8" />
> 
>   
>   <arg name="control_frame_rate"        default="false" />
>   <arg name="frame_rate"                default="30" />
> 
>   <group ns="$(arg camera_name)">
>     
>     <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" cwd="node" output="screen"/>
> 
>     
>     <node pkg="nodelet" type="nodelet" name="spinnaker_camera_nodelet"
>           args="load spinnaker_camera_driver/SpinnakerCameraNodelet camera_nodelet_manager" >
> 
>       <param name="frame_id"                        value="camera" />
>       <param name="serial"                          value="$(arg camera_serial)" />
> 
>       
>       <param name="acquisition_frame_rate_enable"   value="$(arg control_frame_rate)" />
>       <param name="acquisition_frame_rate"          value="$(arg frame_rate)" />
> 
>       
> 	  <param name="image_format_color_coding" 		value="$(arg image_format_color_coding)" />
> 	  
> 	  
> 	  
>       <param name="camera_info_url" if="$(arg calibrated)"
>              value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
>     </node>
> 
>     
>     <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
>           args="load image_proc/debayer camera_nodelet_manager">
>     </node>
>   </group>
> </launch>
> ```

hello, do you know how to set the parameters for camera.

diogon01, 2020-11-25:
> hello, do you know how to set the parameters for camera.

@tiexuedanxin i'm change the color white balance for `once` and this work

ParthAD3, 2021-05-20:
Hi @diogon01 
Did you solve this problem?

diogon01, 2021-05-21:
> Hi @diogon01
> Did you solve this problem?

Yes man, 
This work after change exposure_auto to "Once" in file white balance in the [Spinnaker.cfg](https://github.com/ros-drivers/flir_camera_driver/blob/kinetic-devel/spinnaker_camera_driver/cfg/Spinnaker.cfg) line number 85

diogon01, 2021-05-21:
This work after change exposure_auto to "Once" in file white balance in the Spinnaker.cfg line number 85

closed 2021-05-21 by diogon01