ros-drivers/flir_camera_driver #58: Failed to load nodelet

tiexuedanxin, 2020-11-10:
hello, when i run the command of roslaunch " spinnaker_camera_driver camera.launch "      the following error occured, is there someone know how to solve it,thanks.

[ERROR] [1605011572.432740699]: Failed to load nodelet [/camera/spinnaker_camera_nodelet] of type [spinnaker_camera_driver/SpinnakerCameraNodelet] even after refreshing the cache: Failed to load library /home/yong/catkin_ws/devel/lib//libSpinnakerCameraNodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libSpinnaker.so.1: cannot open shared object file: No such file or directory)
[ERROR] [1605011572.432760831]: The error before refreshing the cache was: Failed to load library /home/yong/catkin_ws/devel/lib//libSpinnakerCameraNodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libSpinnaker.so.1: cannot open shared object file: No such file or directory)
[FATAL] [1605011572.432879403]: Failed to load nodelet '/camera/spinnaker_camera_nodelet` of type `spinnaker_camera_driver/SpinnakerCameraNodelet` to manager `camera_nodelet_manager'
[camera/spinnaker_camera_nodelet-3] process has died [pid 8548, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load spinnaker_camera_driver/SpinnakerCameraNodelet camera_nodelet_manager __name:=spinnaker_camera_nodelet __log:=/home/yong/.ros/log/d9758ff2-2350-11eb-9fb1-a4bb6dc92670/camera-spinnaker_camera_nodelet-3.log].
log file: /home/yong/.ros/log/d9758ff2-2350-11eb-9fb1-a4bb6dc92670/camera-spinnaker_camera_nodelet-3*.log
^C[camera/image_proc_debayer-4] killing on exit
[camera/camera_nodelet_manager-2] killing on exit
^X[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

eleathers, 2021-02-03:
I am also getting this error. Were you able to resolve it?

kylelscott, 2021-02-15:
I am also getting this error. Has a solution been found?

andreucm, 2021-04-21:
Hi
I experience the same runtime error "Failed to load nodelet". In my case with:
- Ubuntu 20.04
- Kernel 5.8.0-50
- Spinnaker 2.3.0.77
- ROS Noetic
- Camera BlackFly S, GigE

the utility spinview works correctly. 

Have you fixed it somehow @tiexuedanxin @kylelscott @eleathers ?

Any idea ?

Thanks

eleathers, 2021-04-21:
> Hi
> I experience the same runtime error "Failed to load nodelet". In my case with:
>
> * Ubuntu 20.04
> * Kernel 5.8.0-50
> * Spinnaker 2.3.0.77
> * ROS Noetic
> * Camera BlackFly S, GigE
>
> the utility spinview [...]

@andreucm ,

I was able to get it working by reverting back to SDK Version 2.0.0.146. 

However, when using Ubuntu 20.04, I cannot get SpinView to work with both SDK versions 2.3.0.77 or 2.0.0.146.

andreucm, 2021-04-21:
ok! thanks @eleathers !

I identified the crashing [line](https://github.com/ros-drivers/flir_camera_driver/blob/faf7cd658821c9d2fa3fbfea20b1e1c33530fe5d/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L56):
`system_(Spinnaker::System::GetInstance())`

However, FLIR testing program "/opt/spinnaker/src/Acquisition/Acquisition.cpp" also calls GetInstance in a quite different way: 
`SystemPtr system = System::GetInstance();` 
In my case , FLIR test Acquisition program runs ok. 

I'm going to modify the code to check this hypothesis....

Any further idea ?

Andreu

andreucm, 2021-04-21:
`roslaunch spinnaker_camera_driver test_spinnaker.launch`

also crashes at this [line](https://github.com/ros-drivers/flir_camera_driver/blob/faf7cd658821c9d2fa3fbfea20b1e1c33530fe5d/spinnaker_camera_driver/src/spinnaker_test_node.cpp#L36):
`  Spinnaker::SystemPtr system = Spinnaker::System::GetInstance();` 

So it seems that we have some ld runtime issue ?

Any further idea ?

thanks

Andreu

andreucm, 2021-04-21:
Focusing my issue at https://github.com/ros-drivers/flir_camera_driver/issues/72