ros-drivers/flir_camera_driver #59: feature upgrade for ubuntu 20.04 and ROS noetic

CrazyPopLin, 2020-11-14:
Hi, thanks for contribution to this driver. 

I am using Ubuntu 20.04 and ROS noetic. Before I install this ros driver, I've installed [GUI Spinnaker SDK - SpinView](https://www.flir.com/support-center/iis/machine-vision/downloads/spinnaker-sdk-and-firmware-download/) and played FLIR BFS camera. It's will install lib and header in **/opt/spinnaker**. Since Ubuntu 20.04 not install python2 by default, the **download_spinnaker** script will not work. 

So I am suggesting to modify the FindSpinnaker.cmake by following:
```cmake
find_path(Spinnaker_INCLUDE_DIRS NAMES
  ...
  /opt/spinnaker/include
) 
find_library(Spinnaker_LIBRARIES NAMES Spinnaker
 ...
 /opt/spinnaker/lib/
)
```
It's easy for people who using Ubuntu 20.04. And SpinView is always good to play for beginners.

niciBume, 2021-02-09:
Hi!
Did you find a way to get the driver running in noetic?

niciBume, 2021-02-09:
This suggestion compiles! Thanks a lot!
But when running `roslaunch spinnaker_camera_driver camera.launch` I get the following error: 
<pre><font color="#CC0000">[FATAL] [1612864400.250026824]: Failed to load nodelet &apos;/camera/spinnaker_camera_nodelet` of type `spinnaker_camera_driver/SpinnakerCameraNodelet` to manager `camera_nodelet_manager&apos;</font>
<font color="#CC0000">[camera/camera_nodelet_manager-1] process has died [pid 19198, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/niclas/.ros/log/3449c85e-6aa9-11eb-bb28-89d65f55becc/camera-camera_nodelet_manager-1.log].</font>
<font color="#CC0000">log file: /home/niclas/.ros/log/3449c85e-6aa9-11eb-bb28-89d65f55becc/camera-camera_nodelet_manager-1*.log</font>
</pre>

Have you been able to overcome this?
Cheers!

closed 2021-02-11 by mhosmar-cpr

buckleytoby, 2023-02-21:
start roscore?

PR #63 (2021-02-11): Add /opt/spinnaker to spinnaker discovery options
Quick fix for finding `spinnaker` (tested using the current SDK version: `2.2.0.48` from [here](https://meta.box.lenovo.com/v/link/view/a1995795ffba47dbbe45771477319cc3)

Closes #59