ros-drivers/flir_camera_driver #71: [ERROR] [1618405746.078545182]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]

xjp99v5, 2021-04-14:
roslauch file:
```
<?xml version="1.0"?>
<!--
Software License Agreement (BSD)

\file      stereo.launch
\authors   Michael Hosmar <mhosmar@clearpathrobotics.com>
\copyright Copyright (c) 2018, Clearpath Robotics, Inc., All rights reserved.

Redistribution and use in source and binary forms, with or without modification, are permitted provided that
the following conditions are met:
 * Redistributions of source code must retain the above copyright notice, this list of conditions and the
   following disclaimer.
 * Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the 
   following disclaimer in the documentation and/or other materials provided with the distribution.
 * Neither the name of Clearpath Robotics nor the names of its contributors may be used to endorse or promote
   products derived from this software without specific prior written permission.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WAR-
RANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR
PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, IN-
DIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT
OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
-->
<launch>
  <!-- Common parameters -->
  <arg name="camera_name"               default="stereo" />

  <!-- When unspecified, the driver will use the default framerate as given by the
      camera itself. Use the parameter 'control_frame_rate' to enable manual frame 
      rate control, and 'frame_rate' to set the frame rate value. -->
  <arg name="control_frame_rate"        default="false" />
  <arg name="frame_rate"                default="15" />

  <arg name="left_camera_serial"        default="20535906" />
  <arg name="left_camera_calibrated"    default="true" />

  <arg name="right_camera_serial"       default="19381267" />
  <arg name="right_camera_calibrated"   default="true" />

  <arg name="run_stereo_image_proc"     default="true" />

  <!-- Nodelet manager -->
  <!-- Both cameras are not loaded into one nodelet manager to avoid the namespacing issue. -->
  <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" />

  <group ns="$(arg camera_name)" >
    <group ns="left" >
      <!-- Camera nodelet -->
      <node pkg="nodelet" type="nodelet" name="camera_nodelet"
            args="load spinnaker_camera_driver/SpinnakerCameraNodelet /camera_nodelet_manager" >
        <param name="frame_id"                        value="camera_left" />
        <param name="serial"                          value="$(arg left_camera_serial)" />

        <!-- Frame rate -->
        <param name="acquisition_frame_rate_enable"   value="$(arg control_frame_rate)" />
        <param name="acquisition_frame_rate"          value="$(arg frame_rate)" />
	
	<!-- Triggering -->
	<param name="line_source" value="ExposureActive" />	
	<param name="line_selector" value="Line2" />
	<param name="line_mode" value="Output" />
	
        <!-- Use the camera_calibration package to create this file -->
        <param name="camera_info_url" if="$(arg left_camera_calibrated)"
               value="file://$(env HOME)/.ros/camera_info/$(arg left_camera_serial).yaml" />
        <!-- <remap from="image_raw" to="image" /> -->
      </node>

      <!-- Debayering nodelet -->
      <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer /camera_nodelet_manager">
      </node>
    </group>

    <group ns="right" >
      <!-- Camera nodelet -->
      <node pkg="nodelet" type="nodelet" name="camera_nodelet"
            args="load spinnaker_camera_driver/SpinnakerCameraNodelet /camera_nodelet_manager" >
        <param name="frame_id"                        value="camera_right" />
        <param name="serial"                          value="$(arg right_camera_serial)" />

        <!-- Frame rate -->
        <param name="acquisition_frame_rate_enable"   value="$(arg control_frame_rate)" />
        <param name="acquisition_frame_rate"          value="$(arg frame_rate)" />
	
	<!-- Triggering -->
	<param name="enable_trigger" value="On" />
	<param name="trigger_source" value="Line3" />
	<param name="trigger_activation_mode" value="RisingEdge" />
	<param name="trigger_overlap_mode" value="ReadOut" />

        <!-- Use the camera_calibration package to create this file -->
        <param name="camera_info_url" if="$(arg right_camera_calibrated)"
               value="file://$(env HOME)/.ros/camera_info/$(arg right_camera_serial).yaml" />
        <!-- <remap from="image_raw" to="image" /> -->
      </node>

      <!-- Debayering nodelet -->
      <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer /camera_nodelet_manager">
      </node>
    </group>

    <!-- Stereo image processing nodelet -->
    <group if="$(arg run_stereo_image_proc)" >
      <node pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc">
        <!-- <remap from="left/image_raw" to="left/image"/>
        <remap from="right/image_raw" to="right/image"/> -->
        <param name="approximate_sync" value="true"/>
      </node>
    </group>

  </group>
</launch>
```

When I output the right/camera_raw, it shows [ERROR] [1618405746.078545182]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]
[ERROR] [1618405746.079345705]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]

FranzAlbers, 2022-03-04:
We run into the same issue. This only occurs whenever we try to set "enable_trigger" to "On".
Has someone any information on how we can solve this issue?

andreucm, 2022-11-11:
Hi , we also experience sometimes this issue, quite randomly in time. Sometimes after some mintues or even tens of minuts. 

- System: Ubuntu 20.04
- ROS: Noetic
- Camera: [BFS-PGE-200S6C-C](https://www.flir.es/products/blackfly-s-gige/?model=BFS-PGE-200S6C-C) , 20Mpx

An hypothesis I do is that it could be related to the big size of the image this camera provides.

Did you solve it or get some clue ?

Thanks

ana-lan, 2023-08-23:
Did anyone solve this issue?

maninka123, 2025-02-03:
Yes, I finally found the reason for the error. The images are too large to stream. To fix this, downsample the images by increasing the image_format_x_binning and image_format_y_binning values. Also, reduce the image_format_roi_width and image_format_roi_height.

You should then see the image output.

In the camera.launch file, this section is commented out. First, uncomment it and make the necessary changes.