ros-drivers/flir_camera_driver #72: Spinnaker::System::GetInstance() fails from spinnaker_test_node.cpp

andreucm, 2021-04-21:
Hi all, 

### Background
Coming from https://github.com/ros-drivers/flir_camera_driver/issues/58, I wanted to focus the issue much more. 

### Description
when I launch :
`roslaunch spinnaker_camera_driver test_spinnaker.launch`
the node crashes at [line](https://github.com/ros-drivers/flir_camera_driver/blob/faf7cd658821c9d2fa3fbfea20b1e1c33530fe5d/spinnaker_camera_driver/src/spinnaker_test_node.cpp#L36):
`Spinnaker::SystemPtr system = Spinnaker::System::GetInstance();`

and If I launch: 
`roslaunch app camera.launch`
the nodelet fails to be loaded, due also to a crash at [line](https://github.com/ros-drivers/flir_camera_driver/blob/faf7cd658821c9d2fa3fbfea20b1e1c33530fe5d/spinnaker_camera_driver/src/SpinnakerCamera.cpp#L56): 
 `system_(Spinnaker::System::GetInstance())`

### Additional Checks
So , I wanted to directly check with a non-ROS program calling the Spinnaker library, such as the example one at opt/spinnaker/src/Acquisition/Acquisition.cpp. This program also calls Spinnaker::System::GetInstance() at line 431. In this case, Acquisition program runs correctly.  Spinview program also runs ok. 

### My system 

- Ubuntu 20.04
- Kernel 5.8.0-50
- Spinnaker 2.3.0.77
- ROS Noetic
- Camera BlackFly S, GigE
- libspinnaker.conf file at /etc/ld.so.conf.d/, with the entry "/opt/spinnaker/lib"

------

Any idea will be welcome!

Thanks to all

Andreu

andreucm, 2021-04-27:
Hi all !
Just to post some extra findings I've found. 

The following code: 
```
#include "Spinnaker.h"
#include "SpinGenApi/SpinnakerGenApi.h"
#include <iostream>
#include "ros/ros.h"

int main(int argc, char** argv)
{
	ros::init(argc, argv, "flir_bfs_camera_test_node");

	std::cout << __LINE__ << std::endl;
	Spinnaker::SystemPtr system = Spinnaker::System::GetInstance();

	std::cout << __LINE__ << std::endl;
	Spinnaker::CameraList camList = system->GetCameras();
	const unsigned int numCameras = camList.GetSize();
	std::cout << "Number of cameras detected: " << numCameras << std::endl << std::endl;

	return 0;
}
```
Returns a segmentation fault just after the first `__LINE__` print. But if I comment the `ros::inint()` line, then the whole program runs correctly. 

So I decided to run it under gdb. And this is the output: 

```
Starting program: /home/andreu/dev/workspaces/jbc_ws/devel/lib/flir_bfs_camera/simple_test 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
10
[New Thread 0x7ffff1ce4700 (LWP 24856)]
[New Thread 0x7ffff14e3700 (LWP 24857)]
[New Thread 0x7ffff0ce2700 (LWP 24858)]
[New Thread 0x7fffebfff700 (LWP 24859)]
[New Thread 0x7fffeb7fe700 (LWP 24860)]
[New Thread 0x7fffeaffd700 (LWP 24861)]
[New Thread 0x7fffea7fc700 (LWP 24862)]
[Thread 0x7fffea7fc700 (LWP 24862) exited]
[Thread 0x7fffeaffd700 (LWP 24861) exited]
[New Thread 0x7fffea7fc700 (LWP 24863)]
[New Thread 0x7fffeaffd700 (LWP 24864)]
[Thread 0x7fffeaffd700 (LWP 24864) exited]
[Thread 0x7fffea7fc700 (LWP 24863) exited]
[New Thread 0x7fffeaffd700 (LWP 24865)]
[New Thread 0x7fffea7fc700 (LWP 24866)]
[Thread 0x7ffff0ce2700 (LWP 24858) exited]

Thread 2 "simple_test" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff1ce4700 (LWP 24856)]
0x00007ffff4ddf540 in boost::detail::thread_data_base::~thread_data_base() () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
```

carlosjoserg, 2021-06-29:
In late April, days or perhaps hours after the last post by @andreucm , Spinnaker 2.4 was available.

We tried again on Ubuntu 20.04 / Noetic / Spinnaker 2.4, and it works.

Attached image of the stream from the camera using RQt Image View:

[image]

andreucm, 2021-06-29:
Great @carlosjoserg ! 
Willing ti try it, and in case of success close the issue. 
thanks!

andreucm, 2021-07-08:
Yeees! Tested successfully with : 

- Ubuntu 20.04
- Kernel 5.8.0-59
- Spinnaker 2.4.0.143
- ROS Noetic
- Camera BlackFly S, GigE
- libspinnaker.conf file at /etc/ld.so.conf.d/, with the entry "/opt/spinnaker/lib"

closed 2021-07-08 by andreucm