ros-drivers/flir_camera_driver #90: make[2]: *** No rule to make target 'flir_camera_driver/spinnaker_camera_driver/usr/lib/libSpinnaker.so', needed by '/home/deanderosa/catkin_ws/devel/lib/libSpinnakerCameraLib.so'.

dderosa1, 2022-02-26:
When I tried to make the repo I got the following error:

make[2]: *** No rule to make target 'flir_camera_driver/spinnaker_camera_driver/usr/lib/libSpinnaker.so', needed by '/home/deanderosa/catkin_ws/devel/lib/libSpinnakerCameraLib.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:4190: flir_camera_driver/spinnaker_camera_driver/CMakeFiles/SpinnakerCameraLib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

I can no longer download Spinnaker using the script. It looks like libSpinnaker is moved to a destination directory specified by ${CMAKE_CURRENT_BINARY_DIR}/usr/include/spinnaker when called by DownloadSpinnaker.cmake.  

Should I copy the files libSpinnaker.so.2.6.0.156, alibSpinnaker.so, and libSpinnaker.so from flir_camera_driver/spinnaker_camera_driver/usr/lib/libSpinnaker.so to ~/usr/include/spinnaker?

fherreroc, 2023-04-28:
Happened to me also (building on noetic), and fixed it by changing the FLIR_LIB_VAR path from `.../user/lib...` to `.../opt/spinnaker/lib...`  in file `spinnaker_camera_driver/cmake/DownloadSpinnaker.cmake`

This is the patch file:
``` 
diff --git a/spinnaker_camera_driver/cmake/DownloadSpinnaker.cmake b/spinnaker_camera_driver/cmake/DownloadSpinnaker.cmake
index 93e8efe..9b11ac9 100644
--- a/spinnaker_camera_driver/cmake/DownloadSpinnaker.cmake
+++ b/spinnaker_camera_driver/cmake/DownloadSpinnaker.cmake
@@ -16,6 +16,6 @@ function(download_spinnaker FLIR_LIB_VAR FLIR_INCLUDE_DIR_VAR)
   message(STATUS "Running download_spinnaker script with arguments: ${FLIR_ARCH} ${FLIR_DIR} ${OS_CODE_NAME}")
   execute_process(
     COMMAND ${DOWNLOAD_SCRIPT} ${FLIR_ARCH} "${FLIR_DIR}" ${OS_CODE_NAME})
-  set(${FLIR_LIB_VAR} "${CMAKE_BINARY_DIR}/usr/lib/libSpinnaker.so" PARENT_SCOPE)
+  set(${FLIR_LIB_VAR} "${CMAKE_BINARY_DIR}/opt/spinnaker/lib/libSpinnaker.so" PARENT_SCOPE)
   set(${FLIR_INCLUDE_DIR_VAR} "${CMAKE_BINARY_DIR}/opt/spinnaker/include/" PARENT_SCOPE)
 endfunction()
```

thebirdgr, 2024-05-27:
Yes agreed! Worked for me as well.