ros-drivers/flir_camera_driver #102: trigger overlap mode is not set when launching spinnaker camera

pauldeee, 2022-09-10:
Hi all,
I'm running ROS Noetic on Ubuntu 20.04. I'm using an Ouster lidar sensor as an external camera trigger for a FLIR BlackFly S USB3 camera. However, it seems when I launch the node the camera FPS is not what it should be.
 
If I start the Ouster sensor then I launch the spinnaker camera.launch, the frame rate is about 20 fps, but I have the lidar set to trigger at 40 fps.

When I open spinview it shows the camera at 20 Hz. However, if I switch the Trigger Overlap to ReadOut inside of SpinView it starts to work and shows 40 Hz as it should.

So basically I am stuck doing this:
(1) open SpinView manually set the Trigger Overlap to ReadOut
(2) close SpinView
(3) roslaunch spinnaker_camera_driver camera.launch.
And everything works as it should.

Is there someway to successfully apply the Trigger Overlap when I roslaunch?
I've tried to add it to my launch file:
`<param name="trigger_overlap_mode"            value="ReadOut" />`

It seems that ReadOut is set as default in the Spinnaker.cfg:
`gen.add("trigger_overlap_mode",                   str_t,    SensorLevels.RECONFIGURE_RUNNING,                "Trigger Overlap Modes",                                                                      "ReadOut",                       edit_method = trigger_overlap_modes)`

If anyone has any ideas or suggestions, please let me know. Maybe I've missed some other setting I should be using.

Thanks

More Info:
```
SUMMARY
========

PARAMETERS
 * /camera/spinnaker_camera_nodelet/acquisition_frame_rate: 30
 * /camera/spinnaker_camera_nodelet/acquisition_frame_rate_enable: False
 * /camera/spinnaker_camera_nodelet/auto_exposure_time_upper_limit: 16666
 * /camera/spinnaker_camera_nodelet/auto_white_balance: Continuous
 * /camera/spinnaker_camera_nodelet/device_type: USB3
 * /camera/spinnaker_camera_nodelet/enable_trigger: On
 * /camera/spinnaker_camera_nodelet/exposure_auto: Continuous
 * /camera/spinnaker_camera_nodelet/exposure_mode: Timed
 * /camera/spinnaker_camera_nodelet/frame_id: camera
 * /camera/spinnaker_camera_nodelet/image_format_color_coding: BayerBG8
 * /camera/spinnaker_camera_nodelet/image_format_x_reverse: True
 * /camera/spinnaker_camera_nodelet/image_format_y_reverse: True
 * /camera/spinnaker_camera_nodelet/isp_enable: False
 * /camera/spinnaker_camera_nodelet/serial: 0
 * /camera/spinnaker_camera_nodelet/trigger_activation_mode: RisingEdge
 * /camera/spinnaker_camera_nodelet/trigger_source: Line3
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    image_proc_debayer (nodelet/nodelet)
    spinnaker_camera_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [43062]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c1f60f74-309a-11ed-985f-ad2a3d1d286b
process[rosout-1]: started with pid [43088]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [43095]
process[camera/spinnaker_camera_nodelet-3]: started with pid [43096]
process[camera/image_proc_debayer-4]: started with pid [43097]
[ INFO] [1662767676.010097821]: Initializing nodelet with 16 worker threads.
[ INFO] [1662767677.497541598]: [SpinnakerCamera]: Number of cameras detected: 1
[ INFO] [1662767677.524932121]: [SpinnakerCamera::connect]: Using Serial: 0
[ INFO] [1662767677.525023563]: [SpinnakerCamera::connect]: Detected device type: USB3Vision
[ INFO] [1662767677.680354668]: [SpinnakerCamera::connect]: Camera model name: Blackfly S BFS-U3-31S4C
[ INFO] [1662767677.683569159]: [SpinnakerCamera]: (22157397) DeviceLinkThroughputLimit set to 500000000.
[ INFO] [1662767677.713169007]: [SpinnakerCamera]: (22157397) BinningHorizontal set to 1.
[ INFO] [1662767677.720400242]: [SpinnakerCamera]: (22157397) BinningVertical set to 1.
[ INFO] [1662767677.727543453]: [SpinnakerCamera]: (22157397) DecimationHorizontal set to 1.
[ INFO] [1662767677.734668980]: [SpinnakerCamera]: (22157397) DecimationVertical set to 1.
[ INFO] [1662767677.741395839]: [SpinnakerCamera]: (22157397) ReverseX set to 1.
[ INFO] [1662767677.748120545]: [SpinnakerCamera]: (22157397) ReverseY set to 1.
[ INFO] [1662767677.752556316]: [SpinnakerCamera]: (22157397) OffsetX set to 0.
[ INFO] [1662767677.756292615]: [SpinnakerCamera]: (22157397) OffsetY set to 0.
[ INFO] [1662767677.760222848]: [SpinnakerCamera]: (22157397) Width set to 2048.
[ INFO] [1662767677.763879107]: [SpinnakerCamera]: (22157397) Height set to 1536.
[ INFO] [1662767677.767202682]: [SpinnakerCamera]: (22157397) OffsetX set to 0.
[ INFO] [1662767677.770256321]: [SpinnakerCamera]: (22157397) OffsetY set to 0.
[ INFO] [1662767677.777424929]: [SpinnakerCamera]: (22157397) PixelFormat set to BayerBG8.
[ INFO] [1662767677.783026447]: [SpinnakerCamera]: (22157397) IspEnable set to 0.
[ INFO] [1662767677.785143148]: [SpinnakerCamera]: (22157397) AcquisitionFrameRateEnable set to 1.
[ INFO] [1662767677.788152082]: [SpinnakerCamera]: (22157397) AcquisitionFrameRate set to 29.9959.
[ INFO] [1662767677.790339516]: [SpinnakerCamera]: (22157397) AcquisitionFrameRateEnable set to 0.
[ INFO] [1662767677.792064703]: [SpinnakerCamera]: (22157397) TriggerMode set to Off.
[ INFO] [1662767677.793481240]: [SpinnakerCamera]: (22157397) TriggerSource set to Line3.
[ INFO] [1662767677.793612085]: [SpinnakerCamera]: (22157397) TriggerSelector set to FrameStart.
[ INFO] [1662767677.795304330]: [SpinnakerCamera]: (22157397) TriggerActivation set to RisingEdge.
[ INFO] [1662767677.796957702]: [SpinnakerCamera]: (22157397) TriggerMode set to On.
[ INFO] [1662767677.797561935]: [SpinnakerCamera]: (22157397) LineSelector set to Line0.
[ INFO] [1662767677.798635559]: [SpinnakerCamera]: (22157397) LineMode set to Input.
[ INFO] [1662767677.799591192]: [SpinnakerCamera]: (22157397) LineSource set to Off.
[ INFO] [1662767677.800980568]: [SpinnakerCamera]: (22157397) ExposureMode set to Timed.
[ INFO] [1662767677.803635218]: [SpinnakerCamera]: (22157397) ExposureAuto set to Continuous.
[ INFO] [1662767677.805338093]: [SpinnakerCamera]: (22157397) AutoExposureExposureTimeUpperLimit set to 16666.
[ INFO] [1662767677.805637354]: [SpinnakerCamera]: (22157397) GainSelector set to All.
[ INFO] [1662767677.808492230]: [SpinnakerCamera]: (22157397) GainAuto set to Continuous.
[ INFO] [1662767677.810399107]: [SpinnakerCamera]: (22157397) BlackLevel set to 1.7.
[ INFO] [1662767677.811142632]: [SpinnakerCamera]: (22157397) GammaEnable set to 1.
[ INFO] [1662767677.812462037]: [SpinnakerCamera]: (22157397) Gamma set to 1.
[ INFO] [1662767677.815479899]: [SpinnakerCamera]: (22157397) BalanceWhiteAuto set to Continuous.
[ INFO] [1662767677.815562754]: [SpinnakerCamera]: (22157397) AutoAlgorithmSelector set to Ae.
[ INFO] [1662767677.816318011]: [SpinnakerCamera]: (22157397) AasRoiEnable set to 1.
[ INFO] [1662767677.817447369]: [SpinnakerCamera]: (22157397) AutoExposureLightingMode set to Normal.
```

Seokbok, 2022-11-19:
[src/camera.cpp] code shoud be add parameter setting code

[src/camera.cpp] line 86 below add code as

**setProperty(node_map_, "TriggerOverlap", config.trigger_overlap_mode);**

and then after cmake enjoy streaming same as ouster trigger!

pauldeee, 2022-11-19:
A+

closed 2022-11-19 by pauldeee