ros-drivers/flir_camera_driver #107: Framerate drops to half of specified value

adthoms, 2023-05-09:
When launching `camera.launch`, the published image framerate is half of what is specified by the parameter [frame_rate](https://github.com/ros-drivers/flir_camera_driver/blob/d72178beee5b627c6be8ef0fc996b9708c9b430e/spinnaker_camera_driver/launch/camera.launch#LL38C55-L38C55). For example, setting:
```
<arg name="frame_rate"                default="30" />
```
gives a published image framerate of 15 Hz

Tested with: 
FLIR BlackFly S
Ubuntu 20.04
ROS Noetic
Spinnaker SDK v2.7.01.128

andreucm, 2023-05-11:
Hi, 
take into account that frame rate is a "maximum desired" param, and other params, specially image resolution or exposure can cause a reduction on actual frame rate. Check that point with the datasheet. 
Best
Andreu

NikolaasBender, 2024-05-21:
I am having a similar issue on Ros2. My frame rate is being capped at 13.66hz until I use spinview to reset the camera then my Ros params are loaded on launch and I can get my desired frame rates and exposure times. I know I am launching correctly because after the reset, with no change to my launch file everything works perfectly the problem is the intermediary step that the camera needs to be reset.

berndpfrommer, 2024-05-22:
If you want the cameras to go to a well defined state, the way to do it is first to configure the camera the way you want it in spinview. Then save the parameters as a UserSet (spinview has an option somewhere for that), e.g. as UserSet1. Then you hack the camera configuration yaml to add some new nodes (i.e. ros parameters) that allow you to load UserSet1 on startup. I have done this for the blackfly camera and it worked for me, but while figuring out the .yaml file I also hosed the cameras a couple of times to the point that I had to reinstall the firmware to get them to work again. So beware.
Unfortunately I lost my blackfly.yaml file so I only have the launch file with the  ros parameters  (see below) which is not super helpful. The 'user_set_load' ros parameter was an execution node, the others where enums.
```
    'default_user_set': 'UserSet1', 
    'user_set_selector': 'UserSet1',    # load UserSet1 at startup
    'user_set_load': 'Yes',  # any string will do to execute the "load" operation
```
I'll try to recreate the .yaml file when I get to it.

berndpfrommer, 2024-05-22:
I added a new config file to the humble-devel branch, and also put an example in the launch file. If using userset parameters doesn't fix the problem, please open another issue for the ROS2 driver, since that code base is completely different.