ros-drivers/flir_camera_driver #110: Frame rate value not going beyond 10 Hz  for RGB image Acquisition

rajatag77, 2023-06-08:
I am using Flir Firefly FFY-U3-16S2C-CS Camera. When I am Launching Roslaunch file it shows the error:
 [ERROR] [1686212095.839087454]: Reconfigure Callback failed with error: [Camera::setNewConfiguration] Failed to set configuration: Spinnaker: GenICam::OutOfRangeException= Value 9.199263 must be smaller than or equal 9.199262. : OutOfRangeException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 85) [-2002]

[image]

gitwikc, 2024-04-12:
## Hi @rajatag77 Bhaiya!

-MOON Lab

gitwikc, 2024-04-12:
Getting the same problem on Flir Firefly S. Please help