ros-drivers/flir_camera_driver #115: [humble-devel] build fails on rosdistro build farm: spinnaker download not working

berndpfrommer, 2023-07-25:
Seems like the first rosdistro build has run against the humble-devel branch.
[Here is a link where you can find the log files](https://build.ros2.org/job/Hbin_uJ64__spinnaker_camera_driver__ubuntu_jammy_amd64__binary/1/)

It fails with the below error message when trying to download the spinnaker SDK.

The  os_code_name is obtained by ``lsb_release -cs`` in ``DownloadSpinnaker.cmake``. Apparently ``lsb_release`` either does not exist or return an empty string such that sys.argv[3] is undefined. Apparently ``lsb_release`` works just fine for the noetic-devel branch  .
@fazzrazz, @tonybaltovski do you have any idea how to fix this?

```
13:33:05 -- Found inc dir: SPINNAKER_INCLUDE_DIRS-NOTFOUND
13:33:05 -- Found variable: FALSE at SPINNAKER_LIBRARIES-NOTFOUND and SPINNAKER_INCLUDE_DIRS-NOTFOUND
13:33:05 -- spinnaker not found
13:33:05 -- libSpinnaker not found in system library path
13:33:05 -- Running download_spinnaker script with arguments: x86_64 /tmp/binarydeb/ros-humble-spinnaker-camera-driver-2.0.0/.obj-x86_64-linux-gnu/usr/lib 
13:33:05 Traceback (most recent call last):
13:33:05   File "/tmp/binarydeb/ros-humble-spinnaker-camera-driver-2.0.0/cmake/download_spinnaker", line 105, in <module>
13:33:05     os_code_name = sys.argv[3]
13:33:05 IndexError: list index out of range
```

berndpfrommer, 2023-07-25:
After PR #116 has been merged, the log file now shows that lsb_release is not present.
Why not?
```
02:19:29 CMake Error at cmake/DownloadSpinnaker.cmake:20 (find_program):
02:19:29   Could not find LSB_RELEASE_EXEC using the following names: lsb_release
02:19:29 Call Stack (most recent call first):
02:19:29   CMakeLists.txt:50 (download_spinnaker)
```

berndpfrommer, 2023-08-02:
@tonybaltovski sorry I have some newbie questions: 
1) do you have to bloom release the package every time just to test if the modified code compiles on the ROS build farms?
2) along these lines: do you want me to do the bloom release (and do I even have the permissions to do so)? I am just learning the ropes with bloom etc as I'm moving my first ROS package (an event camera driver) into rosdistro.

Thanks,
Bernd

tonybaltovski, 2023-08-03:
1. Generally no, usually if it builds on the CI it works on the buildfarm but something is up with this
2. Sure that would be great!  I can help you too if needed.

berndpfrommer, 2023-08-10:
The latest error is this. PR forthcoming...
```
02:10:53 dpkg-shlibdeps: warning: can't extract name and version from library name 'librcutils.so'
02:10:53 dpkg-shlibdeps: error: cannot find library libSpinnaker.so.3 needed by debian/ros-humble-spinnaker-camera-driver/opt/ros/humble/lib/libcamera_driver.so (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
02:10:53 dpkg-shlibdeps: warning: can't extract name and version from library name 'libcamera_info_manager.so'

```

berndpfrommer, 2023-08-10:
@tonybaltovski With respect to pushing into the development branch, the flir_camera_driver repo behaves differently from another repository that I'm working with (ros-event-camera/metavision_driver). There, the ROS2 build farm polls my master branch once an hour and automatically rebuilds if new commits have arrived. That avoids opening a new rosdistro PR for every attempt to debug a build issue.
If you know how to get the flir_camera_driver repo to behave the same way, please let me know. Is the ROS2 build farm polling a different branch maybe?

berndpfrommer, 2023-08-10:
Now the ffmpeg libraries are missing. Will be fixed in next PR
```
17:54:38 dpkg-shlibdeps: error: cannot find library libswscale.so.5 needed by debian/ros-humble-spinnaker-camera-driver/opt/ros/humble/lib/libSpinVideo.so.3.1.0.79 (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
17:54:38 dpkg-shlibdeps: error: cannot find library libavcodec.so.58 needed by debian/ros-humble-spinnaker-camera-driver/opt/ros/humble/lib/libSpinVideo.so.3.1.0.79 (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
17:54:38 dpkg-shlibdeps: error: cannot find library libavutil.so.56 needed by debian/ros-humble-spinnaker-camera-driver/opt/ros/humble/lib/libSpinVideo.so.3.1.0.79 (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
17:54:38 dpkg-shlibdeps: error: cannot find library libavformat.so.58 needed by debian/ros-humble-spinnaker-camera-driver/opt/ros/humble/lib/libSpinVideo.so.3.1.0.79 (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
```

berndpfrommer, 2023-08-12:
Now amd64 builds, but arm64 fails due to Spinnaker SDK binaries not being available for that platform.

berndpfrommer, 2023-08-14:
Closed by PR #127 
https://build.ros2.org/job/Hbin_ujv8_uJv8__spinnaker_camera_driver__ubuntu_jammy_arm64__binary/

closed 2023-08-14 by berndpfrommer