ros-drivers/flir_camera_driver #120: Driver keeps crashing with GigE Camera

devraise, 2023-08-02:
I'm trying to use the driver with a Blackfly S (BFS-PGE-04S2C-CS). After a few seconds of running it keeps giving me the following error and stops sending any images. Any help would be much appreciated. I'm using Spinnaker version 3.1.0.79

```
[ERROR] [1690942495.116732889]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]
[ERROR] [1690942495.116979827]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
```

KDeng98, 2024-04-10:
Hi there,
I also meet the same problem. Any idea of how to fix it? Thanks.

jtaveau, 2024-05-06:
Got the same issue with this camera (Blackfly S BFS-PGE-63S4C), anyone got results on that?

w577658669, 2024-08-30:
Got the same issue with this camera (A70), anyone got results on that?

jtaveau, 2024-08-30:
Hey, have you tried [increasing your socket buffer size](https://www.flir.com/support-center/iis/machine-vision/knowledge-base/lost-ethernet-data-packets-on-linux-systems/?srsltid=AfmBOorxZJcps0eiyUNJa9COR7yExjqK5-CxIMTnuvZNhedTjqU0n8z1) ?
Are Jumbo packages enabled on your NIC?

w577658669, 2024-08-30:
I didn't increase the socket buffer size.I just used the ROS driver according to https://www.flir.cn/support-center/iis/machine-vision/application-note/using-ros-with-spinnaker/  I don't knowJumbo packages

jtaveau, 2024-08-30:
Then try those 2 things, and that should solve your issue, that's how I solved it on my side.

w577658669, 2024-08-30:
Thanks I'll try it

w577658669, 2024-08-30:
I use the commands,but it doesn't work.
sudo sysctl –w net.core.rmem_max=2147483647
sudo sysctl –w net.core.wmem_max=2147483647
sudo sysctl –w net.core.rmem_default=2147483647
sudo sysctl –w net.core.wmem_default=2147483647

jtaveau, 2024-08-30:
What does not work?

w577658669, 2024-08-30:
The terminal also displays:
[ERROR] [1725006449.112335831]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: LogicalErrorException NULL pointer dereferenced [-2005]
[ERROR] [1725006449.112987800]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]

jtaveau, 2024-08-30:
Can you double check if the values defined are you taken into account?

```
cat /proc/sys/net/core/rmem_default
```

Otherwise, are you using jumbo packages or not?

w577658669, 2024-08-30:
[image]

I don't know if I use jumbo packages or not.

jtaveau, 2024-08-30:
Perfect, sent the output of 
```
ip a
```

w577658669, 2024-08-30:
[image]
I set the MTU size to 9000,but also display:
[image]

jtaveau, 2024-08-30:
Then I don't know, good luck in finding where it's coming from!

w577658669, 2024-08-30:
OK，Thanks  a lot even if the problem hasn't been solved.