ros-drivers/flir_camera_driver #130: Teledyne Camera

theDimitron, 2023-09-08:
Hey,

I'm trying to start the ROS-node with my Teledyne Nano-C1940 (over GigE). The instructions I was following: 
https://www.flir.de/support-center/iis/machine-vision/application-note/using-ros-with-spinnaker/
Unfortunately without success. At the initialization phase a bunch of the errors are coming out and if i try to grab any image topic the node simply crashes within a couple  of seconds.

Do I do something wrong, or the driver dose nos support that type (manufacture) of camera?

 Here is the terminal output:

_jetson:~/catkin_ws$ roslaunch spinnaker_camera_driver camera.launch
... logging to /home/horizonuam/.ros/log/3eb09878-4e2b-11ee-903b-48b02da4c8ac/roslaunch-horizonuam-jetson-68216.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson:35823/

SUMMARY

PARAMETERS
 * /camera/spinnaker_camera_nodelet/acquisition_frame_rate: 30
 * /camera/spinnaker_camera_nodelet/acquisition_frame_rate_enable: True
 * /camera/spinnaker_camera_nodelet/auto_white_balance: Continuous
 * /camera/spinnaker_camera_nodelet/device_type: GigE
 * /camera/spinnaker_camera_nodelet/frame_id: camera
 * /camera/spinnaker_camera_nodelet/image_format_color_coding: BayerRG8
 * /camera/spinnaker_camera_nodelet/isp_enable: False
 * /camera/spinnaker_camera_nodelet/serial: 0
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    image_proc_debayer (nodelet/nodelet)
    spinnaker_camera_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [68235]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3eb09878-4e2b-11ee-903b-48b02da4c8ac
process[rosout-1]: started with pid [68256]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [68264]
process[camera/spinnaker_camera_nodelet-3]: started with pid [68265]
process[camera/image_proc_debayer-4]: started with pid [68266]
[ INFO] [1694165814.561969255]: Initializing nodelet with 12 worker threads.
[ INFO] [1694165816.087964356]: [SpinnakerCamera]: Number of cameras detected: 1
[ INFO] [1694165816.135005844]: [SpinnakerCamera::connect]: Using Serial: 10
[ INFO] [1694165816.135230007]: [SpinnakerCamera::connect]: Detected device type: GigEVision
[ INFO] [1694165816.927920305]: [SpinnakerCamera::connect]: Camera model name: Nano-C1940
[ INFO] [1694165816.932048723]: [SpinnakerCamera]: (H2511395) DeviceLinkThroughputLimit set to 125000000.
[ WARN] [1694165816.932232789]: SpinnakerCamera::connect: Could not detect camera model name.
[ERROR] [1694165816.975518409]: [SpinnakerCamera]: (H2511395) Feature name BinningHorizontal not implemented.
[ERROR] [1694165816.975821870]: [SpinnakerCamera]: (H2511395) Feature name BinningVertical not implemented.
[ERROR] [1694165816.976107570]: [SpinnakerCamera]: (H2511395) Feature name DecimationHorizontal not implemented.
[ERROR] [1694165816.976336694]: [SpinnakerCamera]: (H2511395) Feature name DecimationVertical not implemented.
[ INFO] [1694165816.977549866]: [SpinnakerCamera]: (H2511395) ReverseX set to 0.
[ INFO] [1694165816.978600154]: [SpinnakerCamera]: (H2511395) ReverseY set to 0.
[ INFO] [1694165816.983527689]: [SpinnakerCamera]: (H2511395) OffsetX set to 0.
[ INFO] [1694165816.990922399]: [SpinnakerCamera]: (H2511395) OffsetY set to 0.
[ INFO] [1694165816.996864414]: [SpinnakerCamera]: (H2511395) Width set to 1936.
[ INFO] [1694165817.000099186]: [SpinnakerCamera]: (H2511395) Height set to 1216.
[ INFO] [1694165817.001866030]: [SpinnakerCamera]: (H2511395) OffsetX set to 0.
[ INFO] [1694165817.003614346]: [SpinnakerCamera]: (H2511395) OffsetY set to 0.
[ INFO] [1694165817.005441479]: [SpinnakerCamera]: (H2511395) PixelFormat set to BayerRG8.
[ERROR] [1694165817.005655787]: [SpinnakerCamera]: (H2511395) Feature name IspEnable not implemented.
[ INFO] [1694165817.008166579]: [SpinnakerCamera]: (H2511395) AcquisitionFrameRateEnable set to 1.
[... 59 lines cut ...]
[ WARN] [1694165842.410113845]:  - EnumEntry_LineSelector_Line4 (display Line 4, symbolic Line4)
[ WARN] [1694165842.410208439]: [SpinnakerCamera]: (H2511395) Enumeration LineMode not writable.
[ERROR] [1694165842.410298328]: [SpinnakerCamera]: (H2511395) Enumeration name LineSource not implemented.
[ INFO] [1694165842.411260423]: [SpinnakerCamera]: (H2511395) ExposureMode set to Timed.
[ WARN] [1694165842.412185398]: [SpinnakerCamera]: (H2511395) Enumeration ExposureAuto not available.
[ERROR] [1694165842.412290264]: [SpinnakerCamera]: (H2511395) Feature name AutoExposureExposureTimeUpperLimit not implemented.
[ WARN] [1694165842.412397626]: [SpinnakerCamera]: (H2511395) Entry name All not available.
[ WARN] [1694165842.412465371]: Available:
[ WARN] [1694165842.412559548]:  - EnumEntry_GainSelector_SensorAll (display Sensor, symbolic SensorAll)
[ WARN] [1694165842.412640253]:  - EnumEntry_GainSelector_DigitalAll (display Digital, symbolic DigitalAll)
[ WARN] [1694165842.412769343]: [SpinnakerCamera]: (H2511395) Enumeration GainAuto not available.
[ INFO] [1694165842.415180678]: [SpinnakerCamera]: (H2511395) BlackLevel set to 2.
[ERROR] [1694165842.415244551]: [SpinnakerCamera]: (H2511395) Feature name GammaEnable not implemented.
[ERROR] [1694165842.415300872]: [SpinnakerCamera]: (H2511395) Feature name Gamma not implemented.
[ WARN] [1694165842.415381449]: [SpinnakerCamera]: (H2511395) Entry name Continuous not available.
[ WARN] [1694165842.415419146]: Available:
[ WARN] [1694165842.415492971]:  - EnumEntry_BalanceWhiteAuto_Off (display Off, symbolic Off)
[ WARN] [1694165842.415534348]:  - EnumEntry_BalanceWhiteAuto_OnDemand (display On Demand, symbolic OnDemand)
[ WARN] [1694165842.415579340]:  - EnumEntry_BalanceWhiteAuto_Periodic (display Periodic, symbolic Periodic)
[ INFO] [1694165842.482064467]: using default calibration URL
[ INFO] [1694165842.482214453]: camera calibration URL: file:///home/horizonuam/.ros/camera_info/0.yaml
[ INFO] [1694165842.482368280]: Unable to open camera calibration file [/home/horizonuam/.ros/camera_info/0.yaml]
[ WARN] [1694165842.482411032]: Camera calibration file /home/horizonuam/.ros/camera_info/0.yaml not found.
[ERROR] [1694165859.981454366]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]
[ERROR] [1694165859.982058888]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]_

jtaveau, 2024-05-06:
I've exactly the same using a Blackfly S BFS-PGE-63S4C, which Jetson are you using?

jtaveau, 2024-05-07:
I found the issue, I'll open a PR by the end of this week.

theDimitron, 2024-05-15:
> I've exactly the same using a Blackfly S BFS-PGE-63S4C, which Jetson are you using?

Hey, it was Jetson Orin AGX, and do not remember what Jetpack that was

jtaveau, 2024-05-20:
Apparently the socket buffer size is too small on the Orin, even if you;re using the 10G interface, using the [commands at the bottom increasing rmem](https://www.flir.com/support-center/iis/machine-vision/application-note/streaming-cameras-on-embedded-systems/)  will solve this issue. 

```
sudo sysctl –w net.core.rmem_max=33554432
sudo sysctl –w net.core.wmem_max=33554432
sudo sysctl –w net.core.rmem_default=33554432
sudo sysctl –w net.core.wmem_default=33554432
```