ros-drivers/flir_camera_driver #138: Cannot locate rosdep definition

Chiwingum11, 2023-10-30:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
flir_camera_description: Cannot locate rosdep definition for [xacro]
spinnaker_camera_driver: Cannot locate rosdep definition for [camera_info_manager]
ros@ros-CELSIUS-H760:~/flir_ros2_ws$

berndpfrommer, 2023-10-30:
Post output of "rosdep update"
and "rosdep db | grep xacro"

Chiwingum11, 2023-10-30:
Nothing has changed. I saw that this was tested and I'm running on the same environment. ROS2 Galactic
Software:
    ROS2 Galactic under Ubuntu 20.04 LTS

Chiwingum11, 2023-10-30:
CMake Error at CMakeLists.txt:85 (find_package):
  By not providing "Findcamera_info_manager.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "camera_info_manager", but CMake did not find one.

  Could not find a package configuration file provided by
  "camera_info_manager" with any of the following names:

    camera_info_managerConfig.cmake
    camera_info_manager-config.cmake

  Add the installation prefix of "camera_info_manager" to CMAKE_PREFIX_PATH
  or set "camera_info_manager_DIR" to a directory containing one of the above
  files.  If "camera_info_manager" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< spinnaker_camera_driver [0.87s, exited with code 1]

Summary: 2 packages finished [2.83s]
  1 package failed: spinnaker_camera_driver
  3 packages had stderr output: flir_camera_description flir_camera_msgs spinnaker_camera_driver

This is the output when I type colcon build command line. Seems like some dependencies are missing

berndpfrommer, 2023-10-30:
ROS galactic should come with the camera-info-manager package. Do you see it when you do "apt search"?

Chiwingum11, 2023-10-31:
libcamera-info-manager-dev/focal 1.11.13-6build2 amd64
  Robot OS camera_info_manager package - development

libcamera-info-manager0d/focal 1.11.13-6build2 amd64
  Robot OS camera_info_manager package

Seems like it's there. but for xacro, it is only available in noetic. Does it also come with ROS galactic or do I have to install it myself? 

ros-noetic-xacro/now 1.14.16-1focal.20230620.185428 amd64 [installed,local]
  Xacro (XML Macros) Xacro is an XML macro language.

ros-robot/focal,focal 1.15 all
  Python Robot OS robot metapackage

berndpfrommer, 2023-10-31:
xacro is part of the ROS2 galactic distro, but it may not be installed by default. See below what I get when apt searching. The package is there, you just need to install it. Usually this step: ``rosdep install --from-paths src --ignore-src`` (from the instructions) installs all required packages from the distro. Did you do run the rosdep install?

```
pfrommer@xavier:~$ apt search xacro | grep -i galactic

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-galactic-webots-ros2-importer/focal 2022.1.4-1focal.20221203.093345 arm64
ros-galactic-xacro/focal 2.0.7-1focal.20221203.075531 arm64
```

berndpfrommer, 2023-10-31:
Never mind, something is broken about rosdep. The rosdep database does not contain the xacro or camera info packages. You need to install manually via apt install.

Chiwingum11, 2023-10-31:
Works now after manually installing all the remaining dependencies. Thank You :)