ros-drivers/flir_camera_driver #139: ROS Iron

InvincibleRMC, 2023-11-21:
Hello there.

Would it be possible to create a ROS Iron release for this package as well as ROS Humble?
Running the CI on Iron for my Fork passed [here](https://github.com/InvincibleRMC/flir_camera_driver).

berndpfrommer, 2023-11-21:
Fine with me. @tonybaltovski do you have objections?

@tonybaltovski  I'm not quite clear on how you manage your tag space, and the CHANGELOG.rst
The way I do it: 
- master branch (in our case humble-devel) contains *no* CHANGELOG.rst.
- each distro has its own release branch with a CHANGELOG.rst.
- upon release I merge master into the release branch, update changelog and bloom.
- version numbering:
   2.0.x --> Rolling
   2.1.x --> Humble
   2.2.x ---> Iron
   2.3.x ----> Jazzy
At the moment we already have a CHANGELOG.rst in the humble-devel branch and humble releases are tagged 2.0.x.
Any suggestions?

tonybaltovski, 2023-11-23:
@berndpfrommer I generally keep the same release branch unless there is changes needed to create a separate branch.  However, this stems from ROS 1.  Your workflow makes sense to me.

berndpfrommer, 2023-11-24:
OK, I submitted a PR to rosdistro for having the driver under Iron.
It's kinda awkward because we are listing the humble-devel branch now as the source/doc branch under Iron. At some point in the future we can rename the branch to something like "ros2" or maybe "ros2-devel". Not necessary right now.
Next steps I will take (unless there are objections):
- remove the CHANGELOG.rst from humble-devel so it doesn't conflict with the release branches.
- create two new release branches: humble-release (with package numbers starting at 2.1.10) and iron-release (package numbers starting at `2.2.10.)`
- create new changelogs in the development branches and bloom to Humble and Iron from there.

InvincibleRMC, 2023-12-02:
Was added to Iron.

closed 2023-12-02 by InvincibleRMC

berndpfrommer, 2023-12-02:
Just released it today. Will take a few more days until the PR goes through and the package will actually show up in iron.