ros-drivers/flir_camera_driver #151: Cant build with ros2 humble on devel

geoeo, 2024-02-05:
When I try `colcon build`

I get 

```
Error processing idl file: /home/haubenstockm/git/ros2/flir_camera_driver/build/flir_camera_msgs/rosidl_adapter/flir_camera_msgs/msg/ImageMetaData.idl
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 40, in <module>
    sys.exit(main())
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 36, in main
    generate_py(args.generator_arguments_file, args.typesupport_impls.split(';'))
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/generate_py_impl.py", line 61, in generate_py
    generated_files = generate_files(generator_arguments_file, mapping)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_cmake/__init__.py", line 96, in generate_files
    raise(e)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_cmake/__init__.py", line 91, in generate_files
    post_process_callback=post_process_callback)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_cmake/__init__.py", line 131, in expand_template
    em.BUFFERED_OPT: True,
AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
gmake[2]: *** [flir_camera_msgs__py/CMakeFiles/flir_camera_msgs__py.dir/build.make:126: rosidl_generator_py/flir_camera_msgs/_flir_camera_msgs_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:590: flir_camera_msgs__py/CMakeFiles/flir_camera_msgs__py.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< flir_camera_msgs [0.39s, exited with code 2]
```

geoeo, 2024-02-05:
This can be fixed by
https://stackoverflow.com/questions/77642155/attributeerror-module-object-has-no-attribute-raw-opt/77656642#77656642

Now I have the error

```
--- stderr: flir_camera_msgs                             
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(_sre.o): warning: relocation against `PyExc_RecursionError' in read-only section `.text'
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(bytearrayobject.o): relocation R_X86_64_PC32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/flir_camera_msgs__rosidl_generator_py.dir/build.make:125: rosidl_generator_py/flir_camera_msgs/libflir_camera_msgs__rosidl_generator_py.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:475: CMakeFiles/flir_camera_msgs__rosidl_generator_py.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< flir_camera_msgs [2.38s, exited with code 2]
```

berndpfrommer, 2024-02-05:
Is there a reason you are not installing the package via apt from the rosdistro?
This repo [built fine](https://github.com/ros-drivers/flir_camera_driver/actions/runs/7576096121) about 3 weeks ago, and there have been no commits since. So something has changed about the environment.
Also, I see /usr/local/lib in your logs which indicates you have installed some custom python versions under /usr/local/ that could cause the problem. Can you eliminate that as a source of error?

geoeo, 2024-02-06:
@berndpfrommer Both my python and pip default to 3.10. And its still selecting 3.6. I really dont want to install from apt because I dont want to reinstall spinnaker binaries. Im not sure, but it seems your build chain assumes only 1 python version on the system?

berndpfrommer, 2024-02-06:
The build environment assumed is the one that the Ros build farm provides
which is a fresh install of Ubuntu 22.04. It looks like the build problems
already occur with the msgs package, which is about as boiler plate as it
gets.

You should be able to use the apt package alongside the Spinnaker install.
The driver should use its own Spinnaker library, installed under /opt/ros,
and not interfere with spinview. It's easy to try out.

geoeo, 2024-02-06:
Interesting. Ill try the apt then.
Thanks! @berndpfrommer any idea to whom I can escalate this issue?

berndpfrommer, 2024-02-06:
As far as the spinnaker_camera_driver goes I'm probably the most knowledgeable. This honestly doesn't seem to be a driver-specific issue. You can try posting on the robotics stack exchange but complex environment issues like this often get little attention.

There was definitely some issue with the "em" module, somewhere between colcon and Ubuntu, see [here](https://github.com/ros-tooling/setup-ros/pull/632).
If you can find some simple messages package that builds correctly you can try peeling away from flir_camera_msgs until it also builds.

Another option: install flir_camera_msgs from the apt repo, delete the source directory containing flir_camera_msgs from your workspace, and try just building the driver, if you really want to build from source.

geoeo, 2024-02-06:
Thanks. I also get this issue with other packages. It seems to be ros2. I will close this. Thanks for your help

closed 2024-02-06 by geoeo