ros-drivers/flir_camera_driver #162: Can't read any images from ROS2 topics

SangHunHan92, 2024-03-22:
**System details**
- operating system/version: Ubuntu 20.04 focal
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]): ROS2 galactic
- camera model: Flea3 GigE

**Is this behavior also observed when using SpinVIew?**
No, The images/videos shown through SpinVIew are very clear.

**Describe the bug**
    I have already installed "spinnaker-3.1.0.79-amd64-pkg" and "flycapture2-2.13.3.31-amd64-pkg_Ubuntu18.04".
   Also, during build ROS2 pkg using ```rosdep install --from-paths src --ignore-src``` and ```colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=ON```, no warnings or errors were detected.
But, If I run launch```ros2 launch spinnaker_camera_driver driver_node.launch.py```, I can't show any camera/image from ROS2 topic. I check this using "rviz2" and "ros2 topic hz".

Some causes I guess are camera model and Camera calibration file.
The camera model confirmed in the config is Blackfly S GigE, but the camera I use is Flea3 GigE. Can you tell me how to use it in Flea3 GigE? Also, how can I get the Camera calibration file?

**Useful logging output**
Please cut-and-paste console output into this bug report. Do not attach screenshots.
```
$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'20306011'" camera_name:=cam1
[INFO] [launch]: All log files can be found below /home/ros20/.ros/log/2024-03-22-15-35-43-478048-ros20-MS-7B89-11630
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [11632]
[camera_driver_node-1] [INFO] [1711089343.574080942] [cam1]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1711089343.574168628] [cam1]: looking for serial number: 20306011
[camera_driver_node-1] [INFO] [1711089343.574177688] [cam1]: parameter definitions file: /home/ros20/workspace/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[camera_driver_node-1] [INFO] [1711089343.575407802] [cam1]: using default calibration URL
[camera_driver_node-1] [INFO] [1711089343.575421202] [cam1]: camera calibration URL: file:///home/ros20/.ros/camera_info/cam1.yaml
[camera_driver_node-1] [ERROR] [1711089343.575448301] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ros20/.ros/camera_info/cam1.yaml]
[camera_driver_node-1] [WARN] [1711089343.575453780] [cam1]: Camera calibration file /home/ros20/.ros/camera_info/cam1.yaml not found
[camera_driver_node-1] [INFO] [1711089344.857822077] [cam1]: using spinnaker lib version: 3.1.0.79
[camera_driver_node-1] [INFO] [1711089344.858115294] [cam1]: found camera with serial number: 20306011
[camera_driver_node-1] [INFO] [1711089344.913566404] [cam1]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] [INFO] [1711089344.915479619] [cam1]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] [INFO] [1711089344.917322446] [cam1]: setting AcquisitionControl/AcquisitionFrameRateEnable to: 1
[camera_driver_node-1] [WARN] [1711089344.917392103] [cam1]: setting AcquisitionControl/AcquisitionFrameRateEnable failed: node AcquisitionControl/AcquisitionFrameRateEnable does not exist!
[camera_driver_node-1] [INFO] [1711089344.917427322] [cam1]: setting AcquisitionControl/AcquisitionFrameRate to: 30
[camera_driver_node-1] [WARN] [1711089344.918544461] [cam1]: setting AcquisitionControl/AcquisitionFrameRate failed: node AcquisitionControl/AcquisitionFrameRate not available!
[camera_driver_node-1] [INFO] [1711089344.918597639] [cam1]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1711089344.919450850] [cam1]: setting ChunkDataControl/ChunkModeActive to: 1
[camera_driver_node-1] [WARN] [1711089344.919505608] [cam1]: setting ChunkDataControl/ChunkModeActive failed: node ChunkDataControl/ChunkModeActive does not exist!
[camera_driver_node-1] [INFO] [1711089344.919538536] [cam1]: setting ChunkDataControl/ChunkSelector to: FrameID
[camera_driver_node-1] [WARN] [1711089344.919568535] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711089344.919578065] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: FrameID
[camera_driver_node-1] [INFO] [1711089344.919602264] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711089344.919645602] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711089344.919695329] [cam1]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[camera_driver_node-1] [WARN] [1711089344.919742527] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711089344.919754287] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: ExposureTime
[camera_driver_node-1] [INFO] [1711089344.919784215] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711089344.919832193] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711089344.919863342] [cam1]: setting ChunkDataControl/ChunkSelector to: Gain
[camera_driver_node-1] [WARN] [1711089344.919889821] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711089344.919898650] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Gain
[camera_driver_node-1] [INFO] [1711089344.919920929] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711089344.919945208] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711089344.919967227] [cam1]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] [WARN] [1711089344.919991416] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711089344.919999946] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Timestamp
[camera_driver_node-1] [INFO] [1711089344.920020795] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711089344.920044634] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711089344.927256720] [cam1]: camera has pixel format: BayerGB8
[camera_driver_node-1] [INFO] [1711089344.927309158] [cam1]: camera_driver_node started up!
[camera_driver_node-1] [INFO] [1711089348.574146518] [cam1]: rate [Hz] in  31.03 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711089353.574129732] [cam1]: rate [Hz] in  31.01 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711089358.574139210] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711089363.574141878] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
...
```
```
$ ros2 topic hz /cam1/image_raw
(show nothing)
```

berndpfrommer, 2024-03-22:
That you don't have a camera calibration file is *not the problem*. The error message can be ignored so long as you don't need the driver to publish a valid camera_info topic.
If you do need a calibration file, please use [this package](
https://navigation.ros.org/tutorials/docs/camera_calibration.html) in conjunction with the driver. Note that IMO the documentation for the calibration package is in poor shape and I cannot provide support for it.
From the driver log output it looks as if the driver is connecting to the camera and getting input (it reports incoming images at a frame rate of 31Hz. The most likely reason you are not seeing any images is because something is not working right with the ROS2 middleware (rmw), which is not directly related to this driver.
Couple of points to check:
- do you see the driver node when you run ``ros2 node list``
- what do you get when doing ``ros2 node <driver_node_name> info``? What topics is the node publishing?
- On the image topic, what do you see when you do ``ros2 topic info <name_of_image_topic>``?

SangHunHan92, 2024-03-22:
- Do you see the driver node when you run ```ros2 node list```
Yes. I can check valid info from ```ros2 node info```
```
/cam1
  Subscribers:
    /exposure_control/control: flir_camera_msgs/msg/CameraControl
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /cam_test/camera_info: sensor_msgs/msg/CameraInfo
    /cam_test/image_raw: sensor_msgs/msg/Image
    /cam_test/meta: flir_camera_msgs/msg/ImageMetaData
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /cam_test/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /cam_test/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /cam_test/get_parameters: rcl_interfaces/srv/GetParameters
    /cam_test/list_parameters: rcl_interfaces/srv/ListParameters
    /cam_test/set_parameters: rcl_interfaces/srv/SetParameters
    /cam_test/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
    /set_camera_info: sensor_msgs/srv/SetCameraInfo
  Service Clients:

  Action Servers:

  Action Clients:

```
- What do you get when doing ros2 node <driver_node_name> info? What topics is the node publishing?
Answer above.
- On the image topic, what do you see when you do ros2 topic info <name_of_image_topic>?
Sometimes a random ros2 daemon stop && ros2 daemon start clears up issues like that.
I command ```ros2 daemon stop && ros2 daemon start```, but the camera image is still not visible.

First of all, even if it is a Flea3 GigE Camera, does it have to work properly?

berndpfrommer, 2024-03-22:
Define "properly". You are running the config file for a blackfly_s, so probably many of the parameters will not work properly. You will need to create your own file for the flea type cameras. I'd greatly appreciate your PR for that since I do not own a flea camera.

But the config file is not the problem. You should definitely see something with ``ros2 topic hz``. Or ``ros2 topic echo /cam_test/image_raw``. The driver is clearly getting images from the camera.
Please restart your driver node, and your listener node. See if you can get the ROS2 talker/listener examples to work. Then *from the same terminals (shell)* launch the driver and do ``ros2 topic echo``.

SangHunHan92, 2024-03-22:
This seems to be a problem caused by the ROS2 driver not having appropriate settings for the Flea3 GigE camera.
I was lucky enough to get a Blackfly S GigE camera, and it works well without any problems.

So now I'm wondering what I need to modify code to make the Flea3 GigE camera work well.

```
$ ros20@ros20-MS-7B89:~/workspace/install/spinnaker_camera_driver/share/spinnaker_camera_driver/launch$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'20407111'" camera_name:=blackfly
[INFO] [launch]: All log files can be found below /home/ros20/.ros/log/2024-03-22-16-51-55-645147-ros20-MS-7B89-15659
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [15661]
[camera_driver_node-1] [INFO] [1711093915.739765173] [blackfly]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1711093915.739852663] [blackfly]: looking for serial number: 20407111
[camera_driver_node-1] [INFO] [1711093915.739861823] [blackfly]: parameter definitions file: /home/ros20/workspace/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[camera_driver_node-1] [INFO] [1711093915.741027423] [blackfly]: using default calibration URL
[camera_driver_node-1] [INFO] [1711093915.741040813] [blackfly]: camera calibration URL: file:///home/ros20/.ros/camera_info/blackfly.yaml
[camera_driver_node-1] [ERROR] [1711093915.741065833] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ros20/.ros/camera_info/blackfly.yaml]
[camera_driver_node-1] [WARN] [1711093915.741070813] [blackfly]: Camera calibration file /home/ros20/.ros/camera_info/blackfly.yaml not found
[camera_driver_node-1] [INFO] [1711093917.031390957] [blackfly]: using spinnaker lib version: 3.1.0.79
[camera_driver_node-1] [INFO] [1711093917.031664547] [blackfly]: found camera with serial number: 20407111
[camera_driver_node-1] [INFO] [1711093917.333895918] [blackfly]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] [INFO] [1711093917.340885195] [blackfly]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] [INFO] [1711093917.347811954] [blackfly]: setting AcquisitionControl/AcquisitionFrameRateEnable to: 1
[camera_driver_node-1] [INFO] [1711093917.351906842] [blackfly]: setting AcquisitionControl/AcquisitionFrameRate to: 30
[camera_driver_node-1] [INFO] [1711093917.359296260] [blackfly]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1711093917.362767818] [blackfly]: setting ChunkDataControl/ChunkModeActive to: 1
[camera_driver_node-1] [INFO] [1711093917.368112477] [blackfly]: setting ChunkDataControl/ChunkSelector to: FrameID
[camera_driver_node-1] [INFO] [1711093917.371762186] [blackfly]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1711093917.376229814] [blackfly]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[camera_driver_node-1] [INFO] [1711093917.377291394] [blackfly]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1711093917.380904914] [blackfly]: setting ChunkDataControl/ChunkSelector to: Gain
[camera_driver_node-1] [INFO] [1711093917.381947053] [blackfly]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1711093917.385558601] [blackfly]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] [INFO] [1711093917.386611681] [blackfly]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1711093917.410612073] [blackfly]: camera has pixel format: BayerRG8
[camera_driver_node-1] [INFO] [1711093917.410648633] [blackfly]: camera_driver_node started up!
[camera_driver_node-1] [INFO] [1711093920.739829535] [blackfly]: rate [Hz] in  30.05 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093925.739827243] [blackfly]: rate [Hz] in  30.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093930.739816473] [blackfly]: rate [Hz] in  29.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093935.739813486] [blackfly]: rate [Hz] in  29.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093940.739810245] [blackfly]: rate [Hz] in  29.99 out   7.40 drop   0%
[camera_driver_node-1] [INFO] [1711093945.739815599] [blackfly]: rate [Hz] in  29.99 out  30.00 drop   0%
[camera_driver_node-1] [INFO] [1711093950.739815957] [blackfly]: rate [Hz] in  29.99 out  19.60 drop   0%
[camera_driver_node-1] [INFO] [1711093955.739813635] [blackfly]: rate [Hz] in  29.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093960.739823051] [blackfly]: rate [Hz] in  29.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093965.739812564] [blackfly]: rate [Hz] in  29.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093970.739815049] [blackfly]: rate [Hz] in  29.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711093975.739814326] [blackfly]: rate [Hz] in  29.99 out  30.00 drop   0%
[camera_driver_node-1] [INFO] [1711093980.739813002] [blackfly]: rate [Hz] in  29.99 out  30.00 drop   0%
[camera_driver_node-1] [INFO] [1711093985.739812084] [blackfly]: rate [Hz] in  29.99 out  30.00 drop   0%
[camera_driver_node-1] [INFO] [1711093990.739815209] [blackfly]: rate [Hz] in  29.99 out  30.00 drop   0%
[camera_driver_node-1] [INFO] [1711093995.739815587] [blackfly]: rate [Hz] in  29.99 out  30.00 drop   0%
```

berndpfrommer, 2024-03-22:
I have a really hard time believing that. In your earlier log you can see a bunch of error messages that are caused because you were using the blackfly_s.yaml file, but then the driver says this:
```
[camera_driver_node-1] [INFO] [1711089344.927309158] [cam1]: camera_driver_node started up!
[camera_driver_node-1] [INFO] [1711089348.574146518] [cam1]: rate [Hz] in  31.03 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711089353.574129732] [cam1]: rate [Hz] in  31.01 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711089358.574139210] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711089363.574141878] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
```
That means it's getting images, and it should publish those. Once it gets images it should also publish them. Could you please repeat the experiment with the flea to make 100% sure that ``rostopic echo`` does not produce output?

berndpfrommer, 2024-03-22:
Instructions for making your own camera config file are [here](https://github.com/ros-drivers/flir_camera_driver/blob/humble-devel/spinnaker_camera_driver/docs/camera_configuration_files.md)

SangHunHan92, 2024-03-22:
> I have a really hard time believing that. In your earlier log you can see a bunch of error messages that are caused because you were using the blackfly_s.yaml file, but then the driver says this: [...]

Yes, I already did that, but ros2 topic echo shows nothing and hz also shows nothing.

```
$ ros20@ros20-MS-7B89:~/workspace/code$ ros2 topic echo /cam1/image_raw
(show nothing)
```
```
(after command "ros2 topic echo")
[camera_driver_node-1] [INFO] [1711097593.223064011] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711097598.223062704] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711097603.223061516] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1711097608.223063417] [cam1]: rate [Hz] in  31.00 out   0.00 drop   0%
```

berndpfrommer, 2024-03-22:
Thanks for checking!
One more last one: can you post the output of ``ros2 topic info /cam1/image_raw`` while you have both the ``ros2 topic echo`` and the driver running?
If that shows a publisher and subscriber, then it's not a ROS2 rmw issue.

Are you compiling from source? If yes, then can you re-instate [this printout](https://github.com/ros-drivers/flir_camera_driver/blob/33ab80728e3c3d2c6175a1f789fa11688ca65073/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp#L298) to see if that causes the images not to be published? It's the only alternative I can think of right now and it could be an issue with GigE cameras.

berndpfrommer, 2024-03-22:
BTW the "image incomplete failure" is usually due to the NIC MTU not set to 9000 and/or the camera transport MTU not set to 9000. The blackfly_s.yaml file defines parameters for setting the MTU on the camera, but from your driver logs it looks like you are not trying to set the MTU. It would likely fail anyways because you'd have to adapt the node name in the .yaml file to match what you see in SpinView.

berndpfrommer, 2024-03-25:
Ok so the topic info shows that there is indeed a subscriber to the topic, and if anything is being sent by the driver, it should receive it.
You were able to compile the driver from source, so modifying is no big deal. Use any editor you like to modify the file ``spinnaker_wrapper_impl.cpp`` [here](
https://github.com/ros-drivers/flir_camera_driver/blob/33ab80728e3c3d2c6175a1f789fa11688ca65073/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp#L296) such that it looks like this:

```
  if (imgPtr->IsIncomplete()) {
    numIncompleteImages_++;
#if 1
    // Retrieve and print the image status description
    std::cout << "Image incomplete: "
              << Spinnaker::Image::GetImageStatusDescription(imgPtr->GetImageStatus()) << std::endl;
#endif
  } else {
    std::cout << "Image complete!" << std::endl;
    float expTime = 0;
    float gain = 0;
```
and then build the workspace again using colcon.
If you see *any* printouts that say "Image complete" then you hopefully will also see images being published. If not, then none of your images arrive completely, i.e. too many packages are dropped on the network.
Not sure how you got the "Image incomplete" printouts without enabling the error message above. And how you concluded it isn't a big problem.

SangHunHan92, 2024-03-26:
```
ros20@ros20-MS-7B89:~/workspace$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'20306011'" camera_name:=cam1
[INFO] [launch]: All log files can be found below /home/ros20/.ros/log/2024-03-26-14-25-26-706959-ros20-MS-7B89-47305
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [47307]
[camera_driver_node-1] [INFO] [1711430726.797673886] [cam1]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1711430726.797762566] [cam1]: looking for serial number: 20306011
[camera_driver_node-1] [INFO] [1711430726.797772096] [cam1]: parameter definitions file: /home/ros20/workspace/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[camera_driver_node-1] [INFO] [1711430726.798959960] [cam1]: using default calibration URL
[camera_driver_node-1] [INFO] [1711430726.798973320] [cam1]: camera calibration URL: file:///home/ros20/.ros/camera_info/cam1.yaml
[camera_driver_node-1] [ERROR] [1711430726.798996491] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ros20/.ros/camera_info/cam1.yaml]
[camera_driver_node-1] [WARN] [1711430726.799001721] [cam1]: Camera calibration file /home/ros20/.ros/camera_info/cam1.yaml not found
[camera_driver_node-1] [INFO] [1711430728.091736411] [cam1]: using spinnaker lib version: 3.1.0.79
[camera_driver_node-1] [INFO] [1711430728.092019832] [cam1]: found camera with serial number: 20306011
[camera_driver_node-1] [INFO] [1711430728.146473866] [cam1]: setting AnalogControl/GainAuto to: Continuous
[camera_driver_node-1] [INFO] [1711430728.148289873] [cam1]: setting AcquisitionControl/ExposureAuto to: Continuous
[camera_driver_node-1] [INFO] [1711430728.150032850] [cam1]: setting AcquisitionControl/AcquisitionFrameRateEnable to: 1
[camera_driver_node-1] [WARN] [1711430728.150154160] [cam1]: setting AcquisitionControl/AcquisitionFrameRateEnable failed: node AcquisitionControl/AcquisitionFrameRateEnable does not exist!
[camera_driver_node-1] [INFO] [1711430728.150214680] [cam1]: setting AcquisitionControl/AcquisitionFrameRate to: 30
[camera_driver_node-1] [WARN] [1711430728.151328904] [cam1]: setting AcquisitionControl/AcquisitionFrameRate failed: node AcquisitionControl/AcquisitionFrameRate not available!
[camera_driver_node-1] [INFO] [1711430728.151396234] [cam1]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1711430728.152210967] [cam1]: setting ChunkDataControl/ChunkModeActive to: 1
[camera_driver_node-1] [WARN] [1711430728.152265397] [cam1]: setting ChunkDataControl/ChunkModeActive failed: node ChunkDataControl/ChunkModeActive does not exist!
[camera_driver_node-1] [INFO] [1711430728.152309027] [cam1]: setting ChunkDataControl/ChunkSelector to: FrameID
[camera_driver_node-1] [WARN] [1711430728.152356318] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711430728.152374438] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: FrameID
[camera_driver_node-1] [INFO] [1711430728.152416448] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711430728.152463968] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711430728.152516648] [cam1]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[camera_driver_node-1] [WARN] [1711430728.152562158] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711430728.152579498] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: ExposureTime
[camera_driver_node-1] [INFO] [1711430728.152618528] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711430728.152676539] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711430728.152718689] [cam1]: setting ChunkDataControl/ChunkSelector to: Gain
[camera_driver_node-1] [WARN] [1711430728.152763579] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711430728.152781319] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Gain
[camera_driver_node-1] [INFO] [1711430728.152819439] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711430728.152864369] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711430728.152906989] [cam1]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] [WARN] [1711430728.152953090] [cam1]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector does not exist!
[camera_driver_node-1] [WARN] [1711430728.152970810] [cam1]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Timestamp
[camera_driver_node-1] [INFO] [1711430728.153011120] [cam1]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [WARN] [1711430728.153058890] [cam1]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable does not exist!
[camera_driver_node-1] [INFO] [1711430728.219827581] [cam1]: camera has pixel format: BayerGB8
[... cut ...]
               << Spinnaker::Image::GetImageStatusDescription(imgPtr->GetImageStatus()) << std::endl;
   } else {
     std::cout << "Image complete!" << std::endl;
     const Spinnaker::ChunkData & chunk = imgPtr->GetChunkData();
     const float expTime = chunk.GetExposureTime();
     const float gain = chunk.GetGain();
     const int64_t stamp = chunk.GetTimestamp();
     const uint32_t maxExpTime =
       static_cast<uint32_t>(is_readable(exposureTimeNode_) ? exposureTimeNode_->GetMax() : 0);
```

What should I try next?

SangHunHan92, 2024-03-26:
The reason I concluded that it was not a big problem is because after installing the flir driver, there was intermittent data loss (about once every 10 seconds) when reading the camera through cv2 in python. Nevertheless, there was generally no problem reading the camera's image sequence. I think this is a problem with my settings, not a problem with the ROS2 driver.

berndpfrommer, 2024-03-26:
Are you still not seeing any images with e.g. rqt_image_view, although you are now getting "image complete" messages? That's the part I cannot understand.
My guess what's going on (or what *was* going on) is this: Your yaml file is incorrect because it's a new camera type (flea). That means the camera parameters, in particular the MTU is not set right. This in turn causes such severe packet loss that only incomplete images arrive at the ROS driver and hence it doesn't publish any. If you are still not seeing the driver publish images, despite it now printing out "image complete", then I'm out of ideas why it's not working.
I see you are still getting error messages on startup because you have not adjusted the blackfly_s.yaml file. I suggest you make a copy of it (call it flea.yaml). Then hack the launch file to point to that file. Look at the node names in SpinView, and modify your flea.yaml file correspondingly. Then you can set e.g. the frame rate, the MTU etc via ros parameters. No need to write code, you just need to edit the flea.yaml file until you get no error messages on startup. And you only have to touch the ROS parameters that you really need, like frame rate enabled, frame rate etc.

berndpfrommer, 2024-03-26:
Oh, and can you post the target link error message you are getting with the latest humble driver?

SangHunHan92, 2024-03-26:
Sure.

```
$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
Starting >>> flir_camera_msgs
Starting >>> rosbag2_tools                     
Starting >>> flir_camera_description                                              
Starting >>> sbg_driver                                                                                            
Finished <<< flir_camera_description [0.93s]                                                                                           
Finished <<< rosbag2_tools [1.63s]                                                                                         
Starting >>> ros2bag_tools
Finished <<< flir_camera_msgs [2.06s]                                                                                 
Finished <<< sbg_driver [1.85s]                                                     
Starting >>> spinnaker_camera_driver
Finished <<< ros2bag_tools [1.56s]                                                             
--- stderr: spinnaker_camera_driver                         
CMake Error at CMakeLists.txt:96 (add_library):
  Target "camera_driver" links to target "rclcpp_components::component" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:96 (add_library):
  Target "camera_driver" links to target
  "camera_info_manager::camera_info_manager" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at CMakeLists.txt:96 (add_library):
  Target "camera_driver" links to target "image_transport::image_transport"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:129 (add_executable):
  Target "camera_driver_node" links to target "rclcpp_components::component"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:129 (add_executable):
  Target "camera_driver_node" links to target
  "camera_info_manager::camera_info_manager" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at CMakeLists.txt:129 (add_executable):
  Target "camera_driver_node" links to target
  "image_transport::image_transport" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at CMakeLists.txt:96 (add_library):
  Target "camera_driver" links to target "rclcpp_components::component" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:96 (add_library):
  Target "camera_driver" links to target
  "camera_info_manager::camera_info_manager" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at CMakeLists.txt:96 (add_library):
  Target "camera_driver" links to target "image_transport::image_transport"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:460: cmake_check_build_system] Error 1
---
Failed   <<< spinnaker_camera_driver [1.26s, exited with code 2]

Summary: 5 packages finished [5.13s]
  1 package failed: spinnaker_camera_driver
  1 package had stderr output: spinnaker_camera_driver
  1 package not processed
```

berndpfrommer, 2024-03-26:
strange. It's not finding any of the basic ROS packages. Almost as if you hadn't sourced /opt/ros/galactic/setup.bash

I just built successfully on Ubuntu 22.04/Humble.
You may have to do a ``rm -rf build install log`` to clear out state from the existing build. Hard to imagine that the new version would not build on Ubuntu 20.04/Galactic.
```
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
Starting >>> flir_camera_msgs
Starting >>> flir_camera_description
--- stderr: flir_camera_description                                                                       
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_EXPORT_COMPILE_COMMANDS

---
Finished <<< flir_camera_description [1.13s]
Finished <<< flir_camera_msgs [5.11s]                     
Starting >>> spinnaker_camera_driver
[Processing: spinnaker_camera_driver]                         
Finished <<< spinnaker_camera_driver [52.2s]                       
Starting >>> spinnaker_synchronized_camera_driver
Finished <<< spinnaker_synchronized_camera_driver [11.0s]                           

Summary: 4 packages finished [1min 9s]
  1 package had stderr output: flir_camera_description

```

SangHunHan92, 2024-03-26:
I confirmed two facts through several experiments.

1. I modified ``spinnaker_wrapper_impl.cpp``` as below and got meaningful output from blackfly_s camera.
2. Even though I created a yaml file specifically for the flea camera, I still can't get the image.
```
### spinnaker_wrapper_impl.cpp ###
  if (imgPtr->IsIncomplete()) {
    // Retrieve and print the image status description
    std::cout << "Image incomplete: "
              << Spinnaker::Image::GetImageStatusDescription(imgPtr->GetImageStatus()) << std::endl;
  } else {
    // std::cout << "Image complete!" << std::endl;
    const Spinnaker::ChunkData & chunk = imgPtr->GetChunkData();
    const float expTime = chunk.GetExposureTime();
    const float gain = chunk.GetGain();
    const int64_t stamp = chunk.GetTimestamp();
    const uint32_t maxExpTime =
      static_cast<uint32_t>(is_readable(exposureTimeNode_) ? exposureTimeNode_->GetMax() : 0);

#if 1
    std::cout << "got image: " << imgPtr->GetWidth() << "x"
              << imgPtr->GetHeight() << " stride: " << imgPtr->GetStride()
              << " ts: " << stamp << " exp time: " << expTime
              << " gain: " << gain << " bpp: " << imgPtr->GetBitsPerPixel()
              << " chan: " << imgPtr->GetNumChannels()
              << " tl payload type: " << imgPtr->GetTLPayloadType()
              << " tl pix fmt: " << imgPtr->GetTLPixelFormat()
              << " payload type: " << imgPtr->GetPayloadType()
              << " pixfmt enum: " << imgPtr->GetPixelFormat()
              << " fmt: " << imgPtr->GetPixelFormatName()
              << " int type: " << imgPtr->GetPixelFormatIntType()
              << " frame id: " << imgPtr->GetFrameID()
              << " img id: " << imgPtr->GetID() << std::endl;
#endif
```
```
$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'20407111'" camera_name:=blackfly
[INFO] [launch]: All log files can be found below /home/ros20/.ros/log/2024-03-26-16-45-27-670663-ros20-MS-7B89-85098
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [85100]
[camera_driver_node-1] [INFO] [1711439127.759432862] [blackfly]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1711439127.759521882] [blackfly]: looking for serial number: 20407111
[camera_driver_node-1] [INFO] [1711439127.759530702] [blackfly]: parameter definitions file: /home/ros20/workspace/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
(setting blackfly_s config ...)
[camera_driver_node-1] [INFO] [1711439129.435888178] [blackfly]: camera has pixel format: BayerRG8
[camera_driver_node-1] [INFO] [1711439129.435931218] [blackfly]: camera_driver_node started up!
[camera_driver_node-1] got image: 2048x1536 stride: 2048 ts: 3169312330352 exp time: 7789 gain: 0 bpp: 8 chan: 1 tl payload type: 1 tl pix fmt: 17301513 payload type: 5 pixfmt enum: 4 fmt: BayerRG8 int type: 0 frame id: 1 img id: 93834223826448
[camera_driver_node-1] got image: 2048x1536 stride: 2048 ts: 3169345679752 exp time: 7789 gain: 0 bpp: 8 chan: 1 tl payload type: 1 tl pix fmt: 17301513 payload type: 5 pixfmt enum: 4 fmt: BayerRG8 int type: 0 frame id: 2 img id: 93834223826544
[camera_driver_node-1] got image: 2048x1536 stride: 2048 ts: 3169379029000 exp time: 7789 gain: 5.29926 bpp: 8 chan: 1 tl payload type: 1 tl pix fmt: 17301513 payload type: 5 pixfmt enum: 4 fmt: BayerRG8 int type: 0 frame id: 3 img id: 93834223826640
[camera_driver_node-1] got image: 2048x1536 stride: 2048 ts: 3169412378392 exp time: 10165 gain: 8.49917 bpp: 8 chan: 1 tl payload type: 1 tl pix fmt: 17301513 payload type: 5 pixfmt enum: 4 fmt: BayerRG8 int type: 0 frame id: 4 img id: 93834223826768
(...)
```
Even though I created a yaml file specifically for flea camera and no error occurred in the settings, I still cannot get images.
Below are my .yaml file and result.
```
# flea.yaml
parameters:
  #
  # -------- image format control
  #
  - name: pixel_format
    type: enum
    # Check available values with SpinView. Not all are supported by ROS!
    # Some formats are e.g. "Mono8", "BayerRG8", "BGR8", "BayerRG16"
    # default is "BayerRG8"
    node: ImageFormatControl/PixelFormat
  - name: image_width
    type: int
    node: ImageFormatControl/Width
  - name: image_height
[... 102 lines cut ...]
[camera_driver_node-1] [INFO] [1711440347.714952185] [cam1]: looking for serial number: 20306011
[camera_driver_node-1] [INFO] [1711440347.714961045] [cam1]: parameter definitions file: /home/ros20/workspace/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/flea.yaml
[camera_driver_node-1] [INFO] [1711440347.716122812] [cam1]: using default calibration URL
[camera_driver_node-1] [INFO] [1711440347.716137572] [cam1]: camera calibration URL: file:///home/ros20/.ros/camera_info/cam1.yaml
[camera_driver_node-1] [ERROR] [1711440347.716166822] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ros20/.ros/camera_info/cam1.yaml]
[camera_driver_node-1] [WARN] [1711440347.716172053] [cam1]: Camera calibration file /home/ros20/.ros/camera_info/cam1.yaml not found
[camera_driver_node-1] [INFO] [1711440349.031593765] [cam1]: using spinnaker lib version: 3.1.0.79
[camera_driver_node-1] [INFO] [1711440349.031922257] [cam1]: found camera with serial number: 20306011
[camera_driver_node-1] [INFO] [1711440349.085920231] [cam1]: setting ImageFormatControl/Width to: 1280
[... cut ...]
[camera_driver_node-1] [INFO] [1711440349.097261777] [cam1]: setting AcquisitionControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1711440349.105360206] [cam1]: camera has pixel format: BayerGB8
[camera_driver_node-1] [INFO] [1711440349.105403536] [cam1]: camera_driver_node started up!
[camera_driver_node-1] [INFO] [1711440352.714964561] [cam1]: rate [Hz] in  31.01 out   0.00 drop   0%
(...)
```

Therefore I determine that the value(e.f. IsIncomplete(), GetWidth(), ...) does not exist in ```imgPtr```.

berndpfrommer, 2024-03-26:
Great, you got a working yaml file!
I believe the reason you are seeing this problem is because chunkData is not enabled. This will cause an exception that causes the images to be dropped.
I did not catch this problem earlier because I was looking at the most recent source code, where this problem has been fixed. It is however a problem with the older version you are running (2.1.10). So you can try and look at the most recent source code and see the changes, in particular the exception handling [here](https://github.com/ros-drivers/flir_camera_driver/blob/9b63c19247cf122c80d38c7d8b6762b5dc3d9ecc/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp#L321C1-L333C6). There is a comment in the code about that.
The other alternative you have is to enable all Chunk Data (exposure, time stamp, frame number etc, all of it), if that is possible on the flea (should be).

There's something funky going on about the driver not building on galactic.
See also issue #164
I'm debugging it right now under PR #165

SangHunHan92, 2024-03-26:
Unfortunately, unlike the Blackfly S, the Flea3 does not have any options regarding Chunk Data.
Therefore, I don't know if it is possible to enable ChunkData.
I inevitably upload two photos for comparison.
This is why I removed the settings regarding Chunk from flea.yaml.

[image]
[image]

berndpfrommer, 2024-03-26:
Too bad. Can you try the latest driver I pushed under PR #165 ? It should build on galactic and fix the bugs you are seeing.

berndpfrommer, 2024-03-26:
never mind, still some bugs under galactic.

berndpfrommer, 2024-03-27:
The latest driver now builds under galactic. If you don't already have spinnaker installed it will download and compile Spinnaker v2.x which is fairly old. You can build against a more recent version of spinnaker by installing it (under /opt/spinnaker) before you compile the ros driver.
Please let me know if this works for you.

SangHunHan92, 2024-03-27:
I succeeded in building the ROS2 driver, but I cannot run it.
I use Spinnaker-2.2.0.48-Linux, and SpinView is working well.

```
$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
Starting >>> flir_camera_msgs
Starting >>> rosbag2_tools                     
Starting >>> flir_camera_description                                              
Starting >>> sbg_driver                                                                                            
Finished <<< rosbag2_tools [1.55s]                                                                                           
Starting >>> ros2bag_tools
Finished <<< flir_camera_description [1.82s]                                                                      
Finished <<< ros2bag_tools [1.96s]                                                                                        
Finished <<< flir_camera_msgs [4.85s]                                                               
Starting >>> spinnaker_camera_driver
Finished <<< spinnaker_camera_driver [24.8s]                                                                   
Starting >>> spinnaker_synchronized_camera_driver
Finished <<< spinnaker_synchronized_camera_driver [8.33s]
Finished <<< sbg_driver [49.7s]
Summary: 7 packages finished [52.0s]
```
```
$ ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'20407111'"
[INFO] [launch]: All log files can be found below /home/ros20/.ros/log/2024-03-27-20-25-21-951610-ros20-MS-7B89-1058654
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [1058656]
[camera_driver_node-1] /home/ros20/workspace2/install/spinnaker_camera_driver/lib/spinnaker_camera_driver/camera_driver_node: symbol lookup error: /home/ros20/workspace2/install/spinnaker_camera_driver/lib/libcamera_driver.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_
[ERROR] [camera_driver_node-1]: process has died [pid 1058656, exit code 127, cmd '/home/ros20/workspace2/install/spinnaker_camera_driver/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r __node:=flir_camera --params-file /tmp/launch_params_xxwhxmv9 --params-file /tmp/launch_params_45zy2yxb -r ~/control:=/exposure_control/control'].
```

berndpfrommer, 2024-03-27:
Hmm. Works in my Ubuntu 20.04/Galactic docker container (output below). Also works on my laptop (Ubuntu 22.04/Humble). Please double check that you have done a  ``rm -rf build log install`` at the top of your workspace. The error you are getting is really weird. Also double check that you sourced /opt/ros/galactic/setup.bash, and not something else. Maybe best to start a fresh shell.
```
pfrommer@galactic_dev:~/Documents/flir$ ros2 launch spinnaker_camera_driver driver_node.launch.py 
[INFO] [launch]: All log files can be found below /home/pfrommer/.ros/log/2024-03-27-12-43-18-060835-galactic_dev-16997
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [17003]
[camera_driver_node-1] [INFO] [1711543398.199997123] [20435008]: reading ros parameters for camera with serial: 20435008
[camera_driver_node-1] [INFO] [1711543398.200076952] [20435008]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1711543398.200264338] [20435008]: parameter definitions file: /home/pfrommer/Documents/flir/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[camera_driver_node-1] [INFO] [1711543398.202156290] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1711543398.202219497] [flir_camera]: camera calibration URL: file:///home/pfrommer/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [ERROR] [1711543398.202274742] [camera_calibration_parsers]: Unable to open camera calibration file [/home/pfrommer/.ros/camera_info/flir_camera.yaml]
[camera_driver_node-1] [WARN] [1711543398.202283892] [flir_camera]: Camera calibration file /home/pfrommer/.ros/camera_info/flir_camera.yaml not found
[camera_driver_node-1] [INFO] [1711543399.089992477] [20435008]: using spinnaker lib version: 2.2.0.48
[camera_driver_node-1] [WARN] [1711543399.090570630] [20435008]: no camera found with serial: 20435008 on try # 1
```

SangHunHan92, 2024-03-28:
How, incredible!
Both blackfly_s & flea camera are working well !! Thank you for your efforts.

Now I just want to solve 2 simple problem.
1. Both blackfly_s and flea camera show grayscale image captured with ROS2 driver. However, Spinview displays RGB images well. 
2. flea camera image taken with the ROS2 driver show subtle grid-structure noise. 

It seems like these can be solved easily. How can control these? 

[image]
blackfly_s with SpinView

[image]
blackfly_s with ROS2 driver

[image]
flea with SpinView

[image]
flea with ROS2 driver

berndpfrommer, 2024-03-28:
You are looking at a bayer image. Google bayer filters, and how color cameras work.
Try setting the ``pixel_format`` parameter for the blackfly_s to an allowed value, see [here](https://github.com/ros-drivers/flir_camera_driver/blob/d4173638eb91088dc1e52c760942fbd83566c1de/spinnaker_camera_driver/config/blackfly_s.yaml#L11). There could be more values allowed, you can explore those with spinview.
What viewer are you using? rqt_image_view? For me the viewer usually de-bayers the image automatically. Whatever, play around with ``pixel_format`` to see if it helps.
Next, figure out how to set the pixel format for flea (you will need to modify the flea.yaml file to introduce a pixel_format parameter). Would love to see a PR with your flea.yaml file once you have it working.

SangHunHan92, 2024-03-28:
I was able to show a color image after changing setting RGB8Packed in SpinView.
When I set the PixelFormat to various settings via .yaml alone before, I still only show the bayer image with rqt_image_view. I don't know why. 
_However, this is simply a visualization issue;_ there was no problem converting the Bayer image to RGB. Thanks !

flea.yaml was configured with minimal settings to avoid errors. I also need to modify driver_node.launch.py as well.
One thing to note, you should use "BayerGB8" as default and not "BayerRG8" for flea3 camera!

flea3.yaml
```
# 
# config file for Flea3 cameras (GigE)
#
# This file maps the ros parameters to the corresponding Spinnaker "nodes" in the camera.
# For more details on how to modify this file, see the README on camera configuration files.

parameters:
  #
  # -------- image format control
  #
  - name: pixel_format
    type: enum
    # Check available values with SpinView. Not all are supported by ROS!
    # Some formats are e.g. "Mono8", "Mono16", "Mono12Packed", "BayerGB8", "BayerGB12Packed", "BayerGB16", "RGB8Packed"
    # default is "BayerGB8"
    node: ImageFormatControl/PixelFormat
  - name: binning_x
    type: int
    node: ImageFormatControl/BinningHorizontal
  - name: binning_y
    type: int
    node: ImageFormatControl/BinningVertical
  - name: image_width
    type: int
    node: ImageFormatControl/Width
  - name: image_height
    type: int
    node: ImageFormatControl/Height
  - name: offset_x # offset must come after image width reduction!
    type: int
    node: ImageFormatControl/OffsetX
  - name: offset_y
    type: int
    node: ImageFormatControl/OffsetY

  #
  # -------- analog control
  #
  - name: gain_auto
    type: enum
    # valid values are "Continuous", "Off"
    node: AnalogControl/GainAuto
  - name: gain
    type: float
    node: AnalogControl/Gain
  - name: gamma_enabled
    type: bool
    node: AnalogControl/GammaEnabled
  #
  # -------- device link throughput limiting
  #
  - name: device_link_throughput_limit
    type: int
    node: DeviceControl/DeviceLinkThroughputLimit
  #
  # -------- transport layer control (GigE)
  #
  - name: gev_scps_packet_size
    type: int
    # default is 1400. Set to 9000 to enable jumbo frames, ensure NIC MTU set >= 9000
    node: TransportLayerControl/GigEVision/GevSCPSPacketSize
[... 215 lines cut ...]
                output='screen',
                name=[LaunchConfig('camera_name')],
                parameters=[example_parameters[camera_type],
                            {'ffmpeg_image_transport.encoding': 'hevc_nvenc',
                             'parameter_file': parameter_file,
                             'serial_number': [LaunchConfig('serial')]}],
                remappings=[('~/control', '/exposure_control/control'), ])

    return [node]

def generate_launch_description():
    """Create composable node by calling opaque function."""
    return LaunchDescription([
        LaunchArg('camera_name', default_value=['flir_camera'],
                  description='camera name (ros node name)'),
        LaunchArg('camera_type', default_value='blackfly_s',
                  description='type of camera (blackfly_s, chameleon...)'),
        LaunchArg('serial', default_value="'20435008'",
                  description='FLIR serial number of camera (in quotes!!)'),
        LaunchArg('parameter_file', default_value='',
                  description='path to ros parameter definition file (override camera type)'),
        OpaqueFunction(function=launch_setup)
        ])

```

berndpfrommer, 2024-03-28:
Thanks for posting. I will add the file to the repo.
Does the driver spit out any error messages when you set the pixel_format to RGB8Packed? Or does it simply not do anything?
If you do ``ros2 topic echo /image_topic_here | grep encoding``  you can see what encoding is published. Should no longer by bayer then.

SangHunHan92, 2024-03-29:
No, it doesn't spit out error messages depending on pixel_format. 
After setting RGB8Packed just once in SpinView, I can visualize the color image well for all types of pixel_format in ROS2. It seems that ```rqt_image_view``` simply does not display the Bayer image properly in RGB.
I think I confused you a bit.

```ros2 topic echo /image_topic | grep encoding``` also shows ```pixel_format``` properly what I set.

By the way, is there an option to publish topic such like ```/flir_camera/image_RGB``` instead of ```/flir_camera/image_raw``` ? When I record a topic ```image_raw``` published with Bayer, I have to convert it to RGB format. I would like to subscribe RGB images topic.

berndpfrommer, 2024-03-30:
I am indeed confused, but let's not further dwell on that :)

As far as the images go: by ROS  convention the topic name "image_raw" is used to indicate that the image is not rectified. If you want to publish a converted image in a different pixel format, the conventional way would be to start your conversion node in a "name space" such as /flir_camera/converted. Then the converted image would then be  /flir_camera/converted/image_raw (and the bayer image still available under /flir_camera/image_raw). Read up about name spaces, and how to specify them in the launch file.
I'd like to close this issue now as solved, since we are moving further and further away from the original issue.

SangHunHan92, 2024-04-01:
Thanks you !! The flea3 camera and build issue that I had originally concerned were resolved.
I'll close this issue.

closed 2024-04-01 by SangHunHan92