ros-drivers/flir_camera_driver #164: build problem on ros galactic

MichaelRaceUp, 2024-03-23:
**Describe the bug**
when I try to build the workspace, I get the following error in the CMake file of spinnaker_camera_driver:

CMake Error at CMakeLists.txt:102 (add_library):
  Target "camera_driver" links to target "rclcpp_components::component" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

The error occurs for the libraries:
rclcpp_components::component
camera_info_manager::camera_info_manager
image_transport::image_transport
rclcpp_components::component
camera_info_manager::camera_info_manager
image_transport::image_transport

I am currently working in Ubuntu 20.04 (which I cannot change), with Ros Galactic, but I also tried with foxy. What can I do to resolve those errors? Should I try to run it with Ros Humble (even though it is not fully supported by the specified version of Ubuntu)?

**Useful logging output**
Starting >>> spinnaker_camera_driver
--- stderr: spinnaker_camera_driver                         
CMake Error at CMakeLists.txt:102 (add_library):
  Target "camera_driver" links to target "rclcpp_components::component" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

**System details**
- operating system/version: Ubuntu 20.04
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]): [galactic]
- camera model: FLIR BFS-U319S4C-C

berndpfrommer, 2024-03-23:
> @berndpfrommer <https://github.com/berndpfrommer>.

Did you run rosdep to install all dependencies?

berndpfrommer, 2024-03-26:
Being addressed in PR #165

berndpfrommer, 2024-03-27:
Closed with #165 
Please reopen if still a problem

closed 2024-03-27 by berndpfrommer