ros-drivers/flir_camera_driver #172: Running A70 thermal camera

pilarvidalaguilar, 2024-04-16:
I am having trouble with launching the node using the A70 thermal camera. I am using ROS2 Humble.

I'm trying to use the parameter file from the blackfly_s with all the parameters commented on the launch but the pixel_format that I set it to Mono16 as that camera uses it. This is how it looks like:

[image]

I understand that even though it's not the specific parameter file, it should be working anyways with the default config.

Then, I try to launch the node using driver_node.launch.py and the node seems to start. However, the topics are not published. This is a screenshot of the terminal when launching and the ros2 topic list.

[image]

When I run the echo of the image_raw and camera_info topics, they are empty because the node does not publish anything.
Any clue of what may be going on?
The camera is working in spinview.

Thanks!

berndpfrommer, 2024-04-16:
There are just a handful of reasons why this could happen:
1) The images arrive at the camera, but are incomplete (only seen this for GigE cameras so far)
2) The "Chunk" settings are not available, or disabled (this has been fixed lately, but you could be running on an older version).
3) Topic mismatch: you are subscribing to the wrong topic. Unlikely, but please double check by doing ros2 topic info <your_topic>  and check with ros2 node info <your_node_name> that indeed the driver is publishing under that topic.

Would you be able to compile from source? If yes, then change the lines near [here](https://github.com/ros-drivers/flir_camera_driver/blob/83e9885d6431af0656af76d90e0b4bbf24662341/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp#L294)  reading "#if 0" to "#if 1" and rebuild. Then overlay your workspace on top of the installed driver to see if that sheds any light on the situation.

pilarvidalaguilar, 2024-04-16:
Thanks for the quick response.
I cloned the repo again as you said I might had an old version and it worked straight away! 
So thank you for your help

closed 2024-04-16 by berndpfrommer