ros-drivers/flir_camera_driver #178: [humble] Failed to start camera

Marius-N-R, 2024-05-21:
**Description**
The camera is recognized, but a connection can't be established. 
I installed from packages and do not use the Spinnaker SDK. I followed the **_Setting up Linux without Spinnaker SDK_** instructions. Also the ROS topics are visible but nothing is published. Do you have an idea what the issue could be? 

**Useful logging output**
`[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [273]`
`[camera_driver_node-1] [INFO] [1716301740.405509976] [12345678]: reading ros parameters for camera with serial: 12345678`
`[camera_driver_node-1] [INFO] [1716301740.406973736] [12345678]: adjusting time stamps!`
`[camera_driver_node-1] [INFO] [1716301740.412022704] [12345678]: parameter definitions file: /opt/ros/humble/share/spinnaker_camera_driver/config/blackfly_s.yaml`
`[camera_driver_node-1] [INFO] [1716301740.429860805] [flir_camera]: using default calibration URL`
`[camera_driver_node-1] [INFO] [1716301740.430803505] [flir_camera]: camera calibration URL: 
file:///root/.ros/camera_info/flir_camera.yaml`
`[camera_driver_node-1] [ERROR] [1716301740.431214194] [camera_calibration_parsers]: Unable to open camera calibration file [/root/.ros/camera_info/flir_camera.yaml]`
`[camera_driver_node-1] [WARN] [1716301740.431555949] [flir_camera]: Camera calibration file /root/.ros/camera_info/flir_camera.yaml not found`
`[camera_driver_node-1] [INFO] [1716301743.692180593] [12345678]: using spinnaker lib version: 3.1.0.79`
`[camera_driver_node-1] [INFO] [1716301743.693431650] [12345678]: found camera with serial number: 12345678`
`[camera_driver_node-1] [INFO] [1716301744.111673821] [12345678]: setting AnalogControl/GainAuto to: Continuous`
`[camera_driver_node-1] [WARN] [1716301744.114791525] [12345678]: setting AnalogControl/GainAuto failed: node AnalogControl/GainAuto not writable!`
`[camera_driver_node-1] [WARN] [1716301744.114824473] [12345678]: AnalogControl/GainAuto set to: UNKNOWN instead of: Continuous`
`[camera_driver_node-1] [INFO] [1716301744.118313296] [12345678]: setting AcquisitionControl/ExposureAuto to: Continuous`
`[camera_driver_node-1] [WARN] [1716301744.122855568] [12345678]: setting AcquisitionControl/ExposureAuto failed: node AcquisitionControl/ExposureAuto not writable!`
`[camera_driver_node-1] [WARN] [1716301744.124934566] [12345678]: AcquisitionControl/ExposureAuto set to: UNKNOWN instead of: Continuous`
`[camera_driver_node-1] [INFO] [1716301744.129429768] [12345678]: setting AcquisitionControl/AcquisitionFrameRateEnable to: 1`
`[camera_driver_node-1] [WARN] [1716301744.130316782] [12345678]: setting AcquisitionControl/AcquisitionFrameRateEnable failed: node AcquisitionControl/AcquisitionFrameRateEnable not writable!`
`[camera_driver_node-1] [INFO] [1716301744.131527096] [12345678]: setting AcquisitionControl/AcquisitionFrameRate to: 40`
`[camera_driver_node-1] [WARN] [1716301744.132468474] [12345678]: setting AcquisitionControl/AcquisitionFrameRate failed: node AcquisitionControl/AcquisitionFrameRate not writable!`
`[camera_driver_node-1] [INFO] [1716301744.134295561] [12345678]: setting AcquisitionControl/TriggerMode to: Off`
`[camera_driver_node-1] [WARN] [1716301744.135269373] [12345678]: setting AcquisitionControl/TriggerMode failed: node AcquisitionControl/TriggerMode not writable!`
`[camera_driver_node-1] [WARN] [1716301744.135294918] [12345678]: AcquisitionControl/TriggerMode set to: UNKNOWN instead of: Off`
`[camera_driver_node-1] [INFO] [1716301744.139341208] [12345678]: setting ChunkDataControl/ChunkModeActive to: 1`
`[camera_driver_node-1] [WARN] [1716301744.140318536] [12345678]: setting ChunkDataControl/ChunkModeActive failed: node ChunkDataControl/ChunkModeActive not writable!`
`[camera_driver_node-1] [INFO] [1716301744.141108264] [12345678]: setting ChunkDataControl/ChunkSelector to: FrameID`
`[camera_driver_node-1] [WARN] [1716301744.143561806] [12345678]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector not writable!`
`[camera_driver_node-1] [WARN] [1716301744.143587561] [12345678]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: FrameID`
`[camera_driver_node-1] [INFO] [1716301744.144393302] [12345678]: setting ChunkDataControl/ChunkEnable to: 1`
`[camera_driver_node-1] [WARN] [1716301744.145310358] [12345678]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable not writable!`
`[camera_driver_node-1] [INFO] [1716301744.146164066] [12345678]: setting ChunkDataControl/ChunkSelector to: ExposureTime`
`[camera_driver_node-1] [WARN] [1716301744.146327284] [12345678]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector not writable!`
`[camera_driver_node-1] [WARN] [1716301744.146339737] [12345678]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: ExposureTime`
`[camera_driver_node-1] [INFO] [1716301744.147247647] [12345678]: setting ChunkDataControl/ChunkEnable to: 1`
`[camera_driver_node-1] [WARN] [1716301744.147407363] [12345678]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable not writable!`
`[camera_driver_node-1] [INFO] [1716301744.148172332] [12345678]: setting ChunkDataControl/ChunkSelector to: Gain`
`[camera_driver_node-1] [WARN] [1716301744.148332944] [12345678]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector not writable!`
`[camera_driver_node-1] [WARN] [1716301744.148344555] [12345678]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Gain`
`[camera_driver_node-1] [INFO] [1716301744.149084353] [12345678]: setting ChunkDataControl/ChunkEnable to: 1`
`[camera_driver_node-1] [WARN] [1716301744.149245713] [12345678]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable not writable!`
`[camera_driver_node-1] [INFO] [1716301744.150140848] [12345678]: setting ChunkDataControl/ChunkSelector to: Timestamp`
`[camera_driver_node-1] [WARN] [1716301744.150353811] [12345678]: setting ChunkDataControl/ChunkSelector failed: node ChunkDataControl/ChunkSelector not writable!`
`[camera_driver_node-1] [WARN] [1716301744.150371116] [12345678]: ChunkDataControl/ChunkSelector set to: UNKNOWN instead of: Timestamp`
`[camera_driver_node-1] [INFO] [1716301744.151327324] [12345678]: setting ChunkDataControl/ChunkEnable to: 1`
`[camera_driver_node-1] [WARN] [1716301744.151492842] [12345678]: setting ChunkDataControl/ChunkEnable failed: node ChunkDataControl/ChunkEnable not writable!`
`[camera_driver_node-1] failed to switch on continuous acquisition!`
`[camera_driver_node-1] [ERROR] [1716301744.152237775] [12345678]: failed to start camera!`
`[camera_driver_node-1] [INFO] [1716301744.152266186] [flir_camera]: camera_driver_node started up!`
`[camera_driver_node-1] [INFO] [1716301745.402998482] [12345678]: rate [Hz] in   0.00 out   0.00 drop   0%`
`[camera_driver_node-1] [INFO] [1716301750.402980951] [12345678]: rate [Hz] in   0.00 out   0.00 drop   0%`
`[camera_driver_node-1] [INFO] [1716301755.402961241] [12345678]: rate [Hz] in   0.00 out   0.00 drop   0%`

**System details**
- operating system/version: Docker container, ros humble image (Ubuntu 22.04)
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]): humble
- camera model: Blackfly S, connected via a GigE connection

berndpfrommer, 2024-05-21:
> @berndpfrommer <https://github.com/berndpfrommer>.

Can you reboot the machine? Permission changes require logout/login

Marius-N-R, 2024-05-22:
Yes, I already rebooted the machine and restarted the docker container and I got the error above

berndpfrommer, 2024-05-22:
I've had the same issue once, too, but forgot how I resolved it. It is as if the camera refused to take any commands. This was with a blackfly (not S) on gigE.
Can you access the camera from e.g. a laptop with the Spinnaker SDK installed, and modify e.g. the frame rate using spinview?

Marius-N-R, 2024-05-22:
No, unfortunately I can only work from the docker on my Pc at the moment. I will download the spinnaker packages and see if I get it to work like that.

closed 2024-05-23 by Marius-N-R