ros-drivers/flir_camera_driver #180: sync camera cannot be acquisited

li-dahua, 2024-05-28:
**Describe the bug**
A clear and concise description of what the bug is.
WARNING: acquisition timeout, restarting!
It says is waiting for accurate frequency, current estimate: -1
**Useful logging output**
Please cut-and-paste console output into this bug report. Do not attach screenshots.
[INFO] [component_container-1]: process started with pid [13285]
[component_container-1] [INFO] [1716916704.121686510] [cam_sync_container]: Load Library: /home/ray/ws/install/spinnaker_synchronized_camera_driver/lib/libsynchronized_camera_driver.so
[component_container-1] [INFO] [1716916704.157493127] [cam_sync_container]: Found class: rclcpp_components::NodeFactoryTemplate<spinnaker_camera_driver::CameraDriver>
[component_container-1] [INFO] [1716916704.157540863] [cam_sync_container]: Found class: rclcpp_components::NodeFactoryTemplate<spinnaker_synchronized_camera_driver::SynchronizedCameraDriver>
[component_container-1] [INFO] [1716916704.157549679] [cam_sync_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<spinnaker_synchronized_camera_driver::SynchronizedCameraDriver>
[component_container-1] [INFO] [1716916704.160903854] [cam_sync]: created exposure controller: cam0.exposure_controller
[component_container-1] [INFO] [1716916704.161040417] [cam_sync]: created exposure controller: cam1.exposure_controller
[component_container-1] [INFO] [1716916704.161315175] [cam0]: parameter definitions file: /home/ray/ws/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[component_container-1] [INFO] [1716916704.163217344] [cam_sync]: camera calibration URL: file://camera_calibrations/22092307.yaml
[component_container-1] [ERROR] [1716916704.163250891] [cam_sync]: Invalid camera calibration URL: file://camera_calibrations/22092307.yaml
[component_container-1] [INFO] [1716916711.088088982] [cam0]: using spinnaker lib version: 3.1.0.79
[component_container-1] [INFO] [1716916711.088222707] [cam0]: found camera with serial number: 22092307
[component_container-1] [INFO] [1716916711.280220113] [cam0]: camera has pixel format: BayerRG8
[component_container-1] [INFO] [1716916711.280525042] [cam1]: parameter definitions file: /home/ray/ws/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/blackfly_s.yaml
[component_container-1] [INFO] [1716916711.281943109] [cam_sync]: camera calibration URL: file://camera_calibrations/22092309.yaml
[component_container-1] [ERROR] [1716916711.281963654] [cam_sync]: Invalid camera calibration URL: file://camera_calibrations/22092309.yaml
[component_container-1] [INFO] [1716916711.282429162] [cam1]: using spinnaker lib version: 3.1.0.79
[component_container-1] [INFO] [1716916711.282533139] [cam1]: found camera with serial number: 22092309
[component_container-1] [INFO] [1716916711.474881673] [cam1]: camera has pixel format: BayerRG8
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/cam_sync' in container '/cam_sync_container'
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] [INFO] [1716916716.475142633] [cam_sync]: waiting for accurate frequency, current estimate: -1
[component_container-1] WARNING: acquisition timeout, restarting!

**System details**
- operating system/version: 22.04
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]):humble
- camera model:Blacklfy s-U3-31S4C

berndpfrommer, 2024-05-29:
It's not getting any images. Questions:
1) Are you externally triggering the cameras with a sync pulse?
2) Are the cameras producing images when you use SpinView (spinnaker SDK)?
3) Can you get the cameras to work individually with the single-camera spinnaker_camera_driver in a similar configuration (i.e. hardware triggered)?

li-dahua, 2024-05-29:
1. Yes, I follow the hardware synchronize step for blackfly s camera from here.
https://www.flir.eu/support-center/iis/machine-vision/application-note/configuring-synchronized-capture-with-multiple-cameras/

2. Before I ran the code was able to capture the image. But after the code is run. the spinview is not able to capture images.

3. I was able to get a single camera node code running. But not sure if it is hardware triggered or not.

berndpfrommer, 2024-05-29:
1. You can either use an external trigger, or have one camera produce the sync pulse. Are you having one of the cameras produce the sync pulse?

It looks to me like maybe some of the launch parameters are not correct. I suggest you reset the cameras such that they work again with spinview. Make sure that indeed you are using the sync pulse, and the slave camera is not free running. Carefully note the parameters that are set on the two cameras, in particular the digital I/O settings and the trigger/trigger source settings.

In a master/slave configuration, you will need slightly different parameters for the two cameras, so the default launch files will not work out of the box, since they are assuming you are using an external trigger. You will need to adjust your launch file such that the two cameras have different launch parameters. The critical parameters are [the digital I/O parameters](https://github.com/ros-drivers/flir_camera_driver/blob/37bbc5f5c806230534bb5bf6c3e0f70d8dc76e06/spinnaker_camera_driver/config/blackfly_s.yaml#L74), e.g. pulling up the 3.3V for one camera, but not the other. Also [trigger mode](https://github.com/ros-drivers/flir_camera_driver/blob/37bbc5f5c806230534bb5bf6c3e0f70d8dc76e06/spinnaker_camera_driver/config/blackfly_s.yaml#L117) etc will need to be set differently for the two cameras (master has trigger mode OFF, slave has trigger mode ON). Please try that and post your launch file if you can no longer make progress.

As far as the error message when launching the stereo node from the single-camera driver: if you installed from apt packages, this may be due to a bug that I fixed recently. Please compile the driver from source if that is an option. The latest version should at least start up without error message.

li-dahua, 2024-05-29:
Hi, Thank you for helping. 
I still have bug of this after colon build from source file.

[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'sync' of type 'cam_sync_ros2::CamSync' in container '/stereo_camera_container': Could not find requested resource in ament index
[component_container-1] [ERROR] [1716949487.230000991] [stereo_camera_container]: Could not find requested resource in ament index
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'exposure_control' of type 'exposure_control_ros2::ExposureControl' in container '/stereo_camera_container': Could not find requested resource in ament index

**Here is my code.**
https://github.com/li-dahua/flir_ros_node/blob/54e724c1be7c78ff6e149ebb20eafa58413e7fa5/stereo_synced.launch.py

berndpfrommer, 2024-05-29:
The problem is that the launch file is referencing other packages that are outside the flir_camera_driver repository. If you remove the [offending nodes](https://github.com/li-dahua/flir_ros_node/blob/54e724c1be7c78ff6e149ebb20eafa58413e7fa5/stereo_synced.launch.py#L127C1-L146C15) from the launch file, the driver should start up.
The two missing nodes are [exposure control](https://github.com/berndpfrommer/exposure_control_ros2) and [cam_sync](https://github.com/berndpfrommer/cam_sync_ros2)
I strongly discourage you from using these two because they are outdated. The spinnaker_synchronized_camera_driver does both exposure control and synchronization now, and does it much better. So I recommend you remove the two nodes from the launch file, and see what happens. If the two cameras start up then you know your configuration is working, and you can transfer that config to the launch file for the synchronized driver.

li-dahua, 2024-05-29:
I managed to run with these 2 changes, even still with these 2 errors.
'gain_auto': 'Continuous',
'exposure_auto': 'Continuous',

berndpfrommer, 2024-05-29:
I do not understand what you mean with "these 2 errors".  The ones from your last log should be gone if you remove the nodes.

li-dahua, 2024-05-29:
I able to get image from the cam with the two missing nodes are missing.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/left_flir_camera' in container '/stereo_camera_container'
[component_container-1] [ERROR] [1717012095.725982621] [stereo_camera_container]: Could not find requested resource in ament index
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'sync' of type 'cam_sync_ros2::CamSync' in container '/stereo_camera_container': Could not find requested resource in ament index
[component_container-1] [ERROR] [1717012095.730392953] [stereo_camera_container]: Could not find requested resource in ament index
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'exposure_control' of type 'exposure_control_ros2::ExposureControl' in container '/stereo_camera_container': Could not find requested resource in ament index
[component_container-1] [INFO] [1717012098.709500121] [22092309]: rate [Hz] in  38.60 out   0.00 drop   0%
[component_container-1] [INFO] [1717012100.454515811] [22092307]: rate [Hz] in  39.32 out   0.00 drop   0%
[component_container-1] [INFO] [1717012103.709479580] [22092309]: rate [Hz] in  38.50 out   0.00 drop   0%

berndpfrommer, 2024-05-29:
OK, so you didn't remove the nodes, but the camera driver still runs. You will find that the image header stamps (header.stamp) of the ros image messages will not have identical time stamps. This is because the sync_node is missing, which forces the time stamps to be the same.
If you now transfer the camera parameter settings to the spinnaker_synchronized_camera_driver launch file you should be able to use that driver. It uses the same underlying code as the one you got to work just now.

li-dahua, 2024-05-29:
I transfer the setting to follower code.
https://github.com/li-dahua/flir_ros_node/blob/3e26e1dee8808ff853e2fdd30541ec259cf3eac4/follower_example.launch.py
But I get some delays between the outputs.

[component_container-1] [INFO] [1717020266.975855148] [cam1]: camera has pixel format: BGR8
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/cam_sync' in container '/cam_sync_container'
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] [INFO] [1717020271.976126141] [cam_sync]: ------ frequency:     38.633 Hz
[component_container-1] [INFO] [1717020271.976194367] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1717020271.976206594] [cam_sync]: cam0        0    0    -0.03%     4.40%
[component_container-1] [INFO] [1717020271.976219615] [cam_sync]: cam1        0    0     0.00%     0.00%
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] WARNING: acquisition timeout, restarting!
^[a[component_container-1] WARNING: acquisition timeout, restarting!
[component_container-1] [INFO] [1717020276.976156268] [cam_sync]: ------ frequency:     38.429 Hz
[component_container-1] [INFO] [1717020276.976290528] [cam_sync]: camera   drop icmp    offset    jitter
[component_container-1] [INFO] [1717020276.976331665] [cam_sync]: cam0        0    0    -0.04%     5.21%
[component_container-1] [INFO] [1717020276.976375624] [cam_sync]: cam1        0    0     0.00%     0.00%

berndpfrommer, 2024-05-29:
OK, so camera 0 (master) is working, but camera 1 is not producing images. You will have to experiment to see what the problem is. Ideally you'd have an oscilloscope to see if the master is generating a pulse signal. That way you know which one to focus on first.
In the absence of an oscilloscope:
- get both cameras working under spinview again
- run the secondary camera in spinview
- run the *primary only* with the launch file above by commenting out the secondary camera.
- does it work then? If yes, then the problem is with the secondary, if no, it is with the primary.

li-dahua, 2024-05-29:
Solved, really thank you for helping.
It is a hardware wire issue.

berndpfrommer, 2024-05-29:
Could you please post a link to the final version of the launch file you were using? I'd like to add it to the repository. Alternatively, you can file a pull request.

li-dahua, 2024-05-30:
Here is my final launch file.
https://github.com/li-dahua/flir_ros_node.git

closed 2024-06-02 by li-dahua