ros-drivers/flir_camera_driver #186: A700 ros2 humble config

HBSIM0, 2024-06-20:
**System details**
- operating system/version: ubuntu22.04
- ROS version : ros2 humble
- camera model: A700

Can I also use the A700 thermal imaging camera with this package? Can someone provide me the A700 yaml information

berndpfrommer, 2024-06-20:
It should work, since the spinnaker SDK is supported according to FLIR's web site. However, you will have to write your own yaml file. Not too many people have a $12k camera. Writing a file is not that hard, instructions are [here](https://github.com/ros-drivers/flir_camera_driver/tree/humble-devel/spinnaker_camera_driver#how-to-develop-your-own-camera-configuration-file). Start with a blackfly_s config file and fix what is broken. Please report back if you had success, or even better, post a working config file or submit a pull request with a working config.

HBSIM0, 2024-06-28:
I tried working after fixing the confiure file and the thermal camera looks like a black and white screen. Can I get some advice?
[image]

my config is below
     'a700': {
        'debug': False,
        'compute_brightness': False,
        'adjust_timestamp': True,
        'dump_node_map': False,
        # set parameters defined in blackfly_s.yaml
        'gain_auto': 'Continuous',
        'pixel_format': 'Mono16',
        'exposure_auto': 'Continuous',
        # to use a user set, do this:
        'user_set_selector': 'UserSet0',
        'user_set_load': 'Yes',
        # These are useful for GigE cameras
        # 'device_link_throughput_limit': 380000000,
        # 'gev_scps_packet_size': 9000,
        # ---- to reduce the sensor width and shift the crop
        # 'image_width': 1408,
        # 'image_height': 1080,
        # 'offset_x': 16,
        # 'offset_y': 0,
        # 'binning_x': 1,
        # 'binning_y': 1,
        # 'connect_while_subscribed': True,
        'frame_rate_auto': 'Off',
        'frame_rate': 40.0,
        'frame_rate_enable': True,
        'buffer_queue_size': 10,
        'trigger_mode': 'Off',
        'chunk_mode_active': True,
        'chunk_selector_frame_id': 'FrameID',
        'chunk_enable_frame_id': True,
        'chunk_selector_exposure_time': 'ExposureTime',
        'chunk_enable_exposure_time': True,
        'chunk_selector_gain': 'Gain',
        'chunk_enable_gain': True,
        'chunk_selector_timestamp': 'Timestamp',
        'chunk_enable_timestamp': True,
    },

berndpfrommer, 2024-06-28:
Congratulations, you got an image, and the image looks reasonable to me. I suppose increased image brightness is related to higher temperature.
The ROS2 driver does not provide a false color image like the hand-held FLIR cameras do. If you want that you will need to write a node that subscribes to the mono image, and maps grey value to color.

berndpfrommer, 2024-06-30:
Can you post your .yaml file (or submit a PR with it)

HBSIM0, 2024-06-30:
Is it not possible for ros2 driver to show the color of thermal mode?
Does it have to be implemented by the user?
a700 supports thermal mode and digital mode.
Please refer to the picture below that you ran on the webpage.
[image]
[image]

a700.yaml

#
# config file for blackfly S cameras (USB3 and GigE)
#
# This file maps the ros parameters to the corresponding Spinnaker "nodes" in the camera.
# For more details on how to modify this file, see the README on camera configuration files.

parameters:
  #
  # -------- user set control
  #
  - name: default_user_set
    type: enum
    # allowed values: UserSet0 UserSet1
    node: UserSetControl/DefaultUserSet
  - name: user_set_selector
    type: enum
    # allowed values: UserSet0 UserSet1
    node: UserSetControl/UserSetSelector
  - name: user_set_load
    type: command
    node: UserSetControl/UserSetLoad
  #
  # -------- image format control
  #
  - name: pixel_format
    type: enum
    # Check available values with SpinView. Not all are supported by ROS!
    # Some formats are e.g. "Mono8", "BayerRG8", "BGR8", "BayerRG16"
    # default is "BayerRG8"
    node: ImageFormatControl/PixelFormat
  - name: binning_x
    type: int
    node: ImageFormatControl/BinningHorizontal
  - name: binning_y
    type: int
    node: ImageFormatControl/BinningVertical
  - name: image_width
    type: int
    node: ImageFormatControl/Width
  - name: image_height
    type: int
    node: ImageFormatControl/Height
  - name: offset_x # offset must come after image width reduction!
    type: int
    node: ImageFormatControl/OffsetX
  - name: offset_y
    type: int
    node: ImageFormatControl/OffsetY
  - name: image_mode
    type: enum
    # Check available values with SpinView. Not all are supported by ROS!
    # Some formats are e.g. "Mono8", "BayerRG8", "BayerRG16", "YUV444Packed"
    # default is "BayerRG8"
    node: ImageFormatControl/ImageMode

  #
  # -------- analog control
  #
  - name: gain_auto
    type: enum
    # valid values are "Continuous", "Off"
    node: AnalogControl/GainAuto
  - name: gain
    type: float
    node: AnalogControl/Gain
  #
  # -------- device link throughput limiting
  #
  - name: device_link_throughput_limit
    type: int
    node: DeviceControl/DeviceLinkThroughputLimit
  #
  # -------- transport layer control (GigE)
  #
  - name: gev_scps_packet_size
    type: int
    # default is 1400. Set to 9000 to enable jumbo frames, ensure NIC MTU set >= 9000
    node: TransportLayerControl/GigEVision/GevSCPSPacketSize
  #
  # -------- digital IO control
  #
  - name: line0_selector # black wire: opto-isolated input
    type: enum
    node: DigitalIOControl/LineSelector
  - name: line1_selector # white wire: opto-isolated output
    type: enum
    node: DigitalIOControl/LineSelector
  - name: line1_linemode # valid values: "Input", "Output"
    type: enum
    node: DigitalIOControl/LineMode
  - name: line2_selector # red wire: non-isolated input/output
    type: enum
    node: DigitalIOControl/LineSelector
  - name: line2_v33enable # red wire: 3.3V power
    type: bool
    node: DigitalIOControl/V3_3Enable
  - name: line3_selector # green wire: aux voltage input and non-isolated input
    type: enum
    node: DigitalIOControl/LineSelector
  - name: line3_linemode # valid values: "Input", "Output"
    type: enum
    node: DigitalIOControl/LineMode
  #
  # -------- acquisition control
  #
  - name: exposure_auto
    type: enum
    # valid values are "Off", "Continuous"
    node: AcquisitionControl/ExposureAuto
  - name: exposure_time
    type: float
    node: AcquisitionControl/ExposureTime
  - name: frame_rate_enable
    type: bool
    node: AcquisitionControl/AcquisitionFrameRateEnable
  - name: frame_rate
    type: float
    node: AcquisitionControl/AcquisitionFrameRate
  - name: trigger_selector
    type: enum
    # valid values are e.g. "FrameStart", "AcquisitionStart", "FrameBurstStart"
    node: AcquisitionControl/TriggerSelector
  - name: trigger_mode
    type: enum
    # valid values are "On" and "Off"
    node: AcquisitionControl/TriggerMode
  - name: trigger_source
    type: enum
    # valid values are "Line<0,1,2>", "UserOutput<0,1,2>", "Counter<0,1><Start/End>",
    # "LogicBlock<0,1>
    node: AcquisitionControl/TriggerSource
  - name: trigger_delay
    # value >= 9
    type: float
    node: AcquisitionControl/TriggerDelay
  - name: trigger_overlap
    type: enum
    # valid values: "Off" and "ReadOut"
    node: AcquisitionControl/TriggerOverlap
  - name: trigger_activation
    type: enum
    # valid values: "LevelLow", "LevelHigh", "FallingEdge", "RisingEdge", "AnyEdge",
    node: AcquisitionControl/TriggerActivation
  #
  # --------- chunk control
  #
  - name: chunk_mode_active
    type: bool
    node: ChunkDataControl/ChunkModeActive
  - name: chunk_selector_frame_id
    type: enum
    # valid values: "FrameID"
    node: ChunkDataControl/ChunkSelector
  - name: chunk_enable_frame_id
    type: bool
    node: ChunkDataControl/ChunkEnable
  - name: chunk_selector_exposure_time
    type: enum
    # valid values: "ExposureTime"
    node: ChunkDataControl/ChunkSelector
  - name: chunk_enable_exposure_time
    type: bool
    node: ChunkDataControl/ChunkEnable
  - name: chunk_selector_gain
    type: enum
    # valid values: "Gain"
    node: ChunkDataControl/ChunkSelector
  - name: chunk_enable_gain
    type: bool
    node: ChunkDataControl/ChunkEnable
  - name: chunk_selector_timestamp
    type: enum
    # valid values: "Timestamp"
    node: ChunkDataControl/ChunkSelector
  - name: chunk_enable_timestamp
    type: bool
    node: ChunkDataControl/ChunkEnable

berndpfrommer, 2024-07-01:
You are using some web software that I'm not familiar with. Can you switch between Thermal and Color (RGB) image in SpinView as well?
But I'm not sure if you are even interested in the color (RGB) image, or if you just want the colorized thermal image shown in the second picture you posted. There are color controls at the side of the window. The question now is: is the colorization done by the web application you are using, or is the colorization done already on the camera itself? Does SpinView offer any parameters (more precisely: GenICam nodes) to change the coloration? If yes, then we may be able to get the color image via the ROS driver, if not, then you may have to write a greyscale -> pseudocolor conversion node yourself.

berndpfrommer, 2024-07-17:
Closing this issue as there is no response.

closed 2024-07-17 by berndpfrommer