ros-drivers/flir_camera_driver #189: [jazzy] Jazzy release

InvincibleRMC, 2024-07-03:
**Describe the bug**
With https://github.com/ros-drivers/flir_camera_driver/issues/167 Being completed. Is there anything blocking a ros-jazzy release?

**System details**
- ROS version: ros-jazzy

berndpfrommer, 2024-07-03:
It's there on my Ubuntu 24.04 system. Can you please check?
```
apt search ros-jazzy-flir
Sorting... Done
Full Text Search... Done
ros-jazzy-flir-camera-description/noble 2.0.15-2noble.20240514.023209 amd64
  FLIR camera Description package

ros-jazzy-flir-camera-msgs/noble 2.0.15-2noble.20240514.002003 amd64
  messages related to flir camera driver

ros-jazzy-flir-camera-msgs-dbgsym/noble 2.0.15-2noble.20240514.002003 amd64
  debug symbols for ros-jazzy-flir-camera-msgs

```

InvincibleRMC, 2024-07-03:
Has `spinnaker_camera_driver` moved somewhere else? I was using that package and it doesn't seem to be in your list either?

InvincibleRMC, 2024-07-03:
I do see the other three packages however.

berndpfrommer, 2024-07-03:
oops, sorry, didn't look carefully. Let me check.

berndpfrommer, 2024-07-03:
Not sure what's going on. The last build on Jul 1st seemed successful, but the last message says something about pending-description. Weird.

https://build.ros2.org/job/Jbin_uN64__spinnaker_camera_driver__ubuntu_noble_amd64__binary/67/

It could also be that we'll have to wait for the next sync to show up. There was a sync scheduled for June 28th but that one has still not happened AFAIK:
https://discourse.ros.org/t/preparing-for-jazzy-sync-and-patch-release-2024-06-28/38340

At any rate, I have released the packages to rosdistro, it should be just a question of time until the package shows up.
Note that you won't be able to use SpinView on Ubuntu 24.04 since FLIR has not released an official SDK yet.

berndpfrommer, 2024-07-05:
Sync just happened. I tested with my USB3 cams on jazzy and the driver gets images.
I had to set this though:
```
export SPINNAKER_GENTL64_CTI=/opt/ros/jazzy/lib/spinnaker-gentl/Spinnaker_GenTL.cti
```

closed 2024-07-05 by berndpfrommer