ros-drivers/flir_camera_driver #192: Trying to configurate the stream buffer handling mode

Cyraneon, 2024-07-05:
Good day to you.

As written in the title, I am trying to configurate the Stream Buffer Handling Mode to "newest only".
This Node can be found in SpinView in the following submenus of Feature : Stream Parameters/Buffer Handling Control/Stream Buffer Handling Node.

I am trying to access to this Node by adding to the standard yaml configuration file the following lines : 
```yaml
  - name: buffer_handling_control
    type: enum
    node: BufferHandlingControl/StreamBufferHandlingMode
```
And by modifying the launch file by adding the following line in the parameters : 
```python
'buffer_handling_control': 'Newest Only',
```

Yet when launching the node, I obtain the following error : 

```
[camera_driver_node-1] [INFO] [1720183684.466130590] [23574899]: setting StreamParameters/BufferHandlingControl/StreamBufferHandlingMode to: Newest Only
[camera_driver_node-1] [WARN] [1720183684.466213600] [23574899]: setting StreamParameters/BufferHandlingControl/StreamBufferHandlingMode failed: node StreamParameters/BufferHandlingControl/StreamBufferHandlingMode not found!
[camera_driver_node-1] [WARN] [1720183684.466229056] [23574899]: StreamParameters/BufferHandlingControl/StreamBufferHandlingMode set to: UNKNOWN instead of: Newest Only
```

It seems I failed to access to the Node I want to configure. So I tried to set `dump_node_map` to True in my launch file, to check the name of this Node, and I was unable to find it.

My question is thus the following : is there a way to configure the Stream Parameters Nodes ? And if it exist, how does it work?

Thanks in advance for your feeback ! 

**System details**
- Ubuntu/20.04:
- ROS2 humble
- Blackflys3-19S4

berndpfrommer, 2024-07-05:
A) Do you see the node in SpinView?
B) If yes, you can double-click on the node and get its name as described [here](https://github.com/ros-drivers/flir_camera_driver/tree/humble-devel/spinnaker_camera_driver#how-to-develop-your-own-camera-configuration-file)?

berndpfrommer, 2024-07-05:
I was able to reproduce the issue on my Blackfly GigE camera. It turns out the nodes you can control are only the ones that show up under the camera itself in SpinView, e.g. under Blackfly xxx. The SDK does not show the Transport Layer or the Stream Parameters in the root node.

berndpfrommer, 2024-07-08:
Yes, the parameters are accessible from the SDK, but work very differently from the camera parameters. I can implement it but it will  take me a couple of days to do.

berndpfrommer, 2024-07-08:
Never mind, it was a quick fix. Please check out the stream_transport_layer branch from this PR:

https://github.com/ros-drivers/flir_camera_driver/pull/193

It found the nodes for my blackfly GigE cameras, but I could not confirm if setting the parameter makes any difference.

berndpfrommer, 2024-07-08:
oops, merged into humble-devel already....

Cyraneon, 2024-07-10:
I pulled the changes you made on the humble-devel branch and can confirm that the camera is now behaving as intended.
I am streaming the images coming from a BlackflyS3 19S4 and can see that there is now little to no delay between the capture of an image and its display on my screen.

For a s far as I am concerned, this issue has been properly fixed in a timely manner.

Thanks for your help and quick reaction !

closed 2024-07-10 by Cyraneon