ros-drivers/flir_camera_driver #197: [jazzy] Cannot connect to GigE camera

InvincibleRMC, 2024-07-20:
**Describe the bug**
The node starts but the camera fails to connect. I tested it with the config/launch that worked in ROS iron as well as the current example's from this library.

**Useful logging output**
Please cut-and-paste console output into this bug report. Do not attach screenshots.

```
[INFO] [launch]: All log files can be found below /home/rov/.ros/log/2024-07-20-13-17-06-707399-rov-P55V4-10065
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [10068]
[camera_driver_node-1] [INFO] [1721495826.836025713] [23473566]: reading ros parameters for camera with serial: 23473566
[camera_driver_node-1] [INFO] [1721495826.836086463] [23473566]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1721495826.836176290] [23473566]: parameter definitions file: ~/rov-25/src/surface/rov_flir/config/blackfly_s.yaml
[camera_driver_node-1] [ERROR] [1721495826.836231590] [23473566]: error reading parameter definitions: bad file: ~/rov-25/src/surface/rov_flir/config/blackfly_s.yaml
[camera_driver_node-1] [ERROR] [1721495826.836268512] [flir_camera]: startup failed!
[camera_driver_node-1] [INFO] [1721495826.836307431] [flir_camera]: camera_driver_node started up!
[camera_driver_node-1] [WARN] [1721495831.836170167] [23473566]: camera is not online!
[camera_driver_node-1] [WARN] [1721495836.836188742] [23473566]: camera is not online!
[camera_driver_node-1] [WARN] [1721495841.836180248] [23473566]: camera is not online!
[camera_driver_node-1] [WARN] [1721495846.836195137] [23473566]: camera is not online!
```

```bash
rov@rov-P55V4:~/rov-25$ echo $SPINNAKER_GENTL64_CTI 
/opt/ros/jazzy/lib/spinnaker-gentl/Spinnaker_GenTL.cti
```

```bash
rov@rov-P55V4:~/rov-25$ ls /opt/ros/jazzy/lib/spinnaker-gentl/
log4cpp.gentl.property  Spinnaker_GenTL.cti
```

I'm not sure if this is related but when testing without a config it attempts to load a `/home/rov/.ros/camera_info/flir_camera.yaml` which cannot be found since it is not installed.

**System details**
- operating system/version:
Ubuntu 24
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]):
ROS Jazzy
- complete FLIR camera model (e.g. BFS-U3-04S2C-C):
Blackfly S GigE (BFS-PGE-16S2C-CS)

berndpfrommer, 2024-07-20:
Looks like it's not finding the camera definition file:
```
[camera_driver_node-1] [ERROR] [1721495826.836231590] [23473566]: error reading parameter definitions: bad file: ~/rov-25/src/surface/rov_flir/config/blackfly_s.yaml
[camera_driver_node-1] [ERROR] [1721495826.836268512] [flir_camera]: startup failed!
```
I don't think it can handle ``~``. Try using the full path, e.g. ``/home/rov24/src...``.

InvincibleRMC, 2024-07-20:
Ok I did that and get this error instead.

```
[INFO] [launch]: All log files can be found below /home/rov/.ros/log/2024-07-20-13-37-19-656413-rov-P55V4-16951
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [16954]
[camera_driver_node-1] [INFO] [1721497039.786676670] [23473566]: reading ros parameters for camera with serial: 23473566
[camera_driver_node-1] [INFO] [1721497039.786736392] [23473566]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1721497039.786820924] [23473566]: parameter definitions file: /home/rov/rov-25/src/surface/rov_flir/config/test_blackfly_s.yaml
[camera_driver_node-1] [INFO] [1721497039.790494955] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1721497039.790591266] [flir_camera]: camera calibration URL: file:///home/rov/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [ERROR] [1721497039.790714200] [camera_calibration_parsers]: Unable to open camera calibration file [/home/rov/.ros/camera_info/flir_camera.yaml]
[camera_driver_node-1] [WARN] [1721497039.790875885] [flir_camera]: Camera calibration file /home/rov/.ros/camera_info/flir_camera.yaml not found
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1]   what():  Spinnaker: System instance cannot be acquired.
[camera_driver_node-1] Could not load producer. Make sure that the environment variable SPINNAKER_GENTL64_CTI exists, and points to the location of the file Spinnaker_GenTL.cti [-1012]
[ERROR] [camera_driver_node-1]: process has died [pid 16954, exit code -6, cmd '/opt/ros/jazzy/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r __node:=flir_camera --params-file /tmp/launch_params_rl0w4uva --params-file /tmp/launch_params_x083vomv -r ~/control:=/exposure_control/control'].
```

berndpfrommer, 2024-07-20:
Check that you have nothing else running, like another copy of the driver, or SpinView or ....

berndpfrommer, 2024-07-20:
Hmm, you are running Jazzy. Then you are probably running on Ubuntu 24.04, which is not supported by the Spinnaker SDK.
Please try installing the Spinnaker SDK nevertheless. It will croak and not complete because of some libavi dependencies missing. However it should set up all the udev files correctly.

InvincibleRMC, 2024-07-20:
Ok I ran the install script and it crashed on the configuration step of udev if I skip that step the rest of the install completes without issue however still getting the same error as before.

```
Would you like to add a udev entry to allow access to USB hardware?
  If a udev entry is not added, your cameras may only be accessible by running Spinnaker as sudo.
[Y/n] $ 
Launching udev configuration script...

This script will assist users in configuring their udev rules to allow
access to USB devices. The script will create a udev rule which will
add FLIR USB devices to a group called flirimaging. The user may also
choose to restart the udev daemon. All of this can be done manually as well.

Adding new members to usergroup flirimaging...
Current members of flirimaging group: rov
To add a new member please enter username (or hit Enter to continue):
$ 

Writing the udev rules file...
Do you want to restart the udev daemon?
[Y/n] $ 
configure_spinnaker.sh: 77: /etc/init.d/udev: not found
```

berndpfrommer, 2024-07-20:
well, not quite sure if I hacked the install script ... but can you try and run again to see if it works now? After all, something was done by the installer. Don't forget to log out/back in, best to reboot the host.

InvincibleRMC, 2024-07-20:
The install script from spinnaker or from this repo?

berndpfrommer, 2024-07-20:
The spinnaker one, but I don't remember for sure. I do recall that I got it working on jazzy.
But can you try despite the error message you got?

InvincibleRMC, 2024-07-20:
I'm now getting the following error

```
[INFO] [launch]: All log files can be found below /home/rov/.ros/log/2024-07-20-14-52-34-723816-rov-P55V4-4789
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [4792]
[camera_driver_node-1] [INFO] [1721501554.860971092] [23473566]: reading ros parameters for camera with serial: 23473566
[camera_driver_node-1] [INFO] [1721501554.861050434] [23473566]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1721501554.861314943] [23473566]: parameter definitions file: /home/rov/rov-25/src/surface/rov_flir/config/test_blackfly_s.yaml
[camera_driver_node-1] [INFO] [1721501554.866408514] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1721501554.866546972] [flir_camera]: camera calibration URL: file:///home/rov/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [ERROR] [1721501554.866617644] [camera_calibration_parsers]: Unable to open camera calibration file [/home/rov/.ros/camera_info/flir_camera.yaml]
[camera_driver_node-1] [WARN] [1721501554.866682062] [flir_camera]: Camera calibration file /home/rov/.ros/camera_info/flir_camera.yaml not found
[camera_driver_node-1] [INFO] [1721501559.864684785] [23473566]: using spinnaker lib version: 4.0.0.116
[camera_driver_node-1] [WARN] [1721501559.864854600] [23473566]: no camera found with serial: 23473566 on try # 1
[camera_driver_node-1] [INFO] [1721501564.601652809] [23473566]: found camera with serial number: 23473566
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1]   what():  Spinnaker: Error reading from DevicePort [SN:23473566], at register address: 00002090. Please try reconnecting the device. [-1006]
[ERROR] [camera_driver_node-1]: process has died [pid 4792, exit code -6, cmd '/opt/ros/jazzy/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r __node:=flir_camera --params-file /tmp/launch_params_84m_4573 --params-file /tmp/launch_params_byu12bv6 -r ~/control:=/exposure_control/control'].
```

berndpfrommer, 2024-07-20:
It usually finds it first try. Now it needs 2 tries, and fails on the second one.
I assume you have tried reconnecting. Also check that your USB bus is really running at USB3 speed levels.
Did the SDK manage to install the example files in /opt/spinnaker/bin? If yes, then run /opt/spinnaker/bin/Acquisition to see if that works.

InvincibleRMC, 2024-07-20:
I'm not entirely sure what steps I did following your feedback but it now seems to work. The USB3 speed seemed to already be set correctly and /opt/spinnaker/bin/Acquisition worked.

berndpfrommer, 2024-07-20:
Sorry, you are running GigE. Should've looked at your info more carefully.

InvincibleRMC, 2024-07-20:
It worked for a little bit of  time however on a different launch, now on a follow up launch it seems stuck not publishing.

```
camera_driver_node-1] [INFO] [1721506628.111214531] [23473566]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1721506628.115380999] [23473566]: setting ChunkDataControl/ChunkSelector to: Timestamp
[camera_driver_node-1] [INFO] [1721506628.116679443] [23473566]: setting ChunkDataControl/ChunkEnable to: 1
[camera_driver_node-1] [INFO] [1721506628.151788604] [23473566]: camera has pixel format: BayerRG8
[camera_driver_node-1] [INFO] [1721506628.151851055] [flir_camera]: camera_driver_node started up!
[camera_driver_node-1] [INFO] [1721506630.633979569] [23473566]: rate [Hz] in  39.97 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506635.633956127] [23473566]: rate [Hz] in  40.05 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506640.634012295] [23473566]: rate [Hz] in  40.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506645.633984240] [23473566]: rate [Hz] in  40.01 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506650.633974034] [23473566]: rate [Hz] in  40.04 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506655.633949782] [23473566]: rate [Hz] in  40.01 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506660.634026985] [23473566]: rate [Hz] in  40.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506665.633978739] [23473566]: rate [Hz] in  40.03 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506670.633986622] [23473566]: rate [Hz] in  40.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506675.633999761] [23473566]: rate [Hz] in  39.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506680.634022145] [23473566]: rate [Hz] in  39.99 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506685.634008996] [23473566]: rate [Hz] in  40.13 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506690.634015171] [23473566]: rate [Hz] in  40.03 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506695.634015493] [23473566]: rate [Hz] in  40.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506700.634014712] [23473566]: rate [Hz] in  40.04 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506705.633924483] [23473566]: rate [Hz] in  39.84 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506710.634010610] [23473566]: rate [Hz] in  40.07 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506715.633954568] [23473566]: rate [Hz] in  40.13 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506720.634017249] [23473566]: rate [Hz] in  40.16 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506725.634024208] [23473566]: rate [Hz] in  40.13 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1721506730.633979678] [23473566]: rate [Hz] in  40.13 out   0.00 drop   0%
```

I'm subscribing to it but, it does not seem to publish.

```
rov@rov-P55V4:~$ ros2 topic info /front_cam/image_raw 
Type: sensor_msgs/msg/Image
Publisher count: 0
Subscription count: 1
```

InvincibleRMC, 2024-07-20:
Ok I'm now having problems with my specific configuration file before I was testing with the example. I do not have time to continue debugging but, will pick this up next weekend. Leaving my specific files attached.

config file

```yaml
#
# config file for blackfly S cameras (USB3 and GigE)
#
# This file maps the ros parameters to the corresponding Spinnaker "nodes" in the camera.
# For more details on how to modify this file, see the README on camera configuration files.

parameters:
  #
  # -------- image format control
  #
  - name: pixel_format
    type: enum
    # Check available values with SpinView. Not all are supported by ROS!
    # Some formats are e.g. "Mono8", "BayerRG8", "BGR8", "BayerRG16"
    # default is "BayerRG8"
    node: ImageFormatControl/PixelFormat
  - name: image_width
    type: int
    node: ImageFormatControl/Width
  - name: image_height
    type: int
    node: ImageFormatControl/Height
  - name: offset_x  # offset must come after image width reduction!
    type: int
    node: ImageFormatControl/OffsetX
  - name: offset_y
    type: int
    node: ImageFormatControl/OffsetY
    # Added by ROV
  - name: binning_x
    type: int
    node: ImageFormatControl/BinningHorizontal
  - name: binning_y
    type: int
    node: ImageFormatControl/BinningVertical
  #
  # -------- analog control
  #
  - name: gain_auto
    type: enum
    # valid values are "Continuous", "Off"
    node: AnalogControl/GainAuto
  - name: gain
    type: float
    node: AnalogControl/Gain
  #
  # -------- device link throughput limiting
  #
  - name: device_link_throughput_limit
    type: int
    node: DeviceControl/DeviceLinkThroughputLimit
  #
  # -------- transport layer control (GigE)
  #
  - name: gev_scps_packet_size
    type: int
    # default is 1400. Set to 9000 to enable jumbo frames, ensure NIC MTU set >= 9000
    node: TransportLayerControl/GigEVision/GevSCPSPacketSize
  #
  # -------- digital IO control
  #
  - name: line0_selector  # black wire: opto-isolated input
    type: enum
    node: DigitalIOControl/LineSelector
  - name: line1_selector  # white wire: opto-isolated output
[... 160 lines cut ...]
            parameters,
        ],
        condition=IfCondition(front_arg),
    )

    # launches node to run bottom flir camera
    bottom_cam = Node(
        package='spinnaker_camera_driver',
        executable='camera_driver_node',
        name='bottom_cam',
        exec_name='bottom_cam',
        emulate_tty=True,
        output='screen',
        parameters=[
            Parameter('serial_number', '23473577'),
            Parameter('parameter_file', parameter_file),
            parameters,
        ],
        condition=IfCondition(bottom_arg),
    )

    namespace_launch = GroupAction(actions=[PushRosNamespace(ns_arg), front_cam, bottom_cam])

    return LaunchDescription([namespace_launch])
```

berndpfrommer, 2024-07-20:
If this is a different issue, please open a new issue, thanks!

closed 2024-07-27 by InvincibleRMC