ros-drivers/flir_camera_driver #201: [humble]

samchopra2003, 2024-07-25:
**Is this behavior also observed when using SpinVIew?**
No this problem is not observed when using SpinView.

**Describe the bug**
I have a two camera setup: Flir Blackfly S and Flir Boson 640. The Boson is a v4l2 device, and I am using another driver for this. Running `ros2 launch spinnaker_camera_driver driver_node.launch.py` results in v4l2 devices not being detected anymore, despite lsusb showing otherwise.

**Useful logging output**
Here is the output of `ls /dev/video*` and `lsusb -tv` before and during running:`ros2 launch spinnaker_camera_driver driver_node.launch.py`. Note that the Point Grey Research device corresponds to the Blackfly S and the Flir Systems Breach corresponds to the Boson.

_Before running launch file:_
[image]
[image]

_After running launch file:_
[image]
[image]

Here is also a interesting dmesg output after running the launch file (Boson is device number 7 and Blackfly is device number 2):
[image]

**System details**
- operating system/version: Ubuntu 22.04.4 LTS (Jammy Jellyfish)
- ROS version: ROS 2 Humble
- complete FLIR camera model: BFS -U3-32S4C-C

berndpfrommer, 2024-07-25:
Seems like there is a USB bus reset happening also for cameras that are not used.
Question: did you install the driver from binaries or did you compile it from source? If you compiled it from source: there is a chance that some very recent changes (PR #194) are causing this, so I'd like you to use the binaries from the apt repo (which don't have those changes yet), or revert to tag 2.0.20.

Can you please also test if
```
/opt/spinnaker/bin/Acquisition
```
causes the Boson video device to disappear?

samchopra2003, 2024-07-25:
I compiled the driver from source. I have tested `/opt/spinnaker/bin/Acquisition` and it works perfectly. The Boson does not disappear. Let me test the tag 2.0.20.

samchopra2003, 2024-07-25:
This tag works perfectly! Thanks! I am able to get both cameras working!

closed 2024-07-25 by samchopra2003

berndpfrommer, 2024-07-25:
@samchopra2003 Would you mind testing PR #202 to see if that leaves your /dev/video device intact? Thanks!