ros-drivers/flir_camera_driver #204: [iron] Acquisition Timeout Teledyne DALSA Calibir GX

rebeccasc, 2024-08-18:
**Is this behavior also observed when using SpinVIew?**
No

**Describe the bug**
I want to use the ROS flir camera driver with my [Calibir GX](https://www.teledynedalsa.com/en/products/imaging/infrared-detectors/calibir-gx/) GigE camera. I added a configuration file, as described [here](https://docs.ros.org/en/rolling/p/spinnaker_camera_driver/#how-to-develop-your-own-camera-configuration-file). When I try to run the camera via the ROS launch script I get a `[WARNING]: acquisition timeout, restarting!`. However, the camera works fine (without the warning) if it was used with SpinView right before. That means it works properly with SpinView and if I disconnect from SpinView and start running the ROS driver afterwards, everything works fine. However, if I then disconnect the camera from the power supply and reconnect it and then try to restart the ROS driver, I get the acquisition error again. It seems that SpinView sets some important camera parameters that I do not set in the launch file. I have gone through all the parameters in SpinView and cannot find the missing ones.

**Useful logging output**
```
[INFO] [camera_driver_node-1]: process started with pid [30078]
[camera_driver_node-1] [INFO] [1723818024.786684416] [XXXXXXXX]: reading ros parameters for camera with serial: XXXXXXXX
[camera_driver_node-1] [INFO] [1723818024.786808960] [XXXXXXXX]: not adjusting time stamps!
[camera_driver_node-1] [INFO] [1723818024.786926795] [XXXXXXXX]: parameter definitions file: /home/XXX/ros2_ws/install/spinnaker_camera_driver/share/spinnaker_camera_driver/config/calibir_gx.yaml 
[camera_driver_node-1] [INFO] [1723818024.791361078] [lwir_ns. flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1723818024.791457621] [lwir_ns. flir_camera]: camera calibration URL: file:///home/XXX/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] [ERROR] [1723818024.791516247] [camera_calibration_parsers]: Unable to open camera calibration file [/home/XXX/.ros/camera_info/flir_camera.yaml]
[camera_driver_node-1] [WARN] [1723818024.791574289] [lwir_ns. flir_camera]: Camera calibration file /home/XXX/.ros/camera_info/flir_camera.yaml not found
[camera_driver_node-1] [INFO] [1723818025.760135228] [XXXXXXXX]: using spinnaker lib version: 4.0.0.116
[camera_driver_node-1] [INFO] [1723818029.254376158] [XXXXXXXX]: found camera with serial number: XXXXXXXX
[camera_driver_node-1] [INFO] [1723818030.176243386] [XXXXXXXX]: setting ImageFormatControl/PixelFormat to: Mono8
[camera_driver_node-1] [INFO] [1723818030.178422172] [XXXXXXXX]: setting Image FormatControl/pixelPolarity to: WhiteHot
[camera_driver_node-1] [INFO] [1723818030.180143909] [XXXXXXXX]: setting ImageFormatControl/rawOutputMode to: Inactive
[camera_driver_node-1] [INFO] [1723818030.181361054] [XXXXXXXX]: setting DeviceControl/DeviceLinkHeartbeatMode to: On 
[camera_driver_node-1] [INFO] [1723818030.183979390] [XXXXXXXX]: setting DeviceControl/DeviceLinkHeartbeatTimeout to: 5000000
[camera_driver_node-1] [INFO] [1723818030.185482956] [XXXXXXXX]: setting AcquisitionControl/AcquisitionMode to: Continuous 
[camera_driver_node-1] [INFO] [1723818030.186408430] [XXXXXXXX]: setting AcquisitionControl/Acquisition FrameCount to: 1 
[camera_driver_node-1] [INFO] [1723818030.187250486] [XXXXXXXX]: setting AcquisitionControl/transfer FrameSkip to: 0
[camera_driver_node-1] [INFO] [1723818030.187767911] [XXXXXXXX]: setting AcquisitionControl/continuousTransfer Mode to: Off
[camera_driver_node-1] [INFO] [1723818030.188067749] [XXXXXXXX]: setting AcquisitionControl/AcquisitionStatusSelector to: AcquisitionActive
[camera_driver_node-1] [INFO] [1723818030.188843012] [XXXXXXXX]: setting ChunkDataControl/ChunkModeActive to: 0
[camera_driver_node-1] [INFO] [1723818030.189429564] [XXXXXXXX]: setting TriggerControl/TriggerMode to: Off
[camera_driver_node-1] [INFO] [1723818030.189867363] [XXXXXXXX]: setting TransportLayer Control/GevSCPSPacketSize to: 1500
[camera_driver_node-1] [INFO] [1723818030.191231802] [XXXXXXXX]: setting Transport Layer Control/GevSCCFGExtendedChunkData_to: 1
[camera_driver_node-1] [INFO] [1723818030.191968856] [XXXXXXXX]: setting TransportLayer Control/GevHeartbeatTimeout to: 5000
[camera_driver_node-1] [INFO] [1723818030.295947129] [XXXXXXXX]: camera has pixel format: Mono8
[camera_driver_node-1] [INFO] [1723818030.296040464] [lwir_ns.flir_camera]: camera_driver_node started up! 
[camera_driver_node-1] [INFO] [1723818030.296438887] [XXXXXXXX]: rate [Hz] in 0.00 out 0.00 drop 0%
[camera_driver_node-1] WARNING: acquisition timeout, restarting!
[camera_driver_node-1] WARNING: acquisition timeout, restarting!
[camera_driver_node-1] [INFO] [1723818034.787098500] [XXXXXXXX]: rate [Hz] in 0.00 out 0.00 drop 0%
[camera_driver_node-1] [INFO] [1723818039.786950871] [XXXXXXXX]: rate [Hz] in 0.00 out 0.00 drop 0%
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1] what(): Spinnaker: Unable to set StreamBufferCountMax due to failure to determine image size or payload size. [-1009]
[ERROR] [camera_driver_node-1]: process has died [pid 30078, exit code -6, cmd '/home/XXX/ros2_ws/install/spinnaker_camera_driver/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r _node:=flir_camera -r _ns:=/lwir_ns --params-file /tmp/launch_params_36fl1f6s --params-file /tmp/launch_params_t_bwucjo -r ~/control:=/exposure_control/control'].
```

**System details**
- operating system/version: ubuntu-22.04.3
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]): iron
- complete FLIR camera model (e.g. BFS-U3-04S2C-C): Calibir GX

berndpfrommer, 2024-08-18:
Since I don't have a Calibir GX at hand the following remarks are somewhat shots in the dark
1) Some of the other Spinnaker based cameras allow for a UserSet to be loaded. Have a look for instance at the Blackfly S camera configs and the launch file for loading e.g. UserSet0. The procedure would be to save UserSet0 from within SpinView, then loading UserSet0 when launch the ROS2 driver.
2) This line from the log is concerning. No idea what it means and what triggers it.
```
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1] what(): Spinnaker: Unable to set StreamBufferCountMax due to failure to determine image size or payload size. [-1009]
```
3) I'm confused as to why the driver didn't log more error messages. The lines below seem to have invalid spaces (between Transport and Layer), and between TransportLayerControl etc. Why are there no error messages? This should never work.
```
[camera_driver_node-1] [INFO] [1723818030.189867363] [XXXXXXXX]: setting TransportLayer Control/GevSCPSPacketSize to: 1500
[camera_driver_node-1] [INFO] [1723818030.191231802] [XXXXXXXX]: setting Transport Layer Control/GevSCCFGExtendedChunkData_to: 1
```
4) Since this is a GigE camera, you may want to use Jumbo packets (5000 instead of 1500), also make sure your network card MTU is set to 5000. But this should not be a problem since SpinView also operates at 1500 MTU, right?

5) Again in the log it says:
```
[camera_driver_node-1] [INFO] [1723818030.187767911] [XXXXXXXX]: setting AcquisitionControl/continuousTransfer Mode to: Off
```
It seems to me that the capitalization and spacing is wrong (should be ContinuousTransferMode), and if anything should be set to "On" rather than off.

Where those strange inconsistencies introduced when editing the log file to obscure the serial number?

rebeccasc, 2024-08-21:
@berndpfrommer thanks a lot for the fast reply!

Actually, the spacing is some weird copy and paste error. 
Setting `ContinuousTransferMode` to `On` solved the problem. Thank you very much for your help! :)

closed 2024-08-21 by rebeccasc