ros-drivers/flir_camera_driver #210: Errors with Spinnaker Camera Driver during stereo.launch, camera.launch : "Feature name AcquisitionFrameRateEnable not implemented." and "Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]"

mo-aqeel, 2024-09-13:
**Description:**
I am setting up two cameras with external triggering from an IMU (using Line0) at a frequency of 20Hz. My goal is to run OpenVINS. However, when I launch either the stereo.launch or camera.launch files, 
 The error I see is:

1. [ERROR] Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]

This error does not occur every time I run the launch file but is persistent enough to hinder smooth operation.

Additionally, the AcquisitionFrameRateEnable feature appears to be critical for synchronizing the cameras with the external trigger, yet the following error occurs:

1. [ERROR] Feature name AcquisitionFrameRateEnable not implemented.

Since controlling the acquisition frame rate is crucial for my use case, especially for running OpenVINS, I need assistance resolving these issues before proceeding.

Since this is my first time configuring cameras with ROS, I'm unsure if the issue is with my configuration or a driver limitation.

**Setup:**
    ROS Distro: Noetic
    Cameras: [Left: 21377652, Right: 21342999]
    Launch File: stereo.launch

**stereo.launch**
<?xml version="1.0"?>

<launch>
  
  <arg name="camera_name"               default="stereo" />

  
  <arg name="control_frame_rate"        default="True" />
  <arg name="frame_rate"                default="40" />

  <arg name="left_camera_serial"        default="21377652" />
  <arg name="left_camera_calibrated"    default="False" />

  <arg name="right_camera_serial"       default="21342999" />
  <arg name="right_camera_calibrated"   default="False" />

  <arg name="run_stereo_image_proc"     default="True" />
  <arg name="trigger"                   default="On" />

  
  
  <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" />

  <group ns="$(arg camera_name)" >
    <group ns="left" >
      
      <node pkg="nodelet" type="nodelet" name="camera_nodelet"
            args="load spinnaker_camera_driver/SpinnakerCameraNodelet /camera_nodelet_manager" >
        <param name="frame_id"                        value="camera_left" />
        <param name="serial"                          value="$(arg left_camera_serial)" />

        
        <param name="acquisition_frame_rate_enable"   value="$(arg control_frame_rate)" />
        <param name="acquisition_frame_rate"          value="$(arg frame_rate)" />
        <param name="enable_trigger"                  value="$(arg trigger)" />

        
        <param name="camera_info_url" if="$(arg left_camera_calibrated)"
               value="file://$(env HOME)/.ros/camera_info/$(arg left_camera_serial).yaml" />
      </node>

      
      <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer /camera_nodelet_manager">
      </node>
    </group>

    <group ns="right" >
[... 134 lines cut ...]
[ERROR] [1726227603.274541955]: [SpinnakerCamera]: (21342999) Feature name AcquisitionFrameRateEnable not implemented.
[ WARN] [1726227603.279811173]: [SpinnakerCamera]: (21342999) Enumeration LineMode not writable.
[ WARN] [1726227603.282682210]: [SpinnakerCamera]: (21342999) Enumeration LineSource not available.
[ERROR] [1726227603.284272366]: [SpinnakerCamera]: (21342999) Feature name AutoExposureExposureTimeUpperLimit not implemented.
[ WARN] [1726227603.284360802]: [SpinnakerCamera]: (21342999) Enumeration GainSelector not writable.
[ERROR] [1726227603.286105384]: [SpinnakerCamera]: (21342999) Feature name GammaEnable not implemented.
[ WARN] [1726227603.286753284]: [SpinnakerCamera]: (21342999) Feature Gamma not writable.
[ERROR] [1726227603.888668830]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]
[ERROR] [1726227603.889032343]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
[ERROR] [1726227604.303353029]: [SpinnakerCamera::grabImage] Failed to retrieve buffer with error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]
[ERROR] [1726227604.303683625]: Failed to disconnect with error: [SpinnakerCamera::disconnect] Failed to disconnect camera with error: Spinnaker: Can't de-initialize camera. Camera is still streaming. [-1004]
^C[stereo/stereo_image_proc-6] killing on exit
[stereo/right/image_proc_debayer-5] killing on exit
[stereo/right/camera_nodelet-4] killing on exit
[stereo/left/image_proc_debayer-3] killing on exit
[stereo/left/camera_nodelet-2] killing on exit
[camera_nodelet_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

**System details**
- operating system/version: Ubuntu 20.04
- ROS version : noetic
- complete FLIR camera model : BFLY-U3-23S6C-C

berndpfrommer, 2024-09-13:
This is not a ROS2 problem. The ROS1 driver is largely unsupported at this point.

mo-aqeel, 2024-09-13:
Thank you for the response, and I apologize for the confusion regarding the label. I mistakenly marked it as 'ROS2'.

Would you happen to have any insight into where this issue might be originating from in the ROS1

mo-aqeel, 2024-09-13:
Thank you for your response. If anyone has insights on how I can resolve this issue, especially regarding the AcquisitionFrameRateEnable and the : Failed waiting for EventData on NEW_BUFFER_DATA event, please let me know. I’d greatly appreciate any guidance on this!

mgrova, 2024-10-19:
Dear, @mo-aqeel

I have the same problem with FLIR A65. When I run the camera, I sometimes get the same error after a while.
Have you been able to fix it?

[image]

Thanks in advance!

Bests, 
Marco.

mo-aqeel, 2024-10-23:
Dear, @mgrova,

Sorry for the late reply, I was using the setup to perform Visual Inertial Odometry, where my camera was triggered by an IMU. When the IMU failed to provide the trigger signal properly, the terminal displayed this error. After fixing the trigger, everything worked fine.