ros-drivers/flir_camera_driver #212: [jazzy] node not starting

SeapathMaster, 2024-09-30:
**Is this behavior also observed when using SpinVIew?**
Camera works in SpinView under Ubuntu 22.04. However, in Ubuntu 24.04 the currently available versions of SpinView don't work.

**Describe the bug**
I installed the spinnaker-camera-driver under ROS2 Jazzy Jalisco using the standard installation method:
`sudo apt install ros-jazzy-spinnaker-camera-driver`
As there is no valid version for Spinnaker under Ubuntu 24.04, I followed the instruction to set the driver up for use without Spinnaker SDK. I confirmed the correct setup.
Starting the node, it throws an exception and exits with ` what():  Spinnaker: System instance cannot be acquired.`
The spinnaker-camera-driver works with this camera on another PC with Ubuntu 22.04 and ROS2 Humble and Spinnaker 3.2.0.62 installed.

**Useful logging output**
`>>ros2 launch spinnaker_camera_driver driver_node.launch.py camera_type:=blackfly_s serial:="'24134247'"

[INFO] [launch]: All log files can be found below /home/superhuman/.ros/log/2024-09-30-15-24-33-639763-rosdev00-9820
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [camera_driver_node-1]: process started with pid [9827]
[camera_driver_node-1] [INFO] [1727677473.953373414] [24134247]: reading ros parameters for camera with serial: 24134247
[camera_driver_node-1] [INFO] [1727677473.953540126] [24134247]: adjusting time stamps!
[camera_driver_node-1] [INFO] [1727677473.953809714] [24134247]: parameter definitions file: /opt/ros/jazzy/share/spinnaker_camera_driver/config/blackfly_s.yaml
[camera_driver_node-1] [INFO] [1727677473.960790660] [flir_camera]: using default calibration URL
[camera_driver_node-1] [INFO] [1727677473.960900730] [flir_camera]: camera calibration URL: file:///home/rosuser/.ros/camera_info/flir_camera.yaml
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1]   what():  Spinnaker: System instance cannot be acquired.
[camera_driver_node-1] Could not load producer. Make sure that the environment variable SPINNAKER_GENTL64_CTI exists, and points to the location of the file Spinnaker_GenTL.cti [-1012]
[ERROR] [camera_driver_node-1]: process has died [pid 9827, exit code -6, cmd '/opt/ros/jazzy/lib/spinnaker_camera_driver/camera_driver_node --ros-args -r __node:=flir_camera --params-file /tmp/launch_params_b_i927z4 --params-file /tmp/launch_params_02n6a2o4 -r ~/control:=/exposure_control/control'].`

**System details**
- operating system/version: Ubuntu 24.04 LTS
- ROS version (please prepend to issue title in square brackets, e.g. [noetic]): ROS2 Jazzy Jalisco
- complete FLIR camera model (e.g. BFS-U3-04S2C-C): BFS-U3-31S4C

SeapathMaster, 2024-09-30:
I fixed it!
It is necessary to add an environment variable:
`export SPINNAKER_GENTL64_CTI=/opt/ros/jazzy/lib/spinnaker-gentl/Spinnaker_GenTL.cti`

I'd propose to add this to the script `linux_setup_flir` or to the installation process, setting the environment variable in the local_setup files of this package.

berndpfrommer, 2024-11-06:
Closing this with #214

closed 2024-11-06 by berndpfrommer