ros-drivers/flir_camera_driver #216: [HUMBLE] Ubuntu 24.04 - Brightness not computed for non-BayerRG8 pixel formats

GreatAlexander, 2024-11-19:
Hi there, first of all thank you for all your hard work on this ROS driver. We really appreciate this!

**Is this behavior also observed when using SpinView?**
N/A. Brightness computation is an output of the ROS driver.

**Describe the bug**
Brightness value (as produced in the `/meta` topic) is [not computed for non-BayerRG8 pixel formats](https://github.com/ros-drivers/flir_camera_driver/blob/humble-devel/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp#L289) (i.e. BayerBG8).

We are using this driver with some cameras (CM3-U3-50S5C) installed upside down, which flips the Bayer format from BayerRG8 to BayerBG8. We noticed that when `compute_brightness` is enabled, the brightness output is stuck at 0 and the brightness controller does not work as expected. With cameras not upside down, compute brightness with BayerRG8 pixel format works as expected. 

We have tested removing the [if statement](https://github.com/ros-drivers/flir_camera_driver/blob/humble-devel/spinnaker_camera_driver/src/spinnaker_wrapper_impl.cpp#L289) and we can confirm the brightness computation works with BayerBG8 too as expected. We are happy to provide a PR to add BayerBG8 as a pixel format supported and tested for brightness computation if you would prefer.

**Useful logging output**
None, but we are happy to provide evidence and data to demonstrate this behaviour.

**System details**
- Operating system/version: Ubuntu22.04 ([Dockerised](https://github.com/ipab-rad/av_camera))
- ROS version: Humble 
- Complete FLIR camera model: CM3-U3-50S5C

Notes:
- We are running a system with 6 x CM3-U3-50S5C cameras (via USB3), using the synchronised driver and externally triggered via custom hardware. We want to use the brightness controller, as it provides better direct control over the brightness of the image.
- We have added a couple of new camera parameters supported by our camera model, specifically the reverse X/Y params (Since some of our cameras are installed upside down). FYI [our repo](https://github.com/ipab-rad/av_camera)
- We are happy to support this driver and test/evaluate with our specific camera models any versions/configurations as requested.

Our camera parameter configuration (example):
```
[rsp_l]: reading ros parameters for camera with serial: 17496130
[rsp_l]: adjusting time stamps!
[rsp_l]: parameter definitions file: /opt/ros_ws/install/av_camera_launch/share/av_camera_launch/config/chameleon.yaml
[sensor.camera]: camera calibration URL: file:///opt/ros_ws/install/av_camera_launch/share/av_camera_launch/config/rsp_l_calib.yaml
[rsp_l]: using spinnaker lib version: 3.1.0.79
[rsp_l]: found camera with serial number: 17496130
[Spinnaker Wrapper]: Initialized camera [serial: 17496130] from: [USB Interface 0]
[rsp_l]: setting ImageFormatControl/PixelFormat to: BayerBG8
[rsp_l]: setting ImageFormatControl/Width to: 2400
[rsp_l]: setting ImageFormatControl/Height to: 1200
[rsp_l]: setting ImageFormatControl/OffsetX to: 24
[rsp_l]: setting ImageFormatControl/OffsetY to: 522
[rsp_l]: setting ImageFormatControl/ReverseX to: 1
[rsp_l]: setting ImageFormatControl/ReverseY to: 1
[rsp_l]: setting ImageFormatControl/VideoMode to: Mode0
[rsp_l]: setting AnalogControl/GainAuto to: Off
[rsp_l]: setting AcquisitionControl/ExposureAuto to: Off
[rsp_l]: setting AcquisitionControl/ExposureTime to: 1000
[rsp_l]: setting AcquisitionControl/TriggerMode to: On
[rsp_l]: setting AcquisitionControl/AcquisitionMode to: SingleFrame
[rsp_l]: setting AcquisitionControl/TriggerSelector to: FrameStart
[rsp_l]: setting AcquisitionControl/TriggerSource to: Line0
[rsp_l]: setting AcquisitionControl/TriggerDelay to: 0
[rsp_l]: setting AcquisitionControl/TriggerOverlap to: ReadOut
[rsp_l]: setting ChunkDataControl/ChunkModeActive to: 1
[rsp_l]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[rsp_l]: setting ChunkDataControl/ChunkEnable to: 1
[rsp_l]: setting ChunkDataControl/ChunkSelector to: Gain
[rsp_l]: setting ChunkDataControl/ChunkEnable to: 1
[rsp_l]: setting ChunkDataControl/ChunkSelector to: Timestamp
[rsp_l]: setting ChunkDataControl/ChunkEnable to: 1
[rsp_l]: camera has pixel format: BayerBG8
```

berndpfrommer, 2024-11-19:
Thanks for the feedback! Would love to see a PR for the parameters you added to chameleon.yaml

berndpfrommer, 2024-11-19:
Oh, and would you mind testing PR #217 ? I just implemented it, but have very limited time right now for testing.

berndpfrommer, 2024-11-21:
I just submitted PRs to the packaging system. You should see the changes after the next "sync" for humble, which I expect to happen some time early/mid December.

closed 2024-11-21 by berndpfrommer